darkxst/silabs-firmware-builder #4: Zvidar Z-DG-Z01 (EFR32MG21) USB Zigbee Coordinator dongle manifest?

Hedda, 2023-06-02:
FYI, here is another new Silicon Labs EFR32MG21 based Zigbee USB dongle, no more info available yet on the pins/pads used:

https://zigbee.blakadder.com/Zvidar_Z-DG-Z01.html

http://www.zvidar.com

https://www.domadoo.fr/en/interface-domotique/6500-zvidar-zigbee-usb-dongle-efr32mg21-chipset-compatible-thread-and-matter.html

https://shop.smarthome-europe.com/en/interface-domotique/5486-zvidar-zigbee-usb-dongle-efr32mg21-chipset-compatible-thread-and-matter.html

https://www.tronika.no/en/smarthome/zigbee-products/zigbee-interfaces/interface-z-dg-z01.html

PS: I assume this is probably a clone/replica of the ZB-GW04 USB dongle or the Home Assistant SkyConnect USB adapter?

darkxst, 2023-06-11:
I will look at adding this, once there is sufficient info available.

Hedda, 2023-06-19:
Apparently, it is USB discovered by VID/PID "**_1a86:55d4_**" and product USB identifier  "**_QinHeng Electronics USB Single Serial_**" or with serial by-id as "**_usb11a86_USB_Single_Serial_54F7005713-if00_**" (where "54F7005713" is the serial number).

https://www.domo-blog.fr/zvidar-zigbee-matter-thread-dongle-domotique-jeedom-home-assistant/

darkxst, 2023-06-19:
This would be the generic data for the CH9102 chip.