darkxst/silabs-firmware-builder #6: EFR32 MG1B232GG chip build-in Beeline SmartBox TURBO+ WAN-LAN router

oleging, 2023-06-17:
Greetings. Is it possible to create a firmware for this device?

More information https://openwrt.org/inbox/toh/beeline/smartbox_turbo_plus#zigbee
[Latest firmware](url=https://github.com/grobasoz/zigbee-firmware/blob/master/EFR32%20Series%201/EFR32MG1B-256k/NCP/NCP_USW_EFR32MG1B-256_678_PA0-PA1-PB11.ebl)

csharper2005, 2023-06-17:
Would be nice to have this firmware.

darkxst, 2023-06-19:
Which firmware are you after? EZSP or Multipan?
Yes it should be possible, but I dont have hardware to test. This is a somewhat smaller chip flash/RAM than what we currently build for.

It might require a new bootloader as well, which this builder is not currently setup to build.

Is there anyway to reset into bootloader on this device? Is there a GPIO pin hooked up for this? Could be some chance of bricking it otherwise.

https://github.com/grobasoz/zigbee-firmware/issues/3

oleging, 2023-06-19:
Multipan.

darkxst, 2023-06-19:
It maybe possible to flash the bootloader over serial connection, provided the existing bootloader doesnt require signed images. I have never tested this though. Generally messing with bootloaders is a bit beyond the scope of what trying to do in this repo!

There are some bootloader images here, but not clear if these are the newer `Gecko Bootloader` .
https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32%20Series%201/EFR32MG1B-256k/Bootloaders

But if that bootloader was to suffice, I can build  multipan images.

oleging, 2023-06-19:
It seems for me that with this bootloader
[image]

darkxst, 2023-06-20:
Per the issue I posted above, you probably want this bootloader
[image]

If that works and is Gecko Bootloader (and not the older EFR Bootloader) I can build multipan image. Hopefully it will fit in the small 256k flash, no guarantee though.

oleging, 2023-06-20:
I will be very grateful. It's worth a try, if it doesn't work out, then there's no problem 🤷‍♂️

oleging, 2023-06-20:
But Bootloader Entry GPIO must be PB11.

darkxst, 2023-06-20:
I dont know which pin the bootloader is, but that issue I linked above mentioned PD15. Anyway there is no setting for Bootloader activation in the RCP firmware, I think its only set in the bootloader.

This chip does not support EmberZnet v7 but RCP Multipan Seems to build ok. You can find the firmwares at these links. You will need to be running Gecko Bootloader > 1.4.x? to be able to load these. They don't include bootloader so you will need to try load grobasoz bootloader first.

https://github.com/darkxst/silabs-firmware-builder/suites/13727165237/artifacts/759555073 (baudrate 230400)
https://github.com/darkxst/silabs-firmware-builder/suites/13727165237/artifacts/759555074 (baudrate 460800)

oleging, 2023-06-20:
Tnks. I'll try some time later.