darkxst/silabs-firmware-builder #8: Builds missing for Elelabs ELx0x3 (ELR023 and ELU013) Zigbee adapters?

Hedda, 2023-06-20:
https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds

Should also be built automatically or? 

https://github.com/darkxst/silabs-firmware-builder/blob/manifestjs/manifests/elelabs-elx0x3.json

PS: More discussion here for reference -> https://github.com/ksjh/silabs-firmware-builder/issues/17

darkxst, 2023-06-20:
Its not in the main branch just yet. Once I merge the manifest support into main it will be there. Until then the builds in that manifestjs branch should be fine, but haven't been tested.

darkxst, 2023-11-03:
I have rebased the manifest work and its ready to merge in #23.

darkxst, 2023-12-30:
These builds are in the main branch now.

closed 2023-12-30 by darkxst