darkxst/silabs-firmware-builder #10: HomeAssistant addon?

starkillerOG, 2023-07-19:
The [SL Web Flasher](https://darkxst.github.io/silabs-firmware-builder/) is awesom, @darkxst thank you very much for creating this!
The Multi-PAN firmware is working perfectly on my Sonoff ZBDongle-E.

Would you consider making a addon for HomeAssistant such that users (myself included) could update the firmware straight from the HomeAssistant device running HomeAssistant OS (in my case a intel nuc i5)?
Preferably a update enitity would let users know as soon as new firmware is available for their stick.

Or even better build it into the Silicon Labs Multiprotocol addon.
That already support the HomeAssistant Yellow and Skyconnect stick, but your version also supports the Sonoff ZBDongle-E and EasyIot ZB-GW04 v1.1 and v1.2.

darkxst, 2023-07-19:
I am still thinking of the best way to distribute updates!

Right now, you can use the official Silabs Flasher addon and manually provide a custom firmware link to flash updates from your HA box.
https://github.com/home-assistant/addons/tree/master/silabs_flasher

As a test I have created a fork of that addon here, that includes my firmware builds, this would automatically update firmware when an addon update is published (it may require a restart to trigger the update after installing the update).
https://github.com/darkxst/multipan_flasher/tree/main
Feel free to test this, I will push the 4.3.1 update here when its released in the coming weeks. Probably I will more broadly advertise it after testing that upgrade.

> Or even better build it into the HomeAssistant Silicon Labs Multiprotocol addon.

This would be ideal, but we can't include community built images in the addon as it would block the release until they are all ready. I don't think they will include a custom URL option in the multiprotocol addon, since flashing the wrong image can have dire consequences. But will try find a way to integrate with this down the track.

There has been support about setting up a central repo for community built silabs firmwares (since there are a few other silabs dongles that not yet supported by my builder. Once this happens may help with options getting it somehow linked with the official addon.

starkillerOG, 2023-07-23:
That is just awesom!

Is it save to run this addon while the Zigbee2MQTT and Silicon Labs Multiprotocol addon are running?
https://github.com/darkxst/multipan_flasher/blob/main/multipan_flasher/DOCS.md

Much appreciate your work!

darkxst, 2023-07-23:
Yes, it will briefly stop the multiprotocol addon at startup while checking/flashing firmware.

starkillerOG, 2023-07-23:
Nice!

Am I correct the sonoff zigbee dongle-E Multi Pan firmware does currently not support Flow control?

I remember in the default firmware flow control was listed as supported (I think).
But I think I got errors while trying to enable flow control in the siliconlabs multiprotocol addon.

darkxst, 2023-07-24:
That is correct, ZBDongle-E does not support hardware flow control. However I have not seen any issues caused by this, in fact it performs better than the ZB-GW0W with hardware flow control!

darkxst, 2023-07-28:
Ive pushed out an update with 4.3.1 firmwares. Once the addon is updated, it will automatically update the dongle on next reboot of HA (or you can manually run the addon to do it straight away).

starkillerOG, 2023-07-28:
@darkxst just updated the addon and run it, it stoppen the multiprotocol addon, flashen the new firmware and started the multiprotocol addon again perfectly.

Zigbee network back up within 30 seconds, perfect!

Thanks for the greath work!

supaeasy, 2023-08-01:
Soo.. I got this running on my ZBDongle-E and it said something about cross-flashing in the protocoll. Long story short: Z2M does not start anymore. Any Ideas? Where do I find logs of this to post here?
Before I installed Your Firmware 7.3.0 and it worked very well.

darkxst, 2023-08-01:
This addon delivers the RCP Multipan firmware for use with Silabs Multiprotocol Addon enabling simultaneous Zigbee and Thread support.  If that is not what you want you should go back to the NCP EZSP firmware for pure Zigbee, there is no similar addon for that yet, but you can use the official Silabs-Flasher addon and this URL:

https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.3.1.0-zbdonglee-115200.gbl

or my webflasher will be updated with 7.3.1 builds in the next few days.

If you want multiprotocol support you need to install the Silabs Multiprotocol Addon and update the serial port in Z2M (you may need to repair devices if you do this though). I have a [blog post](https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware) with lots more details about this

btw cross-flashing means it switched from EZSP firmware -> RCP Multipan firmware

supaeasy, 2023-08-01:
Yeah, that was my mistake. I used the webflasher to reflash 7.3.0 and it works normally again. 
However, when I use SiLabs Flasher with your provided link I get the following error:

`[20:23:55] INFO: Starting universal-silabs-flasher with /dev/ttyACM0 (bootloader baudrate 115200)
Error: '/root/firmware.gbl' does not appear to be a valid GBL image: ValidationError('Image is truncated: tag value is cut off')
[20:23:56] INFO: universal-silabs-flasher-up script exited with code 1`

Seems to me like something on the stick itself doesn't appear to be right. Everything works fine though, but it wont flash any other firmware. Have to use the webflasher.

darkxst, 2023-08-01:
Sorry you probably need to use the raw download link with silabs flasher
https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.3.1.0-zbdonglee-115200.gbl

supaeasy, 2023-08-01:
Perfect! This worked instantly. This should REALLY make it into Documentation. Almost pulled my hair out.
By the way: is there some sort of documentation about what differences there are in your firmware as opposed to the official one at https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router ?

darkxst, 2023-08-01:
Its pretty much a general rule when using download links from Github, not specific to silabs flasher!

The router firmware turns the ZBDongle-E into a Zigbee Router, that can be used to extend your Zigbee mesh (both range or capacity if you have lots of end-devices). It basically works as a Zigbee Repeater, that joins your network as a end device, and then devices further away can connect to that instead of the Coordinator. Most AC Power connected devices, like smart plugs, lights etc also act as Zigbee Router function. I do not have any equivalent builds to this.

supaeasy, 2023-08-01:
I am sorry. Of course I did not mean that one -  copied the wrong link from Sonoffs manual - but the differences in your firmware and the official coordinator Firmware. One thing I noticed is touchlink supoport. But what's more?
I meant this one
https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP

darkxst, 2023-08-02:
The main difference is that my builds are off the current 7.3.1 SDK vs Iteads official build that uses the older (now unmaintaned) 6.10.3 SDK. Silabs dont provide a nice list of differences, unless you go and read the release notes ;) but you can expect performance improvements, security and bug fixes, and probably some new bugs too!