darkxst/silabs-firmware-builder #14: Problem with flashing sonoff-e with firmware ezsp 7.1.3.0

mamrai1, 2023-08-11:
https://github.com/Koenkk/zigbee-herdsman/issues/319#issuecomment-1674295841

Also tried flashing nvm3 and then reflashing ezsp 7.1.3.0 on sonoff zbdongle e but i can't probe the device

darkxst, 2023-08-11:
Hi Mamrai1, 
Can you try again with the 7.3.1 firmware but add the `--verbose` flag to generate debug logs for universal-silabs-flasher. Do this for both the flash and probe. Once you have these attach the logs to this issue as files (they will be really long otherwise!)

mamrai1, 2023-08-11:
/
[probeafterreflashing7110.log](https://github.com/darkxst/silabs-firmware-builder/files/12322474/probeafterreflashing7110.log)
[probefail.log](https://github.com/darkxst/silabs-firmware-builder/files/12322476/probefail.log)
[flash.log](https://github.com/darkxst/silabs-firmware-builder/files/12322478/flash.log)
[probe7110.log](https://github.com/darkxst/silabs-firmware-builder/files/12322479/probe7110.log)

mamrai1, 2023-08-11:
I downloadwd the firmware using wget.

wastez, 2023-08-11:
As already mentioned before i had the same problem with 7.3.1 but i'm sure you still know it.

mamrai1, 2023-08-11:
> As already mentioned before i had the same problem with 7.3.1 but i'm sure you still know it.

Did you have the problem with 7.3.0 and 7.2.2?

darkxst, 2023-08-11:
> flash.log

This log is missing debug logs and is incomplete, can you try again.

> Did you experience the problem with 7.3.0 and 7.2.3?

No I have never experienced this problem with any of the builds on my repo.

It probably wont help, but have you tried with the web flasher? same issues?
https://darkxst.github.io/silabs-firmware-builder/

wastez, 2023-08-12:
As we already tried it with different flasher (elelabs and silabs) I can’t imagine that the webflasher will help. 

To be fair I have to say I don’t install other build from this repo.
I installed the latest versions from xsp1989 and I had other problems with z2m. 
But like here that the firmware seems to be destroyed after flashing i never had before.

darkxst, 2023-08-12:
> As we already tried it with different flasher (elelabs and silabs) I can’t imagine that the webflasher will help.

The webflasher uses universal-silabs-flasher under the hood, however it also runs completely isolated from the python environment on your machine, just in case something in there is causing the issues.

I have not been able to reproduce this issue myself, and I have flashed dozens of times the various ezsp zbdongle-e builds from 7.2.x and 7.3.x. So unless there is a different hardware revision of the zbdongle-e kicking around, its hard to see how the firmware is completely broken.

mamrai1, 2023-08-12:
I used this command on debian for flashing:
 universal-silabs-flasher --device /dev/ttyACM0 flash --firmware ncp-uart-hw-v7.3.1.0-zbdonglee-115200.gbl 2>&1 | tee flash.log

mamrai1, 2023-08-12:
tried the webflasher.. same results but better logs:)
[silabs_flasher (1).log](https://github.com/darkxst/silabs-firmware-builder/files/12326795/silabs_flasher.1.log)

darkxst, 2023-08-12:
you needed to add `--verbose`, but its ok the webflasher logs look fine.

Flashing seems to go fine, but then dropping connection (probably timeout) as soon as it tries to probe
```
2023-08-12 09:54:39 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.EZSP at 115200 baud
2023-08-12 09:54:39 emscripten bellows.ezsp[42] DEBUG Resetting EZSP
2023-08-12 09:54:39 emscripten bellows.uart[42] DEBUG Resetting ASH
2023-08-12 09:54:39 emscripten bellows.uart[42] DEBUG Sending: b'1ac038bc7e'
2023-08-12 09:54:44 emscripten bellows.uart[42] DEBUG Connection lost: None
2023-08-12 09:54:44 emscripten bellows.uart[42] DEBUG Closed serial connection
```
Does the same thing happen if you try and flash the multipan RCP firmware? can you probe after installing that one?

darkxst, 2023-08-12:
also one more test, with 7.3.1 if you connect to the dongle over serial are you getting the gecko bootloader menu or a lot of random characters? 

If your on linux just run
``` screen /dev/ttyACM0 115200 ```

mamrai1, 2023-08-12:
> also one more test, with 7.3.1 if you connect to the dongle over serial are you getting the gecko bootloader menu or a lot of random characters?
> 
> If your on linux just run `screen /dev/ttyACM0 115200`

nothing just a blank screen

mamrai1, 2023-08-12:
> you needed to add `--verbose`, but its ok the webflasher logs look fine.
> 
> Flashing seems to go fine, but then dropping connection (probably timeout) as soon as it tries to probe
> 
> ```
> 2023-08-12 09:54:39 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.EZSP at 115200 baud
> 2023-08-12 09:54:39 emscripten bellows.ezsp[42] DEBUG Resetting EZSP
> 2023-08-12 09:54:39 emscripten bellows.uart[42] DEBUG Resetting ASH
> 2023-08-12 09:54:39 emscripten bellows.uart[42] DEBUG Sending: b'1ac038bc7e'
> 2023-08-12 09:54:44 emscripten bellows.uart[42] DEBUG Connection lost: None
> 2023-08-12 09:54:44 emscripten bellows.uart[42] DEBUG Closed serial connection
> ```
> 
> Does the same thing happen if you try and flash the 4.3.1 multipan RCP firmware? Can you probe after installing that one?

No it works:

 universal-silabs-flasher --verbose --device /dev/ttyACM0 probe
2023-08-12 13:59:06 omv asyncio[15102] DEBUG Using selector: EpollSelector
2023-08-12 13:59:06 omv universal_silabs_flasher.flasher[15102] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-08-12 13:59:06 omv universal_silabs_flasher.common[15102] DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7f2813aa2a30>, Serial<id=0x7f2813aabdc0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2023-08-12 13:59:06 omv universal_silabs_flasher.common[15102] DEBUG Sending data b'3'
2023-08-12 13:59:08 omv universal_silabs_flasher.flasher[15102] INFO Probing ApplicationType.CPC at 460800 baud
2023-08-12 13:59:08 omv universal_silabs_flasher.common[15102] DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x7f2813aa2a30>, Serial<id=0x7f2813aabdc0, open=True>(port='/dev/ttyACM0', baudrate=460800, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2023-08-12 13:59:08 omv universal_silabs_flasher.cpc[15102] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-08-12 13:59:08 omv universal_silabs_flasher.common[15102] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-08-12 13:59:09 omv universal_silabs_flasher.cpc[15102] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-08-12 13:59:09 omv universal_silabs_flasher.cpc[15102] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-08-12 13:59:09 omv universal_silabs_flasher.common[15102] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-08-12 13:59:10 omv universal_silabs_flasher.cpc[15102] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-08-12 13:59:10 omv universal_silabs_flasher.cpc[15102] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-08-12 13:59:10 omv universal_silabs_flasher.common[15102] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-08-12 13:59:11 omv universal_silabs_flasher.cpc[15102] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-08-12 13:59:11 omv universal_silabs_flasher.cpc[15102] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-08-12 13:59:11 omv universal_silabs_flasher.common[15102] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-08-12 13:59:12 omv universal_silabs_flasher.cpc[15102] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (att [... cut ...]
2023-08-12 13:59:12 omv universal_silabs_flasher.cpc[15102] DEBUG Parsed frame UnnumberedFrameType.POLL_FINAL CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_IS: 6>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=bytearray(b'\x04\x00\x00\x00\x03\x00\x00\x00\x01\x00\x00\x00'))))
2023-08-12 13:59:12 omv universal_silabs_flasher.flasher[15102] INFO Detected ApplicationType.CPC, version '4.3.1' at 115200 baudrate (bootloader baudrate None)

mamrai1, 2023-08-12:
Probing rcp firmware works, screen show just a blank screen with that also but that is normal i think.. With ezsp firmware, no probe and screen also blank.

wastez, 2023-08-12:
> Probing rcp firmware works, screen show just a blank screen with that also but that is normal i think.. With ezsp firmware, no probe and screen also blank.

I also think that the blank screen on rcp is normal because it’s not driven directly by the serial device. 
So the problem seems to be the ncp 7.3.1 firmware…..

darkxst, 2023-08-12:
screen wont show anything useful in either ncp or rcp, only if the device is stuck in bootloader it will show a menu

mamrai1, 2023-08-12:
@darkxst have you ever flashed the nvm3 file on the sonoff you used to check the firmware? I did. Maybe something got erased which was not supposed to be?

darkxst, 2023-08-12:
> @darkxst have you ever flashed the nvm3

yes, I have used it occasionally when downgrading back to 6.10.3 factory firmware...

darkxst, 2023-08-13:
which firmware build are you using that does work?

mamrai1, 2023-08-13:
7.1.1.0 build 273

darkxst, 2023-08-13:
> 7.1.1.0 build 273

Yes, I've seen the version number in the logs, but whose build, which repo?

mamrai1, 2023-08-13:
Xsp1989

mamrai1, 2023-08-13:
Weird.. Attached 2 files of 7.1.4 from @xsp1989
. Old version works but not the new one.

darkxst, 2023-08-13:
You mean these? hw one is zb-gw04
https://github.com/xsp1989/zigbeeFirmware/commit/f813621c63519a289e6f315157e9fe2f8f8fbd96

wastez, 2023-08-13:
As far as I know the 7.1.1 was too from xsp1989 but he removed them. 

One of the last changed he made was to set the default transmit power to 20 dbm.

mamrai1, 2023-08-13:
> You mean these? hw one is for zb-gw04 [xsp1989/zigbeeFirmware@f813621](https://github.com/xsp1989/zigbeeFirmware/commit/f813621c63519a289e6f315157e9fe2f8f8fbd96)

Yes there is a a version with sw flowcontrol. As stated above the latest doesn't work. Attached zip file with both versions. 
[firmware714.zip](https://github.com/darkxst/silabs-firmware-builder/files/12329501/firmware714.zip)

wastez, 2023-08-13:
@darkxst 
So if you now flash your 7.3.1 it will work without problems?
Have somebody tried a other version of 7.3.1 (higher baudrate for example)

darkxst, 2023-08-13:
> So if you now flash your 7.3.1 it will work without problems?

yes, I flashed it the other day and it booted fine and worked in Z2M.

darkxst, 2023-08-13:
> Attached zip file with both versions.

The firmware hasnt changed on git. It was released Apr 27. so why are the two downloads different?

mamrai1, 2023-08-13:
It was removed and re released 27 Apr...

darkxst, 2023-08-13:
that 7.1.4.new flashes and probes ok on my dongle, but has other issues failing in Z2M

wastez, 2023-08-13:
> that 7.1.4.new flashes and probes ok on my dongle, but has other issues failing in Z2M

Thats exactly what i had.....

But not that they are not working at all like the 7.3.1.

darkxst, 2023-08-13:
> But not that they are not working at all like the 7.3.1.

I just flashed my 7.3.1 again (from a fresh download), after 7.1.4 failed and it was fine, again...

> By the way whats the difference in your firmware beetween normal and -none version?

normal has software (xon/xoff) flow control set same as itead builds (this only applies to NCP builds)
`-none` has flow control turned completely off

wastez, 2023-08-13:
Tried it again and the same happening, here the log:

[silabs-log.txt](https://github.com/darkxst/silabs-firmware-builder/files/12329632/silabs-log.txt)

wastez, 2023-08-13:
Here is the probing log:
[probe.txt](https://github.com/darkxst/silabs-firmware-builder/files/12329640/probe.txt)

wastez, 2023-08-13:
Flashed the old firmware and all is as it should......

mamrai1, 2023-08-13:
> Flashed the old firmware and all is as it should......

Which one?

wastez, 2023-08-13:
> > Flashed the old firmware and all is as it should......
> 
> Which one?

Again the 7.1.1.

mamrai1, 2023-08-13:
@wastez where did you buy your stick?

wastez, 2023-08-13:
@mamrai1 
I ordert it from amazon.
I dont´t know the serial, can't see it in z2m.
Have to search for it.

mamrai1, 2023-08-13:
Tim on your firmware power is hardcoded to 20db? 
If so maybe power issues with cable, or usb port or position of the stick?

wastez, 2023-08-13:
@darkxst 

Could it be that your stick has a other bootloader for example? (Sorry i don't know much about these things in ezsp sticks and how they work)

mamrai1, 2023-08-13:
> that 7.1.4.new flashes and probes ok on my dongle, but has other issues failing in Z2M

Sorry you are right. Flashes ok on mine too

wastez, 2023-08-13:
@darkxst 

I saw something on the end of the flash process with 7.3.1 the following was sent:
`2023-08-13 15:37:33 Vortex-i5.local universal_silabs_flasher.xmodemcrc[84506] DEBUG Sending data b'\x01\x1f\xe0sp_version": "7.3.1.0"\n}\n\xfc\x04\x04\xfc\x04\x00\x00\x00\xf0\xda\xb1\xdf\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xf3\x86' (attempt 0)`

And with 7.1.1 this was sent:
`2023-08-13 15:54:27 Vortex-i5.local universal_silabs_flasher.xmodemcrc[84903] DEBUG Sending data b'\x01\x1b\xe4\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfc\x04\x04\xfc\x04\x00\x00\x00\xdf\x95T~\xff\xff\xff\xff\xff\xff\xff\xffd\xd4' (attempt 0)`

Is this correct on 7.3.1? Shouldn't it be all hex?

darkxst, 2023-08-14:
> Is this correct on 7.3.1? Shouldn't it be all hex?

That is correct, its the metadata that is embedded on all my firmware builds
[image]

darkxst, 2023-08-14:
> Could it be that your stick has a other bootloader for example?

My stick also shows bootloader v1.12. The only thing that could really affect the bootloaded is was if its set to only accept signed images, but that is not the case on any -E dongles. Otherwise the RCP image would have failed also.

> my serial: 202208161613231

if that is a date code, the dongle I am currently testing is about 3 months newer than that.

> Tim on your firmware power is hardcoded to 20db? If so maybe power issues with cable, or usb port or position of the stick?

power is left at default, which I believe is 20dBm for EFR32 although I havent specifically checked. I wouldnt expect that to be causing power issues and failing boots

> Maybe i had the same problem from 7.2.2.0 to 7.2.3.0.

Did 7.2.2 work? it would be good to know the exact version that broke, then I can check release notes for SDK and see if anything relevant changed. However its also possible it was triggered by a tooling change in the silabs slc tools, which are not documented well.

darkxst, 2023-08-14:
Also can you try this firmware?
https://github.com/darkxst/silabs-test/raw/7.1/firmware_builds/zbdonglee/ncp-uart-hw-v7.1.5.0-zbdonglee-115200.gbl

mamrai1, 2023-08-14:
> Also can you try this firmware? https://github.com/darkxst/silabs-test/raw/7.1/firmware_builds/zbdonglee/ncp-uart-hw-v7.1.5.0-zbdonglee-115200.gbl
> 
> This is built with same configs as my 7.3.1

Flashing goes ok but fails to probe
[flash715.log](https://github.com/darkxst/silabs-firmware-builder/files/12331453/flash715.log)
[probe715.log](https://github.com/darkxst/silabs-firmware-builder/files/12331454/probe715.log)

mamrai1, 2023-08-14:
with 7.2.3 flashing and probe ok but z2m fails to start:
2023-08-14T04:36:53.002Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success
2023-08-14T04:36:53.002Z zigbee-herdsman:adapter:ezsp:driv Network ready
2023-08-14T04:36:53.003Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 0 e0798dfffe7d7366
2023-08-14T04:36:53.004Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Node Descriptor' for '0'
2023-08-14T04:36:53.005Z zigbee-herdsman:adapter:ezsp:driv ZDO Node_Desc_req params: {"dstaddr":0}
2023-08-14T04:36:53.007Z zigbee-herdsman:adapter:ezsp:driv EZSP nwk=0, IEEE=0xe0798dfffe7d7366
2023-08-14T04:36:53.007Z zigbee-herdsman:adapter:ezsp:ezsp ==> getKey: {"keyType":1}
2023-08-14T04:36:53.007Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"getKey","_id_":106,"_isRequest_":true,"keyType":1}
2023-08-14T04:36:53.008Z zigbee-herdsman:adapter:ezsp:uart --> DATA (5,6,0): 1d00016a0001
2023-08-14T04:36:53.008Z zigbee-herdsman:adapter:ezsp:uart --> [565f21a93e2a14d1337e]
2023-08-14T04:36:53.008Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
2023-08-14T04:36:53.008Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[224,121,141,255,254,125,115,102]}},"extendedTimeout":true}
2023-08-14T04:36:53.017Z zigbee-herdsman:adapter:ezsp:uart <-- [665fa1a93e2a15a85995b5f28fb69e60047c65ddf5535d7a9f766319f87d5e3fa7ebcdde09fc82392458ab8927e27e]
2023-08-14T04:36:53.018Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,6,0): 665fa1a93e2a15a85995b5f28fb69e60047c65ddf5535d7a9f766319f87e3fa7ebcdde09fc82392458ab8927e27e
2023-08-14T04:36:53.018Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2023-08-14T04:36:53.018Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2023-08-14T04:36:53.018Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1d80016a00001a0001ffd725e30c2998324276189d3af162b000900400000000000066737dfeff8d79e0
2023-08-14T04:36:53.019Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x6a: {"_cls_":"getKey","_id_":106,"_isRequest_":false,"status":0,"keyStruct":{"bitmask":26,"type":1,"key":{"contents":[255,215,37,227,12,41,152,50,66,118,24,157,58,241,98,176]},"outgoingFrameCounter":299008,"incomingFrameCounter":0,"sequenceNumber":0,"partnerEUI64":{"type":"Buffer","data":[224,121,141,255,254,125,115,102]}}}
2023-08-14T04:36:53.019Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success
2023-08-14T04:36:53.019Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[224,121,141,255,254,125,115,102]}},"extendedTimeout":true}
2023-08-14T04:36:53.019Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,7,0): 1e00017e0066737dfeff8d79e001
2023-08-14T04:36:53.020Z zigbee-herdsman:adapter:ezsp:uart --> [675c21a92a2a73c1246ab5a8d3b5937b9e7e]
2023-08-14T04:36:53.020Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2023-08-14T04:36:53.020Z zigbee-herdsman:adapter:ezsp:driv TRUST_CENTER_LINK_KEY: {"_cls_":"getKey","_id_":106,"_isRequest_":false,"status":0,"keyStruct":{"bitmask":26,"type":1,"key":{"contents":[255,215,37,227,12,41,152,50,66,118,24,157,58,241,98,176]},"outgoingFrameCounter":299008,"incomingFrameCounter":0,"sequenceNumber":0,"partnerEUI64":{"type":"Buffer","data":[224,121,141,255,254,125,115,102]}}}
2023-08-14T04:36:53.020Z zigbee-herdsman:adapter:ezsp:ezsp ==> getKey: {"keyType":3}
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:uart <-- [775ca1a92a2ace1e7e]
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,7,0): 775ca1a92a2ace1e7e
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1e80017e00
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x7e: {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":false}
2023-08-14T04:36:53.025Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2023-08-14T04:36:53.026Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"getKey","_id_":106,"_isRequest_":true,"keyType":3}
2023-08-14T04:36:53.026Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,0,0): 1f00016a0003
2023-08-14T04:36:53.026Z zigbee-herdsman:adapter:ezsp:uart --> [705d21a93e2a16b5e27e]
2023-08-14T04:36:53.026Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2023-08-14T04:36:53.026Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":0,"apsFrame":{"clusterId":2,"profileId":0,"sequence":2,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":256},"messageTag":3,"message":{"type":"Buffer","data":[2,0,0]}}
2023-08-14T04:36:53.035Z zigbee-herdsman:adapter:ezsp:uart <-- [005da1a93e2a15b15997bd506304bbafca27e0fc6856190a5ef962f96a7d5e3fa7ebcdde6f8fffc7dbd5d2692d8a7e]
2023-08-14T04:36:53.036Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,0,0): 005da1a93e2a15b15997bd506304bbafca27e0fc6856190a5ef962f96a7e3fa7ebcdde6f8fffc7dbd5d2692d8a7e
2023-08-14T04:36:53.036Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-08-14T04:36:53.036Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-08-14T04:36:53.036Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1f80016a0000030003f775c95129e65669c75785987e81a33f0170960000000000000000000000000000
2023-08-14T04:36:53.036Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x6a: {"_cls_":"getKey","_id_":106,"_isRequest_":false,"status":0,"keyStruct":{"bitmask":3,"type":3,"key":{"contents":[247,117,201,81,41,230,86,105, [... cut ...]
2023-08-14T04:36:53.046Z zigbee-herdsman:adapter:ezsp:debg Node descriptor request for '0' failed (Error: ZdoRequest error), retry
Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:573:19)
    at EZSPAdapter.nodeDescriptorInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:373:28)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:363:32)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

darkxst, 2023-08-14:
can you try 7.1.5 again (Same link). Rebuilt to drop token interface added to my configs around 7.3.

mamrai1, 2023-08-14:
> can you try 7.1.5 again (Same link). Rebuilt to drop token interface added to my configs around 7.3.

i was about to write that i suspect something with the token in the nvm

mamrai1, 2023-08-14:
> > can you try 7.1.5 again (Same link). Rebuilt to drop token interface added to my configs around 7.3.
> 
> i was about to write that i suspect something with the token in the nvm

no same results. probing fails

darkxst, 2023-08-14:
> i was about to write that i suspect something with the token in the nvm

had you previously overwritten IEEE address?

darkxst, 2023-08-14:
> with 7.2.20 flashing and probe ok. z2m works ok
> with 7.2.3 flashing and probe ok but z2m fails to start:

were these my builds or xsp1989's?

mamrai1, 2023-08-14:
> > i was about to write that i suspect something with the token in the nvm
> 
> had you previously overwritten IEEE address?

no only flashed nvm3 file after downgrading to 6.10 version.
backup/restore  with bellows works. In the backup file is noticed these lines:

            "ezsp": {
                "manufacturer": "",
                "board": "",
                "version": "7.1.1.0 build 273",
                "stack_version": 9,
                "can_burn_userdata_custom_eui64": true,
                "can_rewrite_custom_eui64": false
            }

mamrai1, 2023-08-14:
> > with 7.2.20 flashing and probe ok. z2m works ok
> > with 7.2.3 flashing and probe ok but z2m fails to start:
> 
> were these my builds or xsp1989's?

no xsp1989's

mamrai1, 2023-08-14:
no probing on both 7.2.2. and 7.2.3
[probe722tim.log](https://github.com/darkxst/silabs-firmware-builder/files/12331824/probe722tim.log)
[probe723tim.log](https://github.com/darkxst/silabs-firmware-builder/files/12331825/probe723tim.log)

mamrai1, 2023-08-14:
Fixed!!
need to flash nvm3_initfile.gbl after flashing your firmwares Tim!!

mamrai1, 2023-08-14:
> Rebuilt to drop token interface added to my configs around 7.3.

@darkxst  Can we also have the correct firmwares v.7.1.5. for zbdongle under your builds?

[23 more replies]

closed 2023-08-14 by mamrai1