darkxst/silabs-firmware-builder #15: Router firmware with latest ezsp

mamrai1, 2023-08-30:
Can you please provide us with a router firmware based on 7.3.1.?
I'm using also a 2nd zbdongle-e with 6.10.3 firmware but old devices like xiaomi and bosch motion sensors based on zigbee 1.2 won't connect through it

mamrai1, 2023-08-31:
Update: After multiple tries i have managed to pair them through the router (near the router)

mamrai1, 2023-09-15:
i dare to declare that using ezp 7.3.1 as coordinator and a 2nd zbdongle-e as router (house with 2 floors-coordinator on upper floor and router on ground floor) my network is extremely stable! At least is way more stable than before using cc2652p as coordinator or router!

supaeasy, 2023-09-15:
what router FW do you use?

mamrai1, 2023-09-15:
6.10.3 on 2nd zbdongle-e.. With cc2652p and latest firmware from koenkk after a double of days the router was not routing messages.. Something was crashing it

darkxst, 2023-10-11:
Building router firmware would take significant changes to the builder scripts. I am not sure there is much benefit over using the 6.10.3 router firmware from iTead, however if there are specific bugs that can be fixed I might look into it at a later date.

Hedda, 2023-10-12:
FYI, [xsp1989](https://github.com/xsp1989)'s Zigbee Router firmware is also old but works on most EFR32MG21 adapters -> https://github.com/xsp1989/zigbeeFirmware/ -> https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

darkxst, 2023-10-13:
iTead's `6.10.3` router firmware will also work on all the same adapters as xsp1989's builds

mamrai1, 2023-10-14:
The trooth is that after 3 months the router with 6.10.3 is very stable..(way more stable than cc2652p). I only have one xiaomi door sensor that refuses to connect through it...Other than that zero problems!

Hedda, 2024-02-14:
> I only have one xiaomi door sensor that refuses to connect through it.

FYI, check out these tips -> https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

Summery, many Aqara and Xiaomi have buggy firmware that makes them stick to the Zigbee Router they first connected to when initially paired/joined and therefore have to be factory reset and re-paired/re-joined manually to connect to other Zigbee Router.

Note that some Aqara and Xiaomi Zigbee devices are not fully compatible with all brands of Zigbee router devices. Xiaomi/Aqara devices are for example known not to work with Zigbee router devices from Centralite, General Electrics, Iris, Ledvance/OSRAM, LIGHTIFY/Sylvania, Orvibo, PEQ, Securifi, and SmartThings/Samsung. Better results can usually be achieved by using mains-powered Zigbee Router devices from IKEA and Nue/3A Home or dedicated DIY routing devices based on Texas Instruments CC253x/CC26x2 and XBee Series 2/3 Zigbee radios.

Also be patient as the pairing of some Zigbee devices may require multiple attempts and you may sometimes need to try again and again. Again, some devices, like example those from Xiaomi/Aqara, are known to not be 100% compliant with the standard Zigbee specifications and may therefore require many paring attempts over 10-20 minutes or longer.

chris-1243, 2024-03-13:
It might not be the right place to ask. Anyone knows if a router firmware is available for the skyconnect? 

I already use a Dongle-E as a router without any issues. I need a second router but the size of the Dongle-E doesn't suit my needs

darkxst, 2024-03-13:
> router firmware is available for the skyconnect?

Router firmware would be a problem for skyconnect as it doesnt have a boot button. Once installed you wouldnt be able to flash ever again.

> (Or any usb small dongle which accepts firmware router😉)

You could try SMLIGHT SLZB-07, I believe that has a router firmware available on the SMLIGHT web flasher.

chris-1243, 2024-03-13:
> Router firmware would be a problem for skyconnect as it doesnt have a boot button. Once installed you wouldnt be able to flash ever again.

Oups, I was not aware of this detail as I do not have one... It fully makes sense not to find any router firmware. Thanks

I will dig a bit internet with my google friend and see what I may find

Hedda, 2024-03-15:
> Router firmware would be a problem for skyconnect as it doesnt have a boot button. Once installed you wouldnt be able to flash ever again.

FYI, if you brick a such adapter where the board has not bootloader button then usually have to buy a debugger (debug probe) programmer to manually flash a new fimrware image via SWD or JTAG, and that needs to be compatible with new EFR32 chips, like example:

https://www.silabs.com/development-tools/wireless/simplicity-link-debugger (other other newish [Segger J-Link debug probes](https://www.segger.com/products/debug-probes/j-link/)).

Alternativly, if there is not a BTL button on the board then could try to figure out what pads/pins could manually short circuit in order to enter into bootloader mode? Similar what can to do on old the HUSBZB-1 adapter -> https://github.com/walthowd/husbzb-firmware?tab=readme-ov-file#husbzb-1-firmware-recovery

Hedda, 2024-03-15:
hmm, maybe I am wrong about this for the SkyConnect as believe that Nabu Casa has for example implemented a manufacturer specific command in their official OpenThread RCP firmware which has "back door" command for reset into bootloader that makes it possible to always access bootloader mode as long as use Nabu Casa's universal-silabs-flasher for flashing:

https://github.com/NabuCasa/silabs-firmware/issues/6

https://github.com/NabuCasa/universal-silabs-flasher/pull/13

tl-sl, 2024-03-15:
> FYI, if you brick a such adapter where the board has not bootloader button then usually have to buy a debugger

Yes you can flash over SWD, JTAG or cJTAG depending what pins are exposed on the pcb. However this is probably beyond the average user that just wants router firmware!

> Nabu Casa has for example implemented a manufacturer specific command in their official OpenThread RCP firmware 

The reset to bootloader support for openthread is upstream now since about 2.3.x, so no longer specific NabuCasa builds. However this has nothing to do with bootloader reset from router firmware.

Hedda, 2024-03-18:
Any chance someone here is interested in putting together a generic always-on Zigbee Router device application for/with Silicon Labs Gecko SDK (GSDK) that could be reused in silabs-firmware-builder for building matching Zigbee Router firmware images to allow users to convert their Zigbee Coorindaotor USB dongles into dedicated Zigbee signal repeater / Zigbee range extender?

https://github.com/SiliconLabs/gecko_sdk

For reference, see Zigbee application examples:

https://github.com/SiliconLabs/zigbee_applications 

Since not all dongles have buttons I guess it would need be configured to automatically always try to join a Zigbee network for 60-seconds when power-on if it is not already joined to a Zigbee network + might also need way to reset devices with no button?

For example check out their "Zigbee Human Detection" example which function as a always-on Zigbee Router device with a join button:

https://github.com/SiliconLabs/zigbee_applications/tree/master/zigbee_human_detection

Also check out these directly or indirectly related discussions about Zigbee Router device firmware for EFR32 based devices:

https://github.com/grobasoz/zigbee-firmware/issues/24

https://github.com/grobasoz/zigbee-firmware/issues/33

https://github.com/zigpy/zigpy/discussions/736

PS: Perhaps ask @xsp1989 if he would be willing to share the source and/or config for the Zigbee Router application he made?

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

darkxst, 2024-03-18:
> Since not all dongles have buttons

This is a problem, there is no way out of router firmware once flashed, as need hardware bootloader reset to re-flash normal firmware.

Hedda, 2024-03-18:
> This is a problem, there is no way out of router firmware once flashed, as need hardware bootloader reset to re-flash normal firmware.

So maybe not build Zigbee Router firmware images for dongles without buttons but instead only build for those with buttons?

Hedda, 2024-03-18:
> Any chance someone here is interested in putting together a generic always-on Zigbee Router device application for/with Silicon Labs Gecko SDK (GSDK) that could be reused in silabs-firmware-builder for building matching Zigbee Router firmware images to allow users to convert their Zigbee Coorindaotor USB dongles into dedicated Zigbee signal repeater / Zigbee range extender?
> 
> https://github.com/SiliconLabs/gecko_sdk
> 
> For reference, see Zigbee application examples:
> 
> https://github.com/SiliconLabs/zigbee_applications
> 
> Since not all dongles have buttons I guess it would need be configured to automatically always try to join a Zigbee network for 60-seconds when power-on if it is not already joined to a Zigbee network + might also need way to reset devices with no button?
> 
> For example check out their "Zigbee Human Detection" example (functions as a always-on Zigbee Router device with join button):
> 
> https://github.com/SiliconLabs/zigbee_applications/tree/master/zigbee_human_detection

By the way, if someone does make a such generic always-on Zigbee Router device application then it would be awesome if also make the the transmit power setting configurable over Zigbee in the same way that Koenkk's Z-Stack Router firmware is, see:

https://github.com/Koenkk/Z-Stack-firmware/issues/341

That is, set the default transmit power to 9dBm but expose an optional attribute to change TX power up to 20 dBm via Zigbee:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/CHANGELOG.md

https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/firmware.patch

[image]

xsp1989, 2024-03-19:
> > Since not all dongles have buttons
> 
> This is a problem, there is no way out of router firmware once flashed, as need hardware bootloader reset to re-flash normal firmware.

It is said that there is an API reset_resetWithReason(BOOTLOADER_RESET_REASON_BOOTLOAD) in the 6. I have not conducted relevant tests on the GSDK of 7.X, but I believe it is available. Please note that this API seems to reference the relevant code in the bootloader through a struct. If there are different versions of the bootloader, incompatibility may occur. The simplest case is to use our ZB-GW04, because it provides a hardware button that can force the bootloader to stay at any time.

Hedda, 2024-03-19:
> The simplest case is to use our ZB-GW04, because it provides a hardware button that can force the bootloader to stay at any time.

@xsp1989 any chance you provide application source for building Zigbee Router device firmware that can be reused on USB dongle devices like ZB-GW04 and similar? -> https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

xsp1989, 2024-03-19:
@Hedda  I have not done any development other than NCP/RCP on the latest SDK. If you use the 6.x version of the SDK to develop routing firmware, it is actually quite simple. You only need to select the device type and generate the source code to compile it. The routing firmware can be It is standard (Range Extender deviceid:0x0008), or it can be a lamp or socket.
Now the 6.10.12 version of the SDK can be downloaded directly in SS without additional registration.

Hedda, 2024-03-19:
> If you use the 6.x version of the SDK to develop routing firmware, it is actually quite simple. You only need to select the device type and generate the source code to compile it. The routing firmware can be It is standard (Range Extender deviceid:0x0008), or it can be a lamp or socket.

So you did not have to add and the press button to join the network + LED status or those extra serial tunneling features either?

## Join the network
1. Press BOOT for 5 seconds until the LED flashes.
2. Gateway allows access to the network.
3. If the LED is always on, then join the network successfully.

## LED status
LED status | Network Status
-----------|---------------
LED off | NO Netwrok
LED blink | steering network
LED on | Joined to the network 

## Extra features
* If you send data through the serial port, the router will forward the data to the gateway.
* If data is sent to the router through the coordinator, the serial port will receive the data.

## Cluster Identifiers
Identifier | Name 
-----------|---
0x0704 | Tunneling (Smart Energy)

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle/README.md#join-the-network

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle/README.md#led-status

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle/README.md#extra-features

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle/README.md#cluster-identifiers

xsp1989, 2024-03-19:
What I mean is that if you make a routing firmware that is exactly the same as the firmware I provided, it may require additional development, but if you only need a routing function, it is actually very simple. You don't even need buttons and LEDs, you only need to set the CLI command line To the UART, you can connect the computer and control it to join the network through CLI commands.

Because my firmware is not specifically used for routing. We have other business needs and just developed this firmware.

Hedda, 2024-04-04:
> You could try SMLIGHT SLZB-07, I believe that has a router firmware available on the SMLIGHT web flasher.

Noticed according to the SMLIGHT SLZB-07 used manual it too does not have a boot buttom so require short circuit workaround:

https://smlight.tech/manual-slzb-07/

## IMPORTANT NOTE FOR THE CURRENT VERSION OF ROUTER FIRMWARE

If you flashed your coordinator to the router and want to flash it back to the coordinator, you have to push the device to flashing mode manually (is not required for other flashing options, only Router -> Coordinator)

Router → Coordinator: Need to push the coordinator physically to flashing mode.
– Unplug your coordinator
– Open enclosure
– Connect pad FLSH with the metal part of the antenna (you can use tweezers or scissors
– Plug the coordinator to your PC (keep FLSH connected to the antenna)
– Run flashing.

[image]

tomtzeng, 2024-05-07:
Is it possible to provide new router firmware based on 7.3.1. for the Sonoff Dongle-E?

supaeasy, 2024-05-07:
Better 7.4.2 since it is ember compatible.

Hedda, 2024-05-07:
> Better 7.4.2 since it is ember compatible.

That is not relevant here. Zigbee Router device compatible is not the same as Zigbee Coordinator compatibility.

Zigbee Router devices are backwards and forward compatible, so they just need to be built for that chip and board.

That is, Zigbee Router is joined/paired like any Zigbee device so it can have any Zigbee stack and firmware version.

darkxst, 2024-05-08:
> Noticed according to the SMLIGHT SLZB-07

New hardware revision will have `--bootloader-reset slzb07` in universal-silabs-flasher

> hmm, maybe I am wrong about this for the SkyConnect as believe that Nabu Casa has for example implemented a manufacturer specific command in their official OpenThread RCP firmware which has "back [...]

The bootloader reset in Thread firmware is now upstream, so any new enough thread firmware will support this. I am not aware of any client side implementations though apart from universal-silabs-flasher and web flasher.

darkxst, 2024-05-08:
I will look at adding router builds after migrating to the new and improved builder puddly has been working on...

Hedda, 2025-10-18:
> We do not provide firmware for the Sonoff Dongle E

@xsp1989 note that above question is posed in darkxst’s repo and that does provide firmware for Sonoff Zigbee Dongle-E:

- https://github.com/darkxst/silabs-firmware-builder