darkxst/silabs-firmware-builder #16: Access to Release notes or more what the new firmware provided

pipiche38, 2023-09-30:
It would be great to get access to a kind of Release notes, where we can see what the firmware version fix/bring

darkxst, 2023-10-11:
Silabs publish full release notes with each release. For example here are the notes for the latest release (which I will be uploading shortly)

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.2

For the most part we dont make many changes on top of those.

panzer-arc, 2024-01-26:
@darkxst 
I'm confused how to tell if there's an update available. I see [here ](https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee)there are two newer builds for the Dongle-E. The flasher selected 7.3.1 for me. There also appear to be newer builds [here](https://github.com/SiliconLabs/gecko_sdk/releases/):

darkxst, 2024-01-26:
> @darkxst I'm confused how to tell if there's an update available.

The web flasher will always flash the currently recommended build. These get more testing before they are updated in the web flasher, compared to the automated builds in the repo.

Current status is that the last two Gecko releases have been very problematic (4.3.2 and 4.4.0). Mainly this affects the RCP multipan builds, but I am keeping the webflasher  with all builds on 4.3.1 for now.

If you are only using NCP build feel free to try 7.4.0 and let us know any feedback, if it causes issues you may need to downgrade to 7.3.1.

supaeasy, 2024-03-07:
As a side note: I was having MASSIVE problems since forever with Aqara Devices (especially Water Leak and Motion Sensors). Super troublesome pairing, regular disconnections after 3-7 Days with all firmwares including 7.3.1. This is completely gone in 7.4.0/1. Great work, I love it. Thank you!