darkxst/silabs-firmware-builder #17: hi

JacekKac, 2023-10-04:
i have a sonoff e-dongle flashed with router firmware (official from sonoff) and i want to flash it back to coordinator but web flasher fails: 

The running firmware could not be detected.

Make sure the USB port works and if you are using a USB extension cable, make sure the cable can transfer data. Unplug the Sonoff ZBDongle-E and plug it back in to reset and try again.

[silabs_flasher (2).log](https://github.com/darkxst/silabs-firmware-builder/files/12810015/silabs_flasher.2.log)

JacekKac, 2023-10-04:
when i open a serial i see it running
 start nwk join EVENT,sta = 0x00
**stack ver. [6.10.3 GA build 297]**
Z3RouterUSBDonlge>NWK Steering scan complete. Beacons heard: 0

darkxst, 2023-10-11:
@JacekKac Once you flash with Router firmware, software reset doesnt work. You can force reset with `universal-silabs-flasher` using the following command

`universal-silabs-flasher --device /dev/ttyACM0 flash --firmware <file> --sonoff-reset`

closed 2023-12-30 by darkxst