darkxst/silabs-firmware-builder #21: Silabs Firmware Flasher

smart-ctrl, 2023-10-22:
Why is Multi-PAN (RCP) 4.3.2 firmware not detected for ZBDongle-E, Multi-PAN (RCP) 4.3.1 is still the latest?
https://darkxst.github.io/silabs-firmware-builder/

smart-ctrl, 2023-10-24:
Never mind, I noticed 4.3.2 is extreamely unstable, maybe that's why?

darkxst, 2023-10-29:
I havent updated the webflasher yet, since havent had a chance to complete testing. I wouldnt expect much has changed from 4.3.1 yet though.

kalpik, 2023-11-16:
Hello!

Any updates on this? Is it safe to flash 4.2.1?

Thanks

darkxst, 2023-11-17:
4.3.2 is safe to flash, I just havent updated the web flasher yet. If that is is what you meant?
4.2.1 is obsolete and no longer works with multiprotocol addon.

kalpik, 2023-11-17:
Apologies. Yes, I meant the 4.3.2 update. I was skeptical since the web flasher wasn't updated yet.

smart-ctrl, 2023-11-17:
4.3.2 isn't stable, It only works for awhile. I had to downgrade to 4.3.1.
I tried to flash 4.3.2 two times, but with the same result. It works fine for some minutes/hours but will eventually stop working.
Something must be very different compared to 4.3.1?

darkxst, 2023-11-18:
@smart-ctrl Can you provide logs for this? Does the multiprotocol addon stop responding or Zigbee side ZHA/Z2M etc?

darkxst, 2023-11-18:
Also according to the release notes, no major changes in 4.3.2
https://www.silabs.com/documents/public/release-notes/gecko-platform-release-notes-4.3.2.0.pdf

smart-ctrl, 2023-11-30:
OK, then I will wait for 4.3.3 and try.

closed 2026-02-02 by github-actions[bot]