darkxst/silabs-firmware-builder #22: [REQUEST] Support for the EFR32MG1B232 chips found in TuYa gateways.

bfayers, 2023-10-28:
The TuYa TYGWZ-01, which were at some point sold in Lidl and Aldi and widely modified for use with ZHA/Z2M with these initial instructions: https://zigbee.blakadder.com/Tuya_TYGWZ-01.html

Currently the firmware used by people for these devices can be found here: https://github.com/grobasoz/zigbee-firmware/blob/master/EFR32%20Series%201/EFR32MG1B-256k/NCP/NCP_UHW_MG1B232_678_PA0-PA1-PB11_PA5-PA4.gbl

Obviously it would be ideal to get NCP, RCP and OT firmware but if this isn't possible NCP & OT firmwares would be very useful.

darkxst, 2023-10-29:
I don't think Series 1 chips support NCP 7.x. They should support RCP, but I am not aware of anyone who has tested RCP via Ethernet gateway, it may not work well. Not sure about OT support for these either.

Mattwestb might know more about support for these.

bfayers, 2023-11-06:
@mattwestb Any ideas for this?

Another options for firmware on these chips that could be useful for some is a router mode firmware for NCP.

MattWestb, 2023-11-07:
I have cooked one EZSP 6.10.8.0 and also have one old RCP that is not working with the current addon https://github.com/MattWestb/EFR32-FW/tree/main/tuya_ZBGW.
For testing is recommended having one SWD probe is the firmware is caching and and cant getting in the bootlader mode. The RCP is untested but i think the EZSP is partly tested of some devs.
If having time we can doing one race and trying getting both working on the chip then my Billy that is having MG1P chip is working great and it shall being possible getting the same with  the MG1B chip if all is working well bit need testing for getting all right.

RCP over Eth was implanted and was working with ESPHome but was broken one time in the addon and shall being fixed but then ESPHome was braking the serial server so i dont knowing the status if its working well or not at this time (i using my IKEA Billy modules with CH340 USB-TTL adapters or direct on UART on  the SOC).

closed 2026-01-31 by github-actions[bot]