darkxst/silabs-firmware-builder #24: ZB-GW04

RichardPar, 2023-11-10:
Hi There,

I have a generic ZB-GW02 v1.2 dongle - flashing the firmware for this device results in 'Secondary is Unresonsive' errors. I go back to my own generic software and it runs. 

Is there perhaps more varieties ?

Thanks
Richard

darkxst, 2023-11-10:
Which firmware are you using and how did you install it? You probably need to make sure you have the correct baudrate, which should be `230400` for the RCP or Openthread firmwares.

RichardPar, 2023-11-12:
> Which firmware are you using and how did you install it? You probably need to make sure you have the correct baudrate, which should be `230400` for the RCP or Openthread firmwares.

Sorry,  Ignore me, its was a sheer moment of stupidity.. I had the baud rate too high (the HA Sky Connect pages)

closed 2023-11-12 by RichardPar