darkxst/silabs-firmware-builder #25: Is it expected that zigpy not work after flash to MultiPAN firmware?

MostHated, 2023-11-14:
Hey there, I flashed my Sonoff Dongle-E with MultiPAN RCP firmware, and later discovered the existence of zigpy. My current setup uses a TI CC2652R devboard, and I used zigpy to backup it's setup and I wanted to test and see if it was possible to use it to restore back to the new device, but I have not been able to get it to connect properly in any way, just trying to confirm if it see's it. I attempted to use the updated baudrate listed on the firmware update page of 460800, etc. I figured it was a good idea to check if this is expected behavior, and whether it would/should be working in the first place with this firmware. 

Unfortunately, I didn't know about zigpy before I flashed the firmware, so I can't say whether it worked or not prior to now, but the flash seemed to work fine and I have no reason to believe that there is an issue, I was just hoping to test out if I could restore to it or not. 
If it is known that zigpy would no longer work with this firmware, that is perfectly fine, I just am hoping to know before I waste any more time on trying to get it to work.

MostHated, 2023-11-14:
Side note, I went back to the 115200 firmware, and then zigpy worked straight away, so I would guess it must be that firmware then.

darkxst, 2023-11-14:
When using multipan firmware with multiprotocol, the zigbee database is moved off the dongle and into zigbeed on the host. So for zigpy to restore a backup it would need to connect the network address and not directly to the serial port of the dongle. I have never actually tried this directly, but guess is must be possible as ZHA can migrate to multiprotocol setup.

MostHated, 2023-11-15:
Ah, I see. 

To see if a restore would work in general from one device to another, I flashed back to the standard firmware, then tried to do a restore, but got two errors of:

```log
bellows.zigbee.application  Couldn't add Key(...)  key: EmberStatus.TABLE_FULL
```
So it might be an issue either way. I guess I never really looked at what the available space of the CC2652R Devboard was compared to the EFR32. Though, I use zigbee2mqtt, so not sure how that plays into all of this. I would imagine it's probably better to just start over and re-pair all the devices in the first place.

Either way, I appreciate the info. Looks like we are all good here. :+1:

darkxst, 2023-11-15:
ZHA uses zigpy under the hood, so it should work with the right commands. I dont think space should be an issue. You will probably find more information on the community forums. There are users that have used zigpy to restore backups for Z2M.

Hedda, 2023-11-23:
If using MultiPAN RCP firmware then you need to install the Silicon Labs Multiprotocol add-on as it acts as EZSP daemon: 

https://github.com/home-assistant/addons/tree/master/silabs-multiprotocol

https://skyconnect.home-assistant.io/procedures/enable-multiprotocol/

https://github.com/zigpy/zigpy/discussions/894