darkxst/silabs-firmware-builder #27: Cannot flash my Sonoff ZBDongle-E with web flasher

GuiPoM, 2023-11-24:
Hello,

I hope this is the right project to report such an issue. I am trying to flash a newly acquired Sonoff ZBDongle-E. Never used, just coming from Amazon.
I tried to flash it from 2 distinct computers, both running Windows 11, with latest versions of Chrome or Edge. Both are failing when flashing. Dongle is plugged on an USB port on the device, not cable, no USB hub.

I am using url : https://darkxst.github.io/silabs-firmware-builder/

There I select ZBDongle-E and Select. Browser then asks me to select a device, I am selecting the one that is listed:
[image]

COM4 on one computer, COM5 here.

Then connecting dialog appears, start progressing, quickly stops and I get the following issue:
[image]

Debug logs:

```
2023-11-24 11:52:25 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-24 11:52:25 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-24 11:52:25 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1412e20>
2023-11-24 11:52:25 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2023-11-24 11:52:27 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-24 11:52:27 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
2023-11-24 11:52:27 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x13c9668>
2023-11-24 11:52:27 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:27 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:28 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-11-24 11:52:28 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:28 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:30 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-11-24 11:52:32 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:32 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:34 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-11-24 11:52:36 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:36 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:38 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-11-24 11:52:39 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 115200 baud
2023-11-24 11:52:39 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-24 11:52:39 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x12754c8>
2023-11-24 11:52:39 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:39 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:42 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-11-24 11:52:44 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:44 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:47 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-11-24 11:52:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:49 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 11:52:51 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-11-24 11:52:51 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 11:52:51 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
[... cut ...] PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-11-24 11:52:57 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.EZSP at 115200 baud
```

I do not know what to do next and how to debug it, I first was thinking that the problem was coming from my PC, but having the same issue on two PCs is more an issue or on the Dongle itself, or maybe this web flasher.

Could you give me any advice ?
Thanks !

darkxst, 2023-11-24:
```
2023-11-24 11:52:57 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.EZSP at 115200 baud
```
Is this log truncated? this stops right where I would expect it connect to factory firmware!

GuiPoM, 2023-11-24:
No, I don't think so, this is what I get when I click on the button to get the logs.

darkxst, 2023-11-24:
Everything up to that point looks fine, CPC will fail as there is not multipan firmware installed.

You could try with the command line universal-silabs-flasher see if it provides more useful logs. I dont know if windows has python these days or not?

https://github.com/NabuCasa/universal-silabs-flasher

GuiPoM, 2023-11-24:
Sure, I will try it today or tomorrow. 
My final goal is to use the web flasher as I initially wanted to create some content about quickly setting up a thread network, so the python way is a lot more complex.
But for my own use and to figure out what the error could be, of course this could be an option !

darkxst, 2023-11-24:
the web flasher should work and its mainly been tested on the ZBDongle-E. I dont use windows myself, but other users have used it with windows.

https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware#web-flasher

oli737, 2023-11-24:
Same "connection failed" problem for me too  when using the webflasher.
Couple weeks before it did work without any problem.

marcobag, 2023-11-24:
Same for me, i've flashed using the SAME browser/machine on 22 November with success, not yesterday and today.
I'm trying to flash different dongles but always ZBDongle-E i'm not able to try re-flash the first dongle to see if works but looks strange to me. 

`2023-11-24 15:40:24 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-24 15:40:24 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-24 15:40:24 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x13572e8>
2023-11-24 15:40:24 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x1a\xc1\x02\x02\x9b{~'
2023-11-24 15:40:24 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x1a\xc1\x02\x02\x9b{~')
2023-11-24 15:40:24 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2023-11-24 15:40:26 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-24 15:40:26 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
2023-11-24 15:40:26 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x117d1d8>
2023-11-24 15:40:26 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 15:40:26 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 15:40:27 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-11-24 15:40:27 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 15:40:27 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 15:40:28 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-11-24 15:40:28 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 15:40:28 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 15:40:29 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-11-24 15:40:29 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-24 15:40:29 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 15:40:30 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-11-24 15:40:30 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 115200 baud
2023-11-24 15:40:30 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-24 15:40:30 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x12c40d0>
2023-11-24 15:40:30 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_ [... cut ...]
2023-11-24 15:40:38 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-24 15:40:39 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-11-24 15:40:39 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.EZSP at 115200 baud`

marcobag, 2023-11-24:
As update, i've flashed using python and now the device is recognized and can be flashed by the web browser.
 i've used:
git clone https://github.com/Elelabs/elelabs-zigbee-ezsp-utility.git
cd elelabs-zigbee-ezsp-utility
python3 -m pip install -r requirements.txt

downloaded the firmware in the same folder

python3 Elelabs_EzspFwUtility.py restart -m btl -p /dev/tty.usbserial-202310081407351 (change with your device port)
python3 Elelabs_EzspFwUtility.py flash -p /dev/tty.usbserial-202310081407351 -f ./rcp-uart-802154-v4.3.1-zbdonglee-460800.gbl

done!

GuiPoM, 2023-11-25:
> Everything up to that point looks fine, CPC will fail as there is not multipan firmware installed.
>
> You could try with the command line universal-silabs-flasher see if it provides more useful logs. [...]

I managed to download and build it. 17Gb of C++ libs :| (web flasher is sooooo convenient !)
But I do not know what to do with it. If I try to flash it, then it might work but we never will have that issue.

I tried probe. Is it what I should have done? 

```
C:\Users\guill>C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\Scripts\universal-silabs-flasher.exe --device COM4 probe
2023-11-25 12:34:07 PC universal_silabs_flasher.flasher[19824] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-25 12:34:09 PC universal_silabs_flasher.flasher[19824] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-25 12:34:14 PC universal_silabs_flasher.flasher[19824] INFO Probing ApplicationType.CPC at 115200 baud
2023-11-25 12:34:18 PC universal_silabs_flasher.flasher[19824] INFO Probing ApplicationType.CPC at 230400 baud
2023-11-25 12:34:22 PC universal_silabs_flasher.flasher[19824] INFO Probing ApplicationType.EZSP at 115200 baud
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\Scripts\universal-silabs-flasher.exe\__main__.py", line 7, in <module>
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\universal_silabs_flasher\flash.py", line 36, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.240.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.240.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.240.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 664, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\universal_silabs_flasher\flash.py", line 230, in probe
    await flasher.probe_app_type()
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\universal_silabs_flasher\flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\universal_silabs_flasher\flasher.py", line 166, in probe_ezsp
    async with self._connect_ezsp(baudrate) as ezsp:
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.240.0_x64__qbz5n2kfra8p0\Lib\contextlib.py", line 204, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\guill\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\universal_silabs_flasher\emberznet.py", line 18, in connect_ezsp
    config.CONF_DEVICE_BAUDRATE: baudrate,
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'bellows.config' has no attribute 'CONF_DEVICE_BAUDRATE'. Did you mean: 'CONF_DEVICE_PATH'?

```

darkxst, 2023-11-25:
thanks, that seems that universal-silabs-flasher is broken! Can you file a bug over [there](https://github.com/NabuCasa/universal-silabs-flasher)  for this error.

I suspect is caused by this update:
https://github.com/zigpy/bellows/releases/tag/0.37.0

BUT i wouldnt have thought that would flow through to the web flasher automatically yet.  I will check it out.

wisewise3, 2023-11-25:
Hello, i'm a newbie. I try to follow many video to flash Sonoff Dongle Usb Plus E detected as "USB-Enhanced-SERIAL-CH9102(COM5) But when i click on "Flash" i selection the Com 5 then it say me "The running firmware could not be detected.

Make sure the USB port works and if you are using a USB extension cable, make sure the cable can transfer data. Unplug the Sonoff ZBDongle-E and plug it back in to reset and try again."...
Here is my log if it could me help. Sorry but i don't understand anything with Python, etc :-(
`2023-11-25 13:06:10 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-25 13:06:10 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-25 13:06:10 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1429b38>
2023-11-25 13:06:10 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\xc1'
2023-11-25 13:06:10 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\xc1')
2023-11-25 13:06:10 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2023-11-25 13:06:12 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-25 13:06:12 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
2023-11-25 13:06:12 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1271a88>
2023-11-25 13:06:12 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-25 13:06:12 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-25 13:06:13 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-11-25 13:06:13 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-25 13:06:13 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-25 13:06:14 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-11-25 13:06:14 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-25 13:06:14 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-25 13:06:15 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-11-25 13:06:15 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-11-25 13:06:15 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-25 13:06:16 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-11-25 13:06:16 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 115200 baud
2023-11-25 13:06:16 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-25 13:06:17 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x14d9158>
[... cut ...]
2023-11-25 13:06:25 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-11-25 13:06:26 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-11-25 13:06:26 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.EZSP at 115200 baud
`

darkxst, 2023-11-25:
Universal-silabs-flasher is broken at the moment so this needs to be fixed there.

Once this bug is resolved, the web flasher should start working again!
https://github.com/NabuCasa/universal-silabs-flasher/issues/46

wisewise3, 2023-11-25:
Thanks for your feedback. I choose my best moment for leaning Home assistant/Zigbee etc lol.

wisewise3, 2023-11-25:
MANY Thanks. It's ok now :-)

GuiPoM, 2023-11-25:
> Universal-silabs-flasher is broken at the moment so this needs to be fixed there.
> 
> Once this bug is resolved, the web flasher should start working again! [NabuCasa/universal-silabs-flasher#46](https://github.com/NabuCasa/universal-silabs-flasher/issues/46)

It is also fixed on my side ! Thanks !

closed 2023-11-25 by GuiPoM