darkxst/silabs-firmware-builder #28: ZBDongle E - Port detected but no connection

Jumperm, 2023-11-28:
Hi, I try to update my ZBDongle-E from Stock to the newest NCP-version. When I plug in the stick, the correct CH9102 driver is loaded. The stick is detected in Chrome/Edge, but no connection is established. The CP2102 driver don't work with the stick. Newest CH9102 driver also don't work. Same at MacOS/Chrome. Any solution?

BR Tobias

darkxst, 2023-11-28:
when did you try this? there were issues with the web flasher for the last week or so, but should be fixed now (as of Sunday).

Jumperm, 2023-11-29:
Hi, i tryed it yesterday.
Do i have a better chance when set the ZBDonglee in bootmode?!

Jumperm, 2023-12-01:
I try to use the universal-silabs-flasher. Maybe the error help you to identify the problem.

```
C:\Users\Tobi\Downloads>universal-silabs-flasher --device COM3 flash --firmware ncp-uart-hw-v7.3.2.0-zbdonglee-115200.gbl
Error: 'ncp-uart-hw-v7.3.2.0-zbdonglee-115200.gbl' does not appear to be a valid GBL image: ValidationError('Image is truncated: tag value is cut off')
```

darkxst, 2023-12-01:
@Jumperm check that you correctly downloaded the raw firmware file (and not just saved the html page!)

Jumperm, 2023-12-01:
> @Jumperm check that you correctly downloaded the raw firmware file (and not just saved the html page!)
> 
> https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.2.3.0-zbdonglee-115200.gbl

Big thanks! With the korrekt raw file it work!
I just thought i have a bad ZBDongle-E.
But what was wrong while using your web tool?!

darkxst, 2023-12-01:
> But what was wrong while using your web tool?!

It was broken for a short period earlier in the week, due to some upstream changes but should be fixed now. See #27 for the details. If its still not working for you now, grab the debug logs from the web flasher and post here.

Jumperm, 2023-12-01:
> > But what was wrong while using your web tool?!
> 
> It was broken for a short period earlier in the week, due to some upstream changes but should be fixed now. See #27 for the details. If its still not working for you now, grab the debug logs from the web flasher and post here.

Still can't connect. How do i get the debug?

darkxst, 2023-12-02:
It looks to be failing to download the flasher code. Do you have some privacy addons etc that block javascript or otherwise interfere?

OnkelBuTscH, 2023-12-02:
Greetings,
I have exactly the same problem where I have the connection loading screen but nothing else happens, there is no timeout or error message.
3 weeks ago it worked for me without any problems.

Jumperm, 2023-12-02:
I have a PiHole, that could be the Problem. I will try it and give you Feedback, but this weekend I have no time.

OnkelBuTscH, 2023-12-02:
That was indeed the case when I temporarily deactivated the pihole for my client it worked again. 
Unfortunately, I couldn't figure out which domain it was because none of them was listed as blocked.

darkxst, 2023-12-02:
> @darkxst where can I buy you a coffee ;) Big thx for support and the great tool!

@Jumperm You can find my page here, if you would like to help support my projects that would be great, thanks
https://www.buymeacoffee.com/darkxst