darkxst/silabs-firmware-builder #30: Web flasher fails

V4ler1an, 2023-12-12:
When I run the web flasher (Chrome or Edge) it finds the dongle on its serial port ok, gets half way round the connecting progress dial then stops and goes no further. Any idea what this might be? Is there any way to diagnose where it's stopping? I'm using Win 11.  Thanks.

[image]

[image]

1000sabords, 2023-12-12:
Same problem for me (Chrome/w10)

azizou016, 2023-12-12:
Same problem chrome/edge w11 and macos

scobby, 2023-12-12:
same problem here under mac os (m2 pro) and windows 11 on a different computer

darkxst, 2023-12-12:
There is currently an issue with the web flasher due to upstream dependency issue See #29 and https://github.com/NabuCasa/sl-web-tools/issues/11

Should be fixed soon.

sormy, 2023-12-13:
As an alternative for zbdonglee:

1. Download firmware from
   https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee
2. Disassemble dongle (take 2 screws of any side), while pressing BOOT button insert into PC/Mac usb port
3. Download SecureCRT for PC/Mac (trial is 30 days):
   https://www.vandyke.com/cgi-bin/releases.php?product=securecrt
4. Follow:
   https://sonoff.tech/wp-content/uploads/2022/11/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf

darkxst, 2023-12-13:
> As an alternative for zbdonglee:

OR just use `Universal-silabs-flasher`
https://github.com/NabuCasa/universal-silabs-flasher

youkorr, 2023-12-13:
Hello, is the problem fixed?

darkxst, 2023-12-13:
Not just yet. Puddly is working on this fix still
https://github.com/NabuCasa/sl-web-tools/pull/12

youkorr, 2023-12-13:
ok

darkxst, 2023-12-14:
This should be fixed now.

V4ler1an, 2023-12-14:
Flashed successfully using the web flasher, thank you!

V4ler1an, 2023-12-14:
Completed.

scobby, 2023-12-14:
can not confirm that this is fixed, just tried to flash the Sonoff Dongle-E with the Multipan firmware, didnt work. 

Used Website:
https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware

This is the output of the console. Its missing a python package SSL:
[image]

V4ler1an, 2023-12-14:
Reopened for you...

darkxst, 2023-12-14:
Actually that one is related, but I forget to push to fix to the blog. That should be fixed as well now.

darkxst, 2023-12-30:
These issues should be all fixed now.

closed 2023-12-30 by darkxst