darkxst/silabs-firmware-builder #32: web flasher stuck on 87,3%

iUnstable0, 2023-12-14:
Im trying to flash ym ZBDongle-E 

[image]

Its stuck at 87.3%
do I give it time or reload the page and try again? console said this

[image]

im flashing OpenThread not multi pan

iUnstable0, 2023-12-14:
seems like this issue is very random i flashed again with zigbee ezsp firmware and it worked then tried thread and got stuck ahain and when i tried zigbee ezsp it got stuck at around 15% with the same error memory out of bounds

darkxst, 2023-12-14:
Ive not seen this error before, however if its really getting stuck part way through download suprised it still gets detected in the web flasher when you reload.

Will probably need to report this error upstream on the sl-web-tools repo.

iUnstable0, 2023-12-14:
yeah im surprised too that the adapter isnt bricked

darkxst, 2023-12-14:
Can you see the progress messages spamming the console logs? 

the zbdonglee is unbrickable though, you can always  force bootloader with `universal-silabs-flasher` and --sonoff-reset flag, or the internal physical boot button.

iUnstable0, 2023-12-26:
Im really sorruy for the late reply i didnt get the notification. Nope i cant see the progress logs but updating chrome fixed this issue. Thank you though! 🙏🏻

closed 2023-12-26 by iUnstable0