darkxst/silabs-firmware-builder #35: Bricked my iHost chip by flashing wrong image

StofLE, 2023-12-25:
I used Silicon Labs Flasher Addon in HAOS and flashed a wrong FW image (i think Yellow or Skyconnect).

Installed universal-silabs-flasher and tried it with --device /dev/ttyS4 --bootloader-baudrate 460800,115200,230400 --ezsp-baudrate 460800,115200,230400 --cpc-baudrate 460800,115200,230400 --spinel-baudrate 460800,115200,230400 probe to check all baudrates and probe-methods. Error: Failed to probe running application type.

Is there a way to debrick it?

darkxst, 2023-12-25:
Yes you can fix with this command
`universal-silabs-flasher --device /dev/ttyACM0 --probe-method bootloader flash --firmware <firmware.gbl> --sonoff-reset`

StofLE, 2023-12-26:
> Yes you can but you need to download universal-silabs-flasher from git, this code hasnt been released yet `universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset ihost flash --firmware <firmware.gbl>`
> 
> Edit: fix command!

Thanks...I installed it from github and started flasher for device ttyS4. It shows "Triggering ihost bootloader" and probes but still failed. 

`homeassistant:/config# universal-silabs-flasher --device /dev/ttyS4 --bootloader-reset ihost flash --firmware ncp-uart-hw-v7.3.2.0-zbdonglee-115200.gbl 
2023-12-26 12:28:31 homeassistant universal_silabs_flasher.flash[137] INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.3.2', ezsp_version='7.3.2.0', ot_rcp_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2023-12-26 12:28:31 homeassistant universal_silabs_flasher.flasher[137] INFO Triggering ihost bootloader
2023-12-26 12:28:31 homeassistant universal_silabs_flasher.flasher[137] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-12-26 12:28:33 homeassistant universal_silabs_flasher.flasher[137] INFO Probing ApplicationType.EZSP at 115200 baud
2023-12-26 12:28:38 homeassistant universal_silabs_flasher.flasher[137] INFO Probing ApplicationType.CPC at 460800 baud
2023-12-26 12:28:43 homeassistant universal_silabs_flasher.flasher[137] INFO Probing ApplicationType.CPC at 115200 baud
2023-12-26 12:28:47 homeassistant universal_silabs_flasher.flasher[137] INFO Probing ApplicationType.CPC at 230400 baud
2023-12-26 12:28:51 homeassistant universal_silabs_flasher.flasher[137] INFO Probing ApplicationType.SPINEL at 460800 baud
Error: Failed to probe running application type
homeassistant:/config# `

No zigbee service is running on iHost (HAOS 11.3.rc1). More ideas? :-)

darkxst, 2023-12-26:
Are you sure nothing else is connected to the serial port? Logs suggest to me that it failed to connect to bootloader. Make sure ZHA or Z2M are disabled.

Can you also run again with the `-v` flag to get more detailed logs
```
universal_silabs_flasher -v --device /dev/ttyS4 --bootloader-reset ihost probe
```
You should be something like this
```
2023-12-26 22:56:03 a0d7b954-ssh universal_silabs_flasher.flasher[3136] INFO Triggering ihost bootloader
2023-12-26 22:56:04 a0d7b954-ssh universal_silabs_flasher.flasher[3136] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-12-26 22:56:04 a0d7b954-ssh universal_silabs_flasher.flasher[3136] INFO Launched application from bootloader
2023-12-26 22:56:04 a0d7b954-ssh universal_silabs_flasher.flasher[3136] INFO Detected bootloader version '1.A.3' (1.3)
```
It will then continue probing, fail with no app found, but it can  re-enter bootloader (again with `INFO Triggering ihost bootloader`) to flash firmware

StofLE, 2023-12-26:
`homeassistant:/config# universal-silabs-flasher -v --device /dev/ttyS4 --bootloader-reset ihost probe
2023-12-26 14:43:24 homeassistant asyncio[215] DEBUG Using selector: EpollSelector
2023-12-26 14:43:24 homeassistant universal_silabs_flasher.flasher[215] INFO Triggering ihost bootloader
2023-12-26 14:43:24 homeassistant universal_silabs_flasher.flasher[215] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-12-26 14:43:24 homeassistant zigpy.serial[215] DEBUG Opening a serial connection to '/dev/ttyS4' (115200 baudrate)
2023-12-26 14:43:24 homeassistant universal_silabs_flasher.common[215] DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0xa4e8b410>, Serial<id=0xa4e849d0, open=True>(port='/dev/ttyS4', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2023-12-26 14:43:24 homeassistant universal_silabs_flasher.common[215] DEBUG Sending data b'3'
2023-12-26 14:43:26 homeassistant universal_silabs_flasher.flasher[215] INFO Probing ApplicationType.CPC at 460800 baud
2023-12-26 14:43:26 homeassistant zigpy.serial[215] DEBUG Opening a serial connection to '/dev/ttyS4' (460800 baudrate)
2023-12-26 14:43:26 homeassistant universal_silabs_flasher.common[215] DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0xa4e8b5f0>, Serial<id=0xa4e84928, open=True>(port='/dev/ttyS4', baudrate=460800, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2023-12-26 14:43:26 homeassistant universal_silabs_flasher.cpc[215] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-12-26 14:43:26 homeassistant universal_silabs_flasher.common[215] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-12-26 14:43:27 homeassistant universal_silabs_flasher.cpc[215] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-12-26 14:43:27 homeassistant universal_silabs_flasher.cpc[215] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-12-26 14:43:27 homeassistant universal_silabs_flasher.common[215] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-12-26 14:43:28 homeassistant universal_silabs_flasher.cpc[215] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-12-26 14:43:29 homeassistant universal_silabs_flasher.cpc[215] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-12-26 14:43:29 homeassistant universal_silabs_flasher.common[215] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-12-26 14:43:30 homeassistant universal_silabs_flasher.cpc[215] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-12-26 14:43:30 homeassistant universal_silabs_flasher.cpc[215] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-12-26 14:43:30 homeassistant universal_silabs_flasher.common[215] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-12-26 14:43:31 homeassistant universal_silabs_flasher.cpc[215] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-12-26 14:43:31 homeassistant universal_silabs_flasher.flasher[215] INFO Probing ApplicationType.CPC at 115200 baud
2023-12-26 14:43:31 homeassistant zigpy.serial[215] DEBUG Opening a serial connection to '/dev/ttyS4' (115200 baudrate)
[... cut ...]
2023-12-26 14:43:48 homeassistant universal_silabs_flasher.spinel[215] DEBUG Sending frame SpinelFrame(header=SpinelHeader(transaction_id=3, network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02')
2023-12-26 14:43:48 homeassistant universal_silabs_flasher.common[215] DEBUG Sending data b'~\x83\x02\x02\xe65~'
2023-12-26 14:43:49 homeassistant universal_silabs_flasher.spinel[215] DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
Error: Failed to probe running application type`

darkxst, 2023-12-27:
There is no response whatsover from the zigbee chip, given the bootloader is unbrickable, this maybe a hardware issue. I suggest you contact iTead for warranty replacement.

StofLE, 2023-12-27:
It is fixed. Thanks for helping me!

I soldered some wires inaccurate at the EFR32 to connect externally and made a shortage over two pins. Awkward. :-)

After cleaning this triggering bootloader and flashing firmware worked fine.

darkxst, 2023-12-28:
Ok, closing this then.

closed 2023-12-28 by darkxst