darkxst/silabs-firmware-builder #38: AttributeError: can_write_custom_eui64 not found in COMMANDS

IgnacioHR, 2023-12-28:
I've just purchased a second dongle from sonoff (EZSP) and wanted to test this new firmware versions.

After successfuly flashed `ncp-uart-hw-v7.3.2.0-zbdonglee-none-115200.gbl` I tried to execute `universal-silabs-flasher --device /dev/tty.usbserial-202305080921111 write-ieee --ieee e0798dfffe77c2f0` in order to write to the new dongle the same address I've in the old one. I'm not 100% sure this is required but I think it would be required to avoid having to re-pair the entire network.

But the command failed:
```
universal-silabs-flasher --device /dev/tty.usbserial-202305080921111 write-ieee --ieee e0798dfffe77c2f0
2023-12-28 18:28:43 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-12-28 18:28:45 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.CPC at 460800 baud
2023-12-28 18:28:49 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.CPC at 115200 baud
2023-12-28 18:28:54 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.CPC at 230400 baud
2023-12-28 18:28:58 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.EZSP at 115200 baud
2023-12-28 18:28:59 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Detected ApplicationType.EZSP, version '7.3.2.0 build 212' (7.3.2.0.212) at 115200 baudrate (bootloader baudrate None)
2023-12-28 18:29:00 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Current device IEEE: e0:79:8d:ff:fe:d9:0e:a8
Traceback (most recent call last):
  File "/Users/ihr/zigbee-sonoff-firmware/venv/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/universal_silabs_flasher/flash.py", line 36, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/Users/ihr/.asdf/installs/python/3.9.13/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Users/ihr/.asdf/installs/python/3.9.13/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/universal_silabs_flasher/flash.py", line 250, in write_ieee
    await ctx.obj["flasher"].write_emberznet_eui64(new_eui64)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/universal_silabs_flasher/flasher.py", line 353, in write_emberznet_eui64
    if not await ezsp.can_write_custom_eui64():
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/bellows/ezsp/__init__.py", line 300, in __getattr__
    return getattr(self._protocol, name)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/bellows/ezsp/protocol.py", line 132, in __getattr__
    raise AttributeError(f"{name} not found in COMMANDS")
AttributeError: can_write_custom_eui64 not found in COMMANDS
```

Q1: ¿is this a hardware limitation at the chip level? so it is not possible to change the address
Q2: Can I replace the zigbee coordinator in my network with another one with a different address? I don't think I can so my only one option will be to flash the old coordinator and pray

Thanks

darkxst, 2023-12-28:
> Q1: ¿is this a hardware limitation at the chip level? so it is not possible to change the address 

No it should work, thats probably a bug. You should file bug report here. here: https://github.com/NabuCasa/universal-silabs-flasher/issues

>Q2: Can I replace the zigbee coordinator in my network with another one with a different address? I don't think I can so my only one option will be to flash the old coordinator and pray

That depends,
ZHA can migrate adapters without changing address and should work without re-pairing
Z2M doesnt support this, and will require repairing if you change address. ( It should be possible to manually migrate backup and restore using zigpy-cli that replicates what ZHA does though)

IgnacioHR, 2023-12-29:
Thanks for your comment. I'm going to close this issue as it looks like it is not related with this project at all.

closed 2023-12-29 by IgnacioHR