darkxst/silabs-firmware-builder #40: new NCP 7.4.0 fw for zbdoongleE: Z2M doesn't start

paolo1983, 2024-01-05:
Hi @darkxst , this morning i updated my sonoff zbdoongleE to the last NCP fw (7.4.0) and I got a big issue: Z2M does not start!!

I read this from log: Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

I come back to previous 7.3.2 and z2m works well

jonyrh, 2024-01-05:
I also have a not start on 7.4...

```
[20:14:59] INFO: Preparing to start...
[20:14:59] INFO: Socat not enabled
[20:15:00] INFO: Starting Zigbee2MQTT...
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Error: {"address":0,"clusterId":32770,"sequence":2} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

anto294, 2024-01-05:
According to https://github.com/home-assistant/addons/pull/3388,
it seems that z2m does not support gecko 4.4.0

darkxst, 2024-01-05:
7.4.0.0 introduces EZSP protocol v13, Z2M probably might need patches to support this new version.

paolo1983, 2024-01-06:
> 7.4.0.0 introduces EZSP protocol v13, Z2M probably might need patches to support this new version.

Ok. So i can close this issue...and waiting for z2m upgrade

darkxst, 2024-01-08:
Follow this issue for tracking upstream when Z2M gets fixed

https://github.com/Koenkk/zigbee-herdsman/issues/854

darkxst, 2024-01-09:
Lets keep this one open for visibility

chris-1243, 2024-02-04:
The latest version of Zigbee2mqtt (1.35.2) has solved this issue. I just had a try and it is working fine

paolo1983, 2024-02-05:
Yes with the last It works

closed 2024-02-05 by paolo1983