darkxst/silabs-firmware-builder #42: [Request] Support Sonoff ZBBridge

aioue, 2024-01-05:
https://zigbee.blakadder.com/Sonoff_ZBBridge.html

"ITead Sonoff ZBBridge Zigbee Bridge is based on an Espressif ESP8266/ESP8285 and an “SM-011 V1.0” radio SiP module by CoolKit-Technologies (based on Silicon Labs EFR32MG21 MCU)."

darkxst, 2024-01-05:
Sonoff ZBBridge requires signed firmware, which I cannot build. as I dont have access to the keys.
I think its possible to flash an unlocked bootloader, but you need to SWD to do this. If the bootloader had been unlocked then you could use the zbdonglee images on this.

This model has also been discontinued by iTead.

Hedda, 2024-01-17:
> Sonoff ZBBridge requires signed firmware, which I cannot build. as I dont have access to the keys.

@aioue FYI, the original firmware firmware was developed by [xsp1989](https://github.com/xsp1989) who make signed builds for the community upon requests:

https://github.com/xsp1989/zigbeeFirmware

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011-signed

https://github.com/xsp1989/ZB-GW03_firmware/tree/master/tasmota

https://github.com/xsp1989/Tasmota/tree/development/tools

https://github.com/xsp1989/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

Those are the same files used by Tasmota for and ZB-GW03

https://github.com/arendst/Tasmota/tree/development/tools

https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

https://tasmota.github.io/docs/Zigbee/

https://github.com/arendst/Tasmota/discussions/12764

https://github.com/arendst/Tasmota/discussions/11536

closed 2024-01-25 by aioue