darkxst/silabs-firmware-builder #47: Unable to flash Sonoff ZBDongle-E

gyenesadam, 2024-02-01:
Hi!

I tried to flash my ZBDongle, but after I start is, it stay at 0.0%...well...forever.
The webflasher deteched, the dongle, the actual firmware, but unable to flash is.

Could anybody help me?

darkxst, 2024-02-01:
Can you check if there are any errors logged in the developer console (F12)?

gyenesadam, 2024-02-02:
I've got the problem.

After i changed the firmware for the original with 3rd party console app, the webflasher worked succesfully.

Maybe there is one strange thing. In Z2M, all devices LQI is 255.

darkxst, 2024-02-02:
> Maybe there is one strange thing. In Z2M, all devices LQI is 255.

Is this with RCP multipan firmware? Its a know issue that child devices connected directly to coordinator dont report correct LQI. However devices connected via another router do report correctly.

gyenesadam, 2024-02-02:
Yes, thats the "problem" thanks for the help!

closed 2024-02-02 by gyenesadam