darkxst/silabs-firmware-builder #48: GSDK 4.4.0

chris-1243, 2024-02-04:
I have just seen this [thread](https://github.com/home-assistant/addons/pull/3429) from Homeassistant community.

According to it, it seems the latest GSDK brings lot of issuse. I guess you are aware of it.
In cass some users are not, it may be good to not update and keep an older GSDK.

darkxst, 2024-02-04:
yes, 4.4.0 is very problematic, Multiprotocol Addon has downgraded to 4.3.1 for the time being.

My webflasher and HA flasher addon are both still serving 4.3.1

chris-1243, 2024-02-04:
Would you mind if I don't closed this issue to keep it visible for others users who might be facing issues with this GSDK ?

darkxst, 2024-02-04:
its not possible to use 4.4.0 at the given multiprotocol addon is on 4.3.x again.

Hedda, 2024-02-15:
FYI, GSDK 4.4.1.0 bug-fix release is now GA -> https://github.com/darkxst/silabs-firmware-builder/issues/54

darkxst, 2024-03-07:
EmberNet NCP 7.4.1 is working well by all reports I have hard.
Multiprotocol is still recommend to use 4.3.1

closed 2024-07-09 by chris-1243