darkxst/silabs-firmware-builder #52: Connection Failed

baustoff, 2024-02-13:
Hey, 
I connected my Sonoff Zigbee Plus v2 (E) Dongle to my Win11-PC and also on different ports, but I always get the error-message "Connection Failed - The running firmware could not be detected". I also stopped pihole and tried it on a Linux Mint System with Chrome Browser, but same error message.

Here is the log:

2024-02-13 17:50:02 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-02-13 17:50:02 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2024-02-13 17:50:02 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x11493e0>
2024-02-13 17:50:02 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2024-02-13 17:50:02 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x11'
2024-02-13 17:50:02 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x11')
2024-02-13 17:50:03 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x1a'
2024-02-13 17:50:03 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x11\x1a')
2024-02-13 17:50:03 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\xc1\x02\x02\x9b{~'
2024-02-13 17:50:03 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x11\x1a\xc1\x02\x02\x9b{~')
2024-02-13 17:50:04 emscripten webserial_transport[42] DEBUG Closing serial port
2024-02-13 17:50:04 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2024-02-13 17:50:04 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
2024-02-13 17:50:04 emscripten webserial_transport[42] DEBUG Closed serial port
2024-02-13 17:50:04 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1b52c90>
2024-02-13 17:50:04 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-02-13 17:50:04 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-02-13 17:50:04 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'x'
2024-02-13 17:50:04 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x00'
2024-02-13 17:50:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2024-02-13 17:50:05 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80'
2024-02-13 17:50:05 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\xf8\x00x\x80\x00\x80\x00\xf8\xf8\x80\xf8\xf8\xf8\x80\xf8'
2024-02-13 17:50:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'\xf8'): ValueError('Invalid flag')
2024-02-13 17:50:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-02-13 17:50:05 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-02-13 17:50:06 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2024-02-13 17:50:06 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-02-13 17:50:06 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-02-13 17:50:07 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2024-02-13 17:50:07 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-02-13 17:50:07 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-02-13 17:50:08 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2024-02-13 17:50:08 emscripten webserial_transport[42] DEBUG Closing serial port
2024-02-13 17:50:08 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 115200 baud
2024-02-13 17:50:08 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2024-02-13 17:50:08 emscripten webserial_transport[42] DEBUG Closed serial port
2024-02-13 17:50:08 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1149430>
2024-02-13 17:50:08 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-02-13 17:50:08 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-02-13 17:50:08 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x11'
2024-02-13 17:50:09 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2024-02-13 17:50:09 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x1a'
2024-02-13 17:50:09 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\xc1\x02\x02\x9b{~'
2024-02-13 17:50:09 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'~'): ValueError('Invalid flag')
2024-02-13 17:50:09 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-02-13 17:50:09 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
[... cut ...]
2024-02-13 17:50:25 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'~\x83\x02\x02\xe65~'
2024-02-13 17:50:26 emscripten universal_silabs_flasher.spinel[42] DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
2024-02-13 17:50:26 emscripten webserial_transport[42] DEBUG Closing serial port
2024-02-13 17:50:26 emscripten webserial_transport[42] DEBUG Closed serial port

darkxst, 2024-02-13:
Not sure about on windows, but on linux you need to your user to 'dialout' group so he has permissions to the usb serial device.

baustoff, 2024-02-14:
> Not sure about on windows, but on linux you need to your user to 'dialout' group so he has permissions to the usb serial device.

Thank you so much, that fixed my problem in Linux at least and now I'm able to use your webflasher 🙂👍
Although it would be also interesting to know which specific permission is necessary for Windows 🤔

closed 2024-02-14 by baustoff