darkxst/silabs-firmware-builder #53: [REQUEST] OpenThread NCP builds (so do not need to use OpenThread Daemon)

Hedda, 2024-02-14:
Please consider also offering OpenThread NCP (Network Co-Processor) builds as an optional alternative to OpenThread RCP.

https://openthread.io/platforms/co-processor

PS: OpenThread NCP firmware will not work with the "OpenThread Border Router Add-on" for Home Assistant as that currently look to require OpenThread RCP firmware but as long as it has Spinel+hdlc+uart serial interface it should still be possible to make it work with Thread based integrations as well as for other Thread implementations such as wpantund:

Reference:

https://openthread.io/platforms/co-processor/wpantund

https://github.com/home-assistant/addons/tree/master/openthread_border_router

https://www.home-assistant.io/integrations/thread/

https://www.home-assistant.io/integrations/otbr/

https://www.home-assistant.io/integrations/matter/

https://github.com/home-assistant/addons-development/issues/38

darkxst, 2024-02-14:
This pretty much requires wpantund to work? At which point not sure what the benefit is over the RCP. Although I suppose it might be useful for people who dont use Home Assistant.

Hedda, 2024-02-14:
Yeah I think it would still be useful for other projects.

I have not researched too deeply into it myself (as do not yet have any Thread devices myself), however from what I understand the [Open Thread Border Router integration](https://www.home-assistant.io/integrations/otbr/) that I believe uses the [OpenThread Border Router Add-on for Home Assistant](https://github.com/home-assistant/addons/tree/master/openthread_border_router) is currently only compatible with OpenThread RCP firmware, but I guess that might change in the future as I think it should technically be possible to make the [Thread integration](https://www.home-assistant.io/integrations/thread/
) (that the [Matter integration](https://www.home-assistant.io/integrations/matter/) in turn depends on) work directly with OpenThread NCP firmware?

That is at least what I understood from this discussion that asks about using a CC2652 based adapter with the Open Thread Border Router integration for Home Assistant -> https://github.com/home-assistant/addons-development/issues/38

PS: slaesh for example offers both OpenThread RCP and OpenThread NCP firmware images for his CC2652 based adapter, with the mentioning that [wpantund](https://github.com/openthread/wpantund) can be used with OpenThread NCP firmware:

https://slae.sh/projects/cc2652/#openthread-border-router

https://slae.sh/projects/cc2652/#matter-support

darkxst, 2024-02-14:
> I think it should technically be possible to make the [Thread integration](https://www.home-assistant.io/integrations/thread/) (that the [Matter integration](https://www.home-assistant.io/integrations/matter/) in turn depends on) work directly with OpenThread NCP firmware?

I dont think the Openthread NCP is standalone like the Zigbee ones. You need wpantund to provide the ipv6 network layer, and then you may still require otbr daemon. However once those requirements are met, HA Thread (and otbr) integration should pick this up automatically as long as its in the same subnet.