darkxst/silabs-firmware-builder #54: Gecko Platform 4.4.1.0 GA released with Zigbee EmberZNet SDK 7.4.1.0 GA

Hedda, 2024-02-15:
Gecko Platform 4.4.1.0 GA (bug-fix release 4.4.1.0) released with Zigbee EmberZNet SDK 7.4.1.0 GA (bug-fix release 7.4.0.0):

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.4.1

MattWestb has summarized the included fixes for Zigbee EmberZNet NCP and RCP here:

https://github.com/zigpy/zigpy/discussions/894#discussioncomment-8477113

Hedda, 2024-02-15:
Related to https://github.com/darkxst/silabs-firmware-builder/issues/48

darkxst, 2024-02-15:
Test builds are here for initial testing
https://github.com/darkxst/silabs-firmware-builder/tree/4.4.1

wastez, 2024-02-17:
Already tried it somebody?

wastez, 2024-02-19:
@darkxst 
Is the firmware a beta (Yes i know everything here is experimental) or a earlier state? 
Because i could'nt find it in the silabs source.
Thinking about to test it but don't have a test net, so it would be my live system.
Thats the reason why i ask.

Hedda, 2024-02-19:
@wastez the previous version has been problematic so risk is that this is too, see https://github.com/darkxst/silabs-firmware-builder/issues/48

wastez, 2024-02-19:
@Hedda 
Yes i know, but seems only to be in the multiprotocol version, no problems here so far on the ncp version.

darkxst, 2024-02-19:
> @darkxst Is the firmware a beta

From Silabs side, I dont think I have ever see them mark a firmware as beta. However the fact they marked 4.4.1 GA, means that 4.4.0 was not production ready.  They tend to just mark new features as experimental, and then later mark the feature production ready.

>  no problems here so far on the ncp version.

NCP Version is not impacted by the stability issues seen with Multipan, it should be safe to use.

> previous version has been problematic so risk is that this is too

I dont see anything in the changelog that addresses the previous issues either.