darkxst/silabs-firmware-builder #56: Cannot flash my Sonoff ZBDongle-E with web flasher

Blazzert77, 2024-02-24:
Good morning

I hope this is the right project to report such an issue. I'm trying to flash a newly acquired Sonoff ZBDongle-E. Never used,

I tried flashing it from 2 separate computers, both running Windows 11, with the latest versions of Chrome or Edge. Both fail

when flashing. The dongle is plugged into a USB port on the device, not a cable or a USB hub.

Project

I use the url: https://darkxst.github.io/silabs-firmware-builder/

There I select ZBDongle-E and Select. The browser then asks me to select a device, I select the one listed:

[image]
[image]

darkxst, 2024-02-25:
Do you have drivers installed? you either need Silabs universal cp2102 driver or Qinheng CH343 drivers. Either should work.

Blazzert77, 2024-02-25:
I think I installed the cp210 driver correctly

Once I download them, I unzip them then double click on silabser.inf, is that right?

darkxst, 2024-02-25:
I though they had an installer, but that would work also.

Blazzert77, 2024-02-25:
it still doesn't work I tried with both drivers

darkxst, 2024-02-25:
in that case it maybe a permission error also.

Blazzert77, 2024-02-25:
how can I correct this? from what I understand that I am the only one to have this problem is in the log nothing indicates any particular problem?

Blazzert77, 2024-02-25:
Or can we do it manually with flash software without going through the web?

Nerivec, 2024-02-26:
```log
2024-02-24 17:45:58 emscripten bellows.uart[42] DEBUG RSTACK Version: 2 Reason: RESET_POWER_ON frame: b'c102029b7b7e'
2024-02-24 17:45:58 emscripten bellows.ezsp[42] INFO NCP entered failed state. No application handler registered, ignoring...
```

That's definitely not the right reset reason here. And once the NCP is in failed state, it will refuse anything but a reset... that won't come because the flasher moves on to another probing method when one fails.

You can try this:
- Install SecureCRT (trial version is fine), or another similar software
- Use it to flash Sonoff [factory NCP firmware](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP) (procedure described in [Sonoff PDF](https://sonoff.tech/wp-content/uploads/2022/07/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf))
- Come back to darkxst flasher and flash the new firmware you want

I got a couple of users' adapters back from the "seemingly dead" using this exact procedure...

darkxst, 2024-02-26:
> I got a couple of users' adapters back from the "seemingly dead" using this exact procedure...

Same can be achieved with [Universal-silabs-flasher](https://github.com/NabuCasa/universal-silabs-flasher) and bootloader reset
```
universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware file.gbl
```

Nerivec, 2024-02-26:
I tried the cmd flasher first, but it didn't work on the first adapter I revived (even with all manual params).
Then I guess I never tried it again, I just did it like I mentioned since I knew that worked...
I'll try again on the next one 😉

darkxst, 2024-02-26:
Bootloader reset forces the dongle into bootloader, just like pressing the physical buttons, from that point it is always flashable even if corrupt firmware on it.

also the `--bootloader-reset` option is new, but previously it existed as `--sonoff-reset`

Blazzert77, 2024-02-26:
hello I tried your technique however I have lots of errors I have a little difficulty I am a beginner would it be possible for you to make a video from A to Z that I can understand because I have this error which persists 

```
universal-silabs-flasher: The term "universal-silabs-flasher" is not recognized as a cmdlet name,
function, script file, or executable program. Check the spelling of the name, or if a path exists,
verify that the path is correct and try again.
To the character Line:1: 1
+ universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sound ...
+ ~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo: ObjectNotFound: (universal-silabs-flasher:String) [], CommandNotFoundException
    + FullyQualifiedErrorId: CommandNotFoundException
```

darkxst, 2024-02-26:
> --device /dev/ttyACM0

Also on windows device will be `COM#` which you check what the number in device manager

darkxst, 2024-02-26:
For COM4 you would do

```
universal-silabs-flasher --device COM4 --bootloader-reset sonoff flash --firmware ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl
```

However its not finding the command so you might need to add python package directory to your PATH Environment variable.
```
C:\Users\blazz\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages
```

Blazzert77, 2024-02-26:
I added the variable to the path we can see that python is recognized however still the same problem

I must be doing something wrong but I can't see
[image]
[image]

Blazzert77, 2024-02-26:
ok I had already tried that I took a skyconnect home assistant key to see if the problem does not come from the key then otherwise I would wait until matter is available on the sky connect

Nerivec, 2024-02-26:
@Blazzert77 
Open folder
`C:\Users\blazz\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\Scripts`
Then right click in that folder and select "Open Terminal"
Then run the command like this in that open terminal:
`.\universal-silabs-flasher.exe --device COM4 --bootloader-reset sonoff flash --firmware ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl`

PS: for future use, you will want to put that folder in the PATH Environment variable, then it should work from anywhere.

Blazzert77, 2024-02-27:
> Ouvrir le dossier Cliquez ensuite avec le bouton droit de la souris dans ce dossier et sélectionnez « Ouvrir le terminal » Exécutez ensuite la commande comme ceci dans ce terminal ouvert :`C:\Users\blazz\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\Scripts` `.\universal-silabs-flasher.exe --device COM4 --bootloader-reset sonoff flash --firmware ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl`
> 
> PS : pour une utilisation future, vous voudrez mettre ce dossier dans la variable d’environnement PATH, alors il devrait fonctionner de n’importe où.

by following your instructions I got a new error that I didn't have before
[image]

Nerivec, 2024-02-27:
That last part `ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl` needs to be the name of the firmware file you downloaded, and it needs to be in the same folder where you are executing the command.

Hedda, 2024-03-19:
> The dongle is plugged into a USB port on the device, not a cable or a USB hub.

Probably not your problem, but FYI, due to their physicial design it is recommend always using a USB cable with the Sonoff ZBDongle-E (and  Sonoff ZBDongle-P) because the made the casing/enclosure for those dongles too large that it affects how long the USB plug actually sticks out so is therefor does not always make proper physical contact, at least on some computers that have USB ports that are partially sunken into the computer chassi

Sr-psycho, 2024-03-22:
<html>
<body>

Guys, what firmware is suitable for this dongle?  Devices are falling off on this one (SDK Version | 4.3.1
-- | --
7.3.1.0) , I want to roll back to the previous version
PS I didn’t reconnect the devices, I just updated the firmware and connected the dongles.

</body>
</html>

lanwin, 2024-04-22:
To make that work, you need to open the device. Inside is a reset and boot button. You need to hold down the boot button before inserting the device and still hold it down until the flasher showing the flash progress. This at least works for me.

darkxst, 2024-04-22:
Or just use `universal-silabs-flasher` with  `--bootloader-reset sonoff`. No disassembly required then.

lanwin, 2024-04-22:
> Or just use `universal-silabs-flasher` with `--bootloader-reset sonoff`. No disassembly required then.

That did not worked for me.

darkxst, 2024-04-22:
Do you have the latest version installed? prior to v0.0.16 it was a different flag.

darkxst, 2024-04-22:
if it shows `--bootloader-reset` on the help page, its new enough, if it doesnt upate using pip

darkxst, 2024-04-23:
then it should work, but depending on firmware you may need to set baudrate to probe for example
`--ezsp-baudrate 230400` or otherwise you can try probe only bootloader `--probe-method bootloader`.

whisdol, 2024-09-03:
> To make that work, you need to open the device. Inside is a reset and boot button. You need to hold down the boot button before inserting the device and still hold it down until the flasher showing the flash progress. This at least works for me.

Thank you, this worked for me in combination with this command (in a Administrator Command Prompt):
```
universal-silabs-flasher --device COM5 --ezsp-baudrate 230400 flash --firmware C:\Users\user\Downloads\ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl
```
After flashing this, performing the same procedure in the Web Flasher also worked (hold boot button, insert dongle, keep holding button, click connect on the flasher, keep holding until the device is connected), so installing/running the universal-silabs-flasher may not be necessary.
The Web Flasher showed a bootloader firmware (`Bootloader (recovery) 1.12.0`) as installed. After then flashing the Multi-PAN firmware, the dongle also gets connected to the Web Flasher without holding the boot button, but flashing a firmware only works when the boot button is pressed during connection (until the dongle is recognized with the bootloader firmware).

andrebrait, 2024-10-20:
Everything in this page failed for me. The only thing that worked was:

1. Use https://github.com/Nerdiyde/Sonoff_zigbee_3.0_USB_dongle_plus_bootloader_script to force it to enter bootloader mode
2. Use SecureCRT to connect to the right COM port and the right baud rate. No menu will appear.
3. Type 1. It'll now be expecting you to send an xmodem tranfer. 
4. Click Transfer -> Send Xmodem and select the firmware file
5. When finished, type 2

After this, this script now works fine, but before it seems it simply could not put the device into bootloader mode no matter what. When connecting to the COM port, it received `~{` instead of the expected menu, and it then crashed with a Timeout error. 

Unfortunately I forgot to copy the result from my verbose run, but that's what I remember from it.

Alexander-Ger-Reich, 2024-10-22:
I can confirm that the situation is still the same

mdddev, 2024-10-26:
Same here. No matter what, web flasher does not work and shows above error message that the current firmware version cannot be detected. I'm on Ubuntu not Windows, though.

darkxst, 2024-10-26:
> Same here. No matter what, web flasher does not work and shows above error message that the current firmware version cannot be detected. I'm on Ubuntu

I use Ubuntu myself, so it should be working there. However your user needs to be part of `dialout` group. Though please file a new issue and attach logs etc.

mdddev, 2024-10-27:
Thank you. That would have been it, I'm sure. My user was not a group member. I interpreted the tutorials in a way that that step was only necessary for interactive flashing via command line. My bad. I got it working now with the dockerised Home Assistant add-on.using firmware in this repo. Thank you!

blackhead1981, 2024-10-28:
> > After flashing this, performing the same procedure in the Web Flasher also worked (hold boot button, insert dongle, keep holding button, click connect on the flasher, keep holding until the device is connected), so installing/running the universal-silabs-flasher may not be necessary. The Web Flasher showed a bootloader firmware (`Bootloader (recovery) 1.12.0`) as installed. After then flashing the Multi-PAN firmware, the dongle also gets connected to the Web Flasher without holding the boot button, but flashing a firmware only works when the boot button is pressed during connection (until the dongle is recognized with the bootloader firmware).
> 
> I can confirm this. The web flasher works by holding the boot button while connecting the dongle to your computer. I haven't installed nor run anything else (except the drivers).
> 
> So: -Disassemble device (unscrewing the 2 philips screws on the USB-A side is enough) -Locate the boot and reset button (close to the USB connector, two little grey dots) -Hold boot button -Insert device -Click connect in the web flasher -Select the right device -Wait for the dialog to confirm the bootloader version -Release boot button
> 
> This worked for me to flash the new firmware.

This works for me. THX

Doinkydoink, 2024-11-12:
[image]
Unable to update even though I try the hold boot button method :(
Not sure what is wrong.

tootla, 2024-11-13:
@Doinkydoink Please try this solution：Please try this solution：[https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961](https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961)

Doinkydoink, 2024-11-13:
> @Doinkydoink Please try this solution：Please try this solution：[#107 (comment)](https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961)

Thank you for the tip

So maybe i install the wrong driver..do I need to uninstall or I it will overwrite when I install the VCP version. 
I am not sure which driver I should be uninstalling from the device manager screen?

Doinkydoink, 2024-11-14:
> > @Doinkydoink Please try this solution：Please try this solution：[#107 (comment)](https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961)
> 
> Update 14 Nov: after trying will reset/boot button a few times, it manage to jump into the next screen. I managed to update the latest firmware. When I try to connect to the flasher again.. same message appear. All good for now but is going to be pain again when I need to update future firmware. Seems like a roll of dice.
> 
> Many thanks to tootla for the tip regarding using the VCP version,
> 
> Update 13 Nov: Manage to find the driver and delete and install VCP version. Still the same issue :(
> 
> Thank you for the tip
> 
> So maybe i install the wrong driver..do I need to uninstall or I it will overwrite when I install the VCP version. I am not sure which driver I should be uninstalling from the device manager screen?

Ok.. I actually managed to install another driver from the link that seller has just share with me.

https://itead.cc/sonoff-zigbee-devices-work-with-home-assistant/

under the first section, there is a write up (I paste it here)

Install Driver：

ZBDongle-P and New batch of ZBDongle-E (30623XXXXX and above):
Windows: [CP210x_Windows_Drivers_with_Serial_Enumeration](https://itead.cc/wp-content/uploads/2023/05/CP210x_Windows_Drivers_with_Serial_Enumeration.zip)

Old batch of ZBDongle-E: Windows: [CH343SER](https://itead.cc/wp-content/uploads/2023/05/CH343SER.zip)  macOS: [ch34xser_macos-main](https://itead.cc/wp-content/uploads/2023/05/ch34xser_macos-main.zip)

red-black-bonsai, 2024-11-14:
My case:

- On Ubuntu
- Used the webflasher to install `Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl` from Sonoff
- Wanted to use the webflasher to install `ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl` - but got the error mentioned at the top.
- Tried to install `ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl` with universal-silabs-flasher as per instructions in this thread:

      universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl
  
  It seemed to flash alright but in the webinstaller I still got same error message as before.
- Tried to install `ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl` with universal-silabs-flasher:

      universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

  Seemed to flash fine, too - and afterwards the webinstaller could detect the firmware again and I was able to use it as normal :partying_face: 

- On both tries I got a warning during/after baudrate probing:

      INFO Triggering sonoff bootloader
      WARNING Bootloader did not launch a valid application

engineerofshit, 2024-11-22:
> I install the driver under old batch and is now able to get through to the screen without having to press boot/reset button..

Thanks, this one is worked for me.

MankiniChykan, 2024-12-24:
> I can confirm this. The web flasher works by holding the boot button while connecting the dongle to your computer. I haven't installed nor run anything else (except the drivers).
>
> So: -Disassemble [...]

Thanks this worked perfect for flashing with Windows and Chrome.

BenBen54-HA, 2025-04-12:
> > > Après avoir flashé ce périphérique, la même procédure a fonctionné dans le flasheur Web (maintenir le bouton de démarrage enfoncé, insérer la clé, maintenir le bouton enfoncé, cliquer sur « Connecter » sur le flasheur, et maintenir le bouton enfoncé jusqu'à ce que l'appareil soit connecté). L'installation/exécution de l'utilitaire universal-silabs-flasher n'est donc peut-être pas nécessaire. Le flasheur Web a indiqué qu'un firmware de chargeur de démarrage ( `Bootloader (recovery) 1.12.0`) était installé. Après avoir flashé le firmware Multi-PAN, le dongle se connecte également au flasheur Web sans maintenir le bouton de démarrage enfoncé. Cependant, le flashage d'un firmware ne fonctionne que lorsque le bouton de démarrage est enfoncé pendant la connexion (jusqu'à ce que le dongle soit reconnu par le firmware du chargeur de démarrage).
> > 
> > 
> > Je peux le confirmer. Le flasher web fonctionne en maintenant le bouton de démarrage enfoncé tout en connectant la clé à l'ordinateur. Je n'ai rien installé ni exécuté d'autre (à l'exception des pilotes).
> > Donc : -Démonter l'appareil (dévisser les 2 vis cruciformes côté USB-A suffit) -Localiser le bouton de démarrage et de réinitialisation (près du connecteur USB, deux petits points gris) -Maintenir le bouton de démarrage -Insérer l'appareil -Cliquer sur connecter dans le flasher Web -Sélectionner le bon appareil -Attendre la boîte de dialogue pour confirmer la version du chargeur de démarrage -Relâcher le bouton de démarrage
> > Cela a fonctionné pour moi pour flasher le nouveau firmware.
> 
> Merci, cela a parfaitement fonctionné pour le flashage avec Windows et Chrome.

Merci, ça a fonctionné également pour moi en maintenant le bouton S2

Hedda, 2025-04-12:
@BenBen54-HA This is an English-speaking repository.