darkxst/silabs-firmware-builder #57: zb-gw04-1v2 460800 support

alexyuoouy, 2024-02-24:
Hello, why does zb-gw04-1v2 not support baud rate of 460800?

darkxst, 2024-02-24:
Its unstable and causes lots of crc errors at 460800.

alexyuoouy, 2024-03-01:
Why does zb-gw04-1v2 have so many crc errors? Is it a problem with CH340 or the ZYZBP008 module? The efr32mg21 chip should not be problematic.

darkxst, 2024-03-01:
it may be an issue with the CH340, however atleast in theory this should handle 460800. I never dug too deeply though. Stability of the ZBDongle-E is much better even though it doesnt use hw flow control.

alexyuoouy, 2024-03-02:
Alright, but this has been quite helpful. Recently, I've been using the zb-gw04-1v2 dongle for debugging OTBR, and I found that it doesn't work properly at 460800. Now that you've mentioned having similar stability issues, we can essentially pinpoint the cause of the problem. closed.

alexyuoouy, 2024-03-02:
Alright, but this has been quite helpful. Recently, I've been using the zb-gw04-1v2 dongle for debugging OTBR, and I found that it doesn't work properly at 460800. Now that you've mentioned having similar stability issues, we can essentially pinpoint the cause of the problem.  Done

Hedda, 2024-03-19:
@alexyuoouy Are you having issues with [ZHA integration in Home Assistant ](https://www.home-assistant.io/integrations/zha)or [Zigbee2MQTT (and if so "ezsp" or "ember")](https://www.zigbee2mqtt.io/)?

Zigbee2MQTT only having experimental support for Silicon Labs however they have recently been making a lot of progress on their new "ember" adapter driver for zigbee-herdsman that probably will eventually replace their older "ezsp" adapter driver for zigbee-herdsman, however note that needs to use at least latest EmberZNet 7.3.x and preferably mberZNet 7.4.x firmware, see ->  https://github.com/Koenkk/zigbee2mqtt/discussions/21462

Anyway, WCH CH340 should be able to handle higher speeds so maybe driver issue in your specific operating system or kernel?

https://wch-ic.com/products/CH340.html

@xsp1989 perhaps you could help answer why give CRC errors and is unstable if using firmware with 460800 baudrate?

Not same firmware build but you also offer a firmware image with 460800 bauds speed for ZB-GW04 (easyiot dongle):

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

@alexyuoouy Have you also posted issue to xsp1989's repo -> https://github.com/xsp1989/zigbeeFirmware

xsp1989, 2025-03-31:
@darkxst Can you provide your version of the firmware that has a lot of CRC errors? I will also conduct corresponding tests. I built a firmware based on 7.4.5, using hardware flow control and a baud rate of 921600, and did not find any CRC errors.

If it is convenient, can you test the firmware I provided to see if it will also produce CRC errors on your ZB-GW04?

[ncp-uart-hw_7.4.5.0_921600.zip](https://github.com/user-attachments/files/19529186/ncp-uart-hw_7.4.5.0_921600.zip)