darkxst/silabs-firmware-builder #60: possibly bricked Dongle E

sagittarius06, 2024-03-08:
The SL Web Flasher is an usefull and convenient tool, a great job you've done.

I've used it for my Sonoff Dongle E with build firmware available here: https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee version ncp-uart-hw-v7.4.1.0-zbdonglee-230400.gbl. It went well on the dialog box but I guess my stick is bricked.
Is there a way to get it unbrick considering I can still see it:

[103416.234721] usb 1-3: new full-speed USB device number 66 using xhci_hcd
[103416.384186] usb 1-3: New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.42
[103416.384192] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[103416.384194] usb 1-3: Product: SONOFF Zigbee 3.0 USB Dongle Plus V2
[103416.384196] usb 1-3: Manufacturer: ITEAD
[103416.384197] usb 1-3: SerialNumber: 20220812192633
[103416.388075] cdc_acm 1-3:1.0: ttyACM0: USB ACM device

/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812192633-if00

I've tried universal-silabs-flasher -v --device /dev/ttyACM0 flash --firmware ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl with no luck

I may have to buy another stick don't you think ?

darkxst, 2024-03-08:
web flasher doesnt probe 230400, but it will stick work if you set correct baudrate in ZHA or Z2M etc.

You need to specify baudrate for 230400
 ```
  universal-silabs-flasher -v --device /dev/ttyACM0 ezsp-baudrate 230400 probe
  ```

You can't brick these dongles as they have a hardware bootloader reset

```
 universal-silabs-flasher -v --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl
 ```

sagittarius06, 2024-03-09:
It was quick ! So thank you so much for your answer. Because it works by adjusting the baud rate.
Thank you again.

closed 2024-03-09 by darkxst