darkxst/silabs-firmware-builder #61: Sonoff-E - Connection failed

talormanda, 2024-03-11:
[image]

[image]

I receive the error above when trying to connect to my device on COM4. I am unsure what I'm doing wrong or what I can do to fix this.

darkxst, 2024-03-11:
Can you post the Debug logs.

talormanda, 2024-03-11:
> Can you post the Debug logs.

`2024-03-10 22:46:49 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-03-10 22:46:49 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x13f9028>
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\nBL > \x00'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Detected version string '1.12.00'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'2'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'Reset info: 0x04 (PWR)\r\nExtended Reset info: 0x0401 ('
2024-03-10 22:46:49 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'Reset info: 0x04 (PWR)\r\nExtended Reset info: 0x0401 (')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.flasher[42] INFO Launched application from bootloader
2024-03-10 22:46:49 emscripten webserial_transport[42] DEBUG Closing serial port
2024-03-10 22:46:49 emscripten universal_silabs_flasher.flasher[42] INFO Detected bootloader version '1.12.0'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2024-03-10 22:46:49 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
2024-03-10 22:46:49 emscripten webserial_transport[42] DEBUG Closed serial port
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1bb3fb8>
2024-03-10 22:46:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\tx\x80x\x80\x80x\x00\x80x\xf8x\xf8x\x00x\xf8x\x00x\xf8x\x80xxx\x80x\x80x\x00\x80xx\x00\x80xx\x80x\x80x\x80\x80x\x00\x80x\xf8x\xf8x\x00x\xf8x\x00x\xf8x\x80xxx'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'x'): ValueError('Invalid flag')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80x\x80x\x00\x80xx\x00\x80x'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'x'): ValueError('Invalid flag')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'x'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x00x\xf8\x80x\x00xxx\x80x\xf8x\x80\x80xxxxxxx\x80x\x80x\xf8xxxxxxx\x00\x80\x00\x80\x00x\x00x\x00\x80\x80\x80\x80xx\xf8\x00\x80x'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'x'): ValueError('Invalid flag')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80xxxxx\x80\xf8\x80x\x00\xf8xxxxx\x80x\x80\xf8x\xf8x\x80x\x80\xf8\x80x\xf8x\x00x\x00xxx\x80\xf8\x80x\xf8x\x00x\xf8xxx\x80x\x80\x00\x80\x00x\x80\x80\x00\x80x\x00'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'\x00'): ValueError('Invalid flag')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'x\x80\x80\x80\x80\x80\x80\x80\x80\x00\x80\x80xx\xf8x\x80\xf8x\xf8xxx\x80x\x80\xf8\x00x\xf8x\x00x\xf8x\x80x\x80\x00\x80\x80\x80\x80\x00\x80\x00x\x00\x80\x00xxx\xf8x\x80\x80x\x80x\x80'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'\x80'): ValueError('Invalid flag')
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'x\x80'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x00'
2024-03-10 22:46:49 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80\x00\x80\x00x\xf8\x80\x00\x00x\x00x\xf8\x80\x00x\x00\x80\x80\xf8x\xf8xxx\x00x\x80x\x80x\x80x\x80x\x80\x00\x80\x00\x80\x00x\x00x\x00 [... cut ...]
2024-03-10 22:47:10 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'~\x83\x02\x02\xe65~'
2024-03-10 22:47:11 emscripten universal_silabs_flasher.spinel[42] DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
2024-03-10 22:47:11 emscripten webserial_transport[42] DEBUG Closing serial port
2024-03-10 22:47:11 emscripten webserial_transport[42] DEBUG Closed serial port
`

darkxst, 2024-03-11:
What firmware is installed on your dongle?

talormanda, 2024-03-11:
> What firmware is installed on your dongle?

I had used this web tool before to put on https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router

but the device stopped working on home assistant ZHA so i went back here to try and start over and couldn't get anything to work.

darkxst, 2024-03-11:
Okay, web flasher wont work with Router firmware as it doesnt respond to software resets.

Likely best option is to flash directly with `universal-silabs-flasher`, its a bit of a pain on windows though!

This command will get you sorted using a downloaded firmware:

```universal-silabs-flasher --device COM4 --bootloader-reset sonoff flash --firmware <firmwarefile>.gbl```

darkxst, 2024-03-11:
This thread provides some info for getting it to work on Windows.
https://github.com/darkxst/silabs-firmware-builder/issues/56

talormanda, 2024-03-11:
> This thread provides some info for getting it to work on Windows. #56

Yeah I did read through all of that before I posted. I tried using the universal flasher by running:

universal-silabs-flasher --device COM4 --bootloader-reset sonoff flash --firmware Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl

and this is the error i get:

[image]

darkxst, 2024-03-11:
"Permission Denied". Try running as administator

talormanda, 2024-03-11:
> "Permission Denied". Try running as administator

that's the fun part, it is running as administrator.

darkxst, 2024-03-11:
I not sure then, I dont run windows myself.

talormanda, 2024-03-11:
> I not sure then, I dont run windows myself.

i can try this elsewhere where you are familiar. what are you running, linux?

darkxst, 2024-03-11:
Yes Linux.
It works well on Linux, just make sure your user is a member of the `dialout` group to get permissions to the serial port. 

```
universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware <file>.gbl
```

talormanda, 2024-03-11:
> Yes Linux. It works well on Linux, just make sure your user is a member of the `dialout` group to get permissions to the serial port.
> 
> ```
> universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware <file>.gbl
> ```

okay, i will install this quick. is there a particular python version required or should i just need to install the flasher and good to go?

darkxst, 2024-03-11:
It will work with any recent version of Python 3.9+ or so.

just `pip install universal-silabs-flasher` and off you go ;)

darkxst, 2024-03-11:
> is there a good way to get a list of devices to see what the path is on linux?

it will most likely be `/dev/ttyACM0`. However you can view available devices via `ls /dev/tty*`

talormanda, 2024-03-11:
> > is there a good way to get a list of devices to see what the path is on linux?
> 
> it will most likely be `/dev/ttyACM0`. However you can view available devices via `ls /dev/tty*`

super frustrating. i seem to be getting the same error.

[image]

darkxst, 2024-03-11:
Add your user to the `dialout` group, logout and back in again, then try again.

talormanda, 2024-03-11:
> Add your user to the `dialout` group, logout and back in again, then try again.

[image]

does this look better?

if this is finished. do i need to flash with the web or is this good? do you remember how to repair it to ZHA after this?

darkxst, 2024-03-11:
yes looks good. You have re-flashed with router, it will automatically pair as soon as you goto add device in ZHA.

talormanda, 2024-03-11:
> yes looks good. You have re-flashed with router, it will automatically pair as soon as you goto add device in ZHA.

is there some sort of pairing mode to get into? i am doing a search with my skyconnect on ZHA and nothing is appearing.

darkxst, 2024-03-11:
It should automatically enter pairing mode when powered up, unless it still has an old connection stored.

You can try connect to the console on router over serial

```screen /dev/ttyACM0 115200```

from there you should see periodic message that its trying to pair... if not you can run this command to force it to disconnect (I think this is the right command, but been a long time since I played with the router firmware!)

```network leave```

talormanda, 2024-03-11:
> It should automatically enter pairing mode when powered up, unless it still has an old connection stored.
>
> You can try connect to the console on router over serial
>
> `screen /dev/ttyACM0 115200` [...]

yeah, i am not seeing any messages. do i just type network leave? or what kind of command do i need?

talormanda, 2024-03-11:
> It should automatically enter pairing mode when powered up, unless it still has an old connection stored.
>
> You can try connect to the console on router over serial
>
> `screen /dev/ttyACM0 115200` [...]

[image]

any idea what this means?

darkxst, 2024-03-11:
if its display periodically it might be it trying to pair. 

Just try type in network leave and see if that changes anything

talormanda, 2024-03-11:
> if its display periodically it might be it trying to pair.
> 
> Just try type in network leave and see if that changes anything

Yeah that did it.

[image]

I had this, then I did network leave, but I couldnt see what I was typing in the terminal. Then ZHA instantly did:

[image]

Why would it hold onto that if I re-flashed it? That's crazy.

darkxst, 2024-03-11:
NVMEM area is not wiped when you flash normally.

talormanda, 2024-03-11:
> NVMEM area is not wiped when you flash normally.

gotcha. im building documentation for next time. how did you determine the path was /dev/ttyACM0 ?

darkxst, 2024-03-11:
CDC driver puts devices there, if you plug a second Dongle-E in it will get /dev/ttyACM1

darkxst, 2024-03-11:
> NVMEM area is not wiped when you flash normally.

you can however wipe it using this gbl.
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl

talormanda, 2024-03-11:
> > NVMEM area is not wiped when you flash normally.
> 
> you can however wipe it using this gbl. https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl

thanks. and as for the 115200. how did we know what this was?

darkxst, 2024-03-11:
baudrate is set by the firmware build, default on sonoff firmwares is always 115200

talormanda, 2024-03-11:
okay thanks. marking this closed for now.

closed 2024-03-11 by talormanda

Suffice, 2024-08-26:
I have this same problem. Not really looking forward to booting up Linux to fix this...