darkxst/silabs-firmware-builder #65: Silicon Labs MultiPan Flasher Add On always installs 4.3.1 although newer firmwares are available

hartmood, 2024-03-18:
Silicon Labs MultiPan Flasher Add-On Installed 0.2.1
Latest version available 4.4.1
Device: Sonoff Donge E
Core 2024.3.1
Supervisor 2024.03.0
Operating System 12.1
Frontend 20240307.0

The add-on always falls back to 4.3.1, although newer versions are available (4.4.0 and 4.4.1). Even manual updates to 4.4.1 are reset to 4.3.1.

darkxst, 2024-03-18:
Newer firmware are unstable. (I dont know if 4.4.1 has been tested yet for the addon side).

> Even manual updates to 4.4.1 are reset to 4.3.1.

SkyConnect would downgrade, Dongle-E will just fail to start with wrong firmware

hartmood, 2024-03-19:
That makes sense. Thanks for the clarification.

closed 2024-03-19 by hartmood