darkxst/silabs-firmware-builder #69: Some devices with firmware are disconnected | 4.3.1 -- | -- 7.3.1.0 |

Sr-psycho, 2024-03-22:
Guys, what firmware is suitable for this dongle from firmware_builds? Devices are falling off on this one (SDK Version | 4.3.1 -- | -- 7.3.1.0) , I want to roll back to the previous version 
PS I didn’t reconnect the devices, I just updated the firmware and connected the dongles.

darkxst, 2024-03-22:
What are you running? ZHA/Z2M/Multipan? and what devices are dropping of network.

7.3.1 has been pretty good by all accounts, and the recommended build at the moment.
7.2.x are problematic with Z2M but seemed ok with ZHA
7.1.x builds were also pretty good, but I dont have any in this repo.

sagittarius06, 2024-03-22:
I use ncp-uart-hw-v7.3.2.0-zbdonglee-115200.gbl with my Sonoff ZBDongle-E and it works great

Sr-psycho, 2024-03-22:
> Что вы бежите? ZHA/Z2M/Multipan? и какие устройства сбрасывают сеть.
> 
> 7.3.1 была довольно хороша по всем показателям, и рекомендуемая сборка на данный момент. 7.2.x проблематичны с Z2M, но с ZHA сборки 7.1.x тоже были довольно хорошими, но у меня их нет в этом репозитории.

I use Z2M. Such devices are disconnected from the network: ZBMINI, SNZB-02, TS011F_plug_1

darkxst, 2024-03-22:
You could try 7.4.1 with the new experimental 'ember' backend
https://github.com/Koenkk/zigbee2mqtt/discussions/21462

lanwin, 2024-04-22:
I have that running 7.4.1 running with ember and still some devices went offline after a while. Had Multipan running before with the same problem.

darkxst, 2024-04-22:
Is it specific devices that drop offline?
If your using ember you can try the 7.4.2 builds:
https://github.com/darkxst/silabs-firmware-builder/tree/4.4.2/firmware_builds

lanwin, 2024-04-22:
I also have a ZBDongle-E. In the mean time I tryied 7.3.2.0 and now I am on 7.4.2.0 

Interestingly now only my "Hue Bloom, Hue Iris, LivingColors Aura" devices are offline while the Hue Spots, and Aqura Sensors are shown as online. I am not sure why. But it can be that it happens since I tried z2m with latest-dev and 7.4.1. They stayed offline with 7.3.2.0 and now with 7.4.2.0 too.

LDprg, 2024-05-06:
Same issue here with zigbee donglee and 7.4.2 firmware and ember/ezsp driver of zigbee2mqtt.

woloss, 2024-06-10:
Same issue, devices disappearing from z2m UI sometimes after z2m restart. In my case it is LYWSD03MMC sensor so far, one gone 3 times and another one just once, rest fine.
7.4.3 dongle-e, zigbee2mqtt 1.37.0 & 1.38.0

Due to multiple updates at the same time: dongle `stock -> 7.4.3`, z2m `1.36.1 -> >1.37.0`, `ezsp -> ember`, hard to track what caused it.

LDprg, 2024-06-10:
@woloss you migth take a look at #22271. I did build up my network from scratch (as I described in a comment there). It fixed all my problems. I guess when changing from ezsp to ember you need to rebuild your network.

woloss, 2024-06-17:
@LDprg tried from scratch at another site, still saw some "Route" error right after pairing, after that, 0 noticeable issues, but didn't check logs tbh. However, for main site, probably gonna wait a bit, kinda painful to repair specific devices.