darkxst/silabs-firmware-builder #71: [REQUEST] EmberZNet Zigbee NCP and OpenThread RCP firmware for "Arduino Nano Matter" development board with USB is based on Silabs MGM240S module (containing Silicon Labs EFR32MG24 SoC)?

Hedda, 2024-03-26:
Spotted at [cnx-software.com tech blog](https://www.cnx-software.com/2024/03/26/arduino-nano-matter-board-specifications-and-price/) and thought this hardware could make a nice MGM240S based board for prototype testing, etc.

"**_Arduino Nano Matter_**" development board with USB is based on Silabs MGM240S module (containing Silicon Labs EFR32MG24 SoC) 

Any chance we could get EmberZNet Zigbee NCP and OpenThread RCP firmware builds for this MGM240S based board?

Guess it similar to MGM240P based Sparkfun Things Plus Matter board might require Silabs Gecko bootloader to be flashed first? -> https://github.com/darkxst/silabs-firmware-builder/pull/31

**Arduino Nano Matter**"  (SKU: ABX00112) is now available for $25(US)

* https://store-usa.arduino.cc/pages/nano-matter
  * https://docs.arduino.cc/hardware/nano-matter/#features
    * https://docs.arduino.cc/resources/pinouts/ABX00112-full-pinout.pdf
    * https://docs.arduino.cc/resources/datasheets/ABX00112-datasheet.pdf
    * https://docs.arduino.cc/resources/schematics/ABX00112-schematics.pdf
      * https://docs.arduino.cc/hardware/nano-matter/
      * https://docs.arduino.cc/tutorials/nano-matter/user-manual/

Key Specifications:
 
* Silabs MGM240S module MGM240SD22VNA (MGM240SD22VNA2R U5)
  * MCU Core = ARM Cortex-M33		
    * Core Frequency (MHz) = 78		
  * Flash (kB) = 1536
  * RAM (kB) = 256
  * Output Power Max (dBm) = 10
  * Silicon Labs Secure Vault™ = High
* Antenna = 2.4GHz Ceramic Chip Antenna surface mounted component on-board
* USB – USB-C port for power and data
  * USB bridge via Microchip ATSAM-D11-D14A-MUT (a.k.a. Atmel SAMD11 / Atmel SAM D11) chip on-board
* Debugging – JTAG/SWD debug port (accessible through the board’s test pads)
* Peripherals
  * Analog – 12-bit ADC (x19), up to 12-bit DAC (x2)
  *Digital – 22x GPIO pins (2x UART, 2x I2C, 2x SPI, 5x PWM)
  *External interrupts are available for all digital pins
* Misc
  * 2x push buttons (RST and user button)
  * RGB LED
  * Castellated pins for integrating the board into final solutions
* Power
  ' 5V via USB-C or header connector pins (VIN, IN5V)
  ' Operating Voltage: 3.3V
* Dimensions – 45 x 18 mm

MGM240S module:
 * https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-modules/device.mgm240sd22vna?tab=specs
* https://www.silabs.com/documents/public/data-sheets/mgm240s-datasheet.pdf

EFR32MG24 SoC:
* https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-socs
* https://www.silabs.com/documents/public/data-sheets/efr32mg24-datasheet.pdf

darkxst, 2024-03-26:
Yes should be able to add this, once I get my hands on one. I would expect it also needs bootloader installed first.

Hedda, 2024-03-26:
> * IEEE 802.15.4 Radio Transmit Output Power (dBm) = 10
> * Antenna = 2.4GHz Ceramic Chip Antenna surface mounted component on-board

When comparing the main hardware pros and cons with these two boards are the Transmit Output Power and Antennas used:

[Arduino Nano Matter](https://store-usa.arduino.cc/pages/nano-matter) pros and cons:
  * Arduino Nano Matter - Pros = Ceramic Chip Antenna (usually offers better performance than a PCB Trace antenna)
  * Arduino Nano Matter - Cons = Based on MGM240**S** which only offers +10 dBm TX Power (however that should work good)

[Sparkfun Things Plus Matter](https://www.sparkfun.com/products/20270) pros and cons:
  * Sparkfun Things Plus Matter - Pros = Based on MGM240**P** which offers +20 dBm TX Power
  * Sparkfun Things Plus Matter - Cons = PCB Trace antenna (usually offers worse performance than Ceramic Chip Antenna)

PS: If want to test external u.fl antenna see the official xG24-RB4187C (BRD4187C) and xG24-RB4186C (BRD4186C) radio boards.

* xG24-RB4187C (BRD4187C) based on the same SoC as MGM240P

https://www.silabs.com/development-tools/wireless/xg24-rb4187c-efr32xg24-wireless-gecko-radio-board

https://www.silabs.com/documents/public/reference-manuals/brd4187c-rm.pdf

* xG24-RB4186C (BRD4186C) based on the same SoC as MGM240S

https://www.silabs.com/development-tools/wireless/xg24-rb4186c-efr32xg24-wireless-gecko-radio-board

https://www.silabs.com/documents/public/reference-manuals/brd4186c-rm.pdf

Hedda, 2024-03-28:
> Also re power, for use as a Thread RCP TX power should be limited (probably < 10dBm) to avoid asynchronous power levels

Zigbee Coordinator also works better if limited to around 10 dBm, because otherwise it can transmit messages directly to Zigbee devices far away but those devices might have a harder time to transmit back messages.

darkxst, 2024-05-08:
These are now available:

https://store.arduino.cc/pages/nano-matter

Hedda, 2024-05-08:
€21 (21 Euro) as recommended retail price in the European Union for this MGM240S based Arduino Nano Matter is not bad at all.

https://store.arduino.cc/products/nano-matter

> I will note that the Sparkfun board has a full blown j-link debugger integrated. This should be superior to the debugger on the Arduino board, particularly if developing with Simplicity studio.

Compared to the MGM240P based Sparkfun Things Matter which cost $24,95 US in the United States but much more in the EU.

https://www.sparkfun.com/products/20270

https://opencircuit.shop/product/sparkfun-thing-plus-matter-mgm240p

https://www.electrokit.com/en/sparkfun-thing-plus-matter-mgm240p

(Not as high price for it in the United Kingdom though https://thepihut.com/products/sparkfun-thing-plus-matter-mgm240p)

darkxst, 2024-05-16:
I got this today, so far not looking good for firmware:

- Only seems to have CMSIS-DAP debug interface and no UART pass-through for the MGM240 chip
- Only works with Arduino IDE (or openOCD), doesnt seem to be supported by any of the Silabs tools Simplicity Studio or Commander. 

So it seems primarily intended for use with Silicon Labs Arduino Core + Arduino IDE only.