darkxst/silabs-firmware-builder #73: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0"

Outsidewall, 2024-04-06:
dongle-z using ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl
Latest dev branch of z2m

Getting hundreds of the following message from the logs

Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0"

Any ideas?

John

darkxst, 2024-04-11:
I have not seen this error before, you are probably best asking on Z2M channels.

One thing to check is that you dont have a panid conflict (with multiple networks).

fred-gb, 2024-04-23:
Hi,

I think, we are in same situation @Outsidewall 😟 

https://github.com/Koenkk/zigbee2mqtt/issues/22271

Outsidewall, 2024-04-26:
@fred-gb I have updated the firmware of the dongle-e to 7.4.2 and updated z2m to latest dev release, and so far it's looking good

fred-gb, 2024-04-26:
OK thanks @Outsidewall ! I will try asap.

fred-gb, 2024-04-26:
😞 

I just try now...

With `latest-dev`same errors

```
error 2024-04-26 17:59:27Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":251} messageTag=255]
warning 2024-04-26 17:59:29Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

😭

closed 2026-01-22 by github-actions[bot]