darkxst/silabs-firmware-builder #74: Flashing RCP Multipan Firmware on slzb-06m fails

Tobi9111, 2024-04-07:
Hi,
i've try to flash the Multipan Firmware on slzb-06m. I follow the steps on https://smlight.tech/manual/slzb-06/guide/thread-matter/thread-slzb-06m.html

I try flashing the Firmware multiple Times with two different PC Systems without success. When i try to flashing by follow the steps from the Link i can choose the Multipan Firmware and the flashing  window appears. After that it stucks at 0%. 
When i try flashing with skip Step 10 (Refresh the page (e.g., by pressing F5) and press the Zigbee Flash Mode button again) the Firmware flashing beginns normal until 100% and it seems that is complete successfully flashed,  but when i try to scan the zigbee Network after reset i get an error Massage. Furthermore in web OS under zigbee Firmware Update it shows me the Stock Firmware number and not the 4.3.1.  Is there any was to flash the zigbee 06m without the web flash builder? Or can somebody tell me what ist wrong? Thank you in advance and best regards!
Tobias

darkxst, 2024-04-07:
> firmware flashing beginns normal until 100% and it seems that is complete successfully flashed, but when i try to scan the zigbee network after reset, i get an error Massage.

How are you trying to scan the zigbee network? If it flashed successfully, your next step should be to setup the Multiprotol addon. See below blog post for details. slzb-06M in usb mode is equivalent to sonoff dongle.

https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware

> Furthermore in the web OS under "zigbee Firmware Update" it shows me the stock firmware number and not the 4.3.1.

This is expected, the version in the web UI only updates for firmware updates done from the web UI.

Finally to use RCP firmware over ethernet you need the latest core development 2.1.0-dev firmware and then you can set the baudrate on Z2M/ZHA page for 460800

Tobi9111, 2024-04-07:
Hi darkxst,
thank you very much for your fast answer and explain. I think i missunderstand that before.
But when i want to configure the Silikon labs multiprotocol addon i must choose a usb port. My slzb-06m ist connected via Network. I know the IP Adresse but it hasn't a usb port adress. I can't configure the addon without choose an USB port. 
I've buy this one because i want to use it away from my homeassistant-host via Network. Is it impossible with the multipan firmware? I ready only the rcp-thread Firmware is only for USB-use at this time.
Best regards 
Tobias

tl-sl, 2024-04-07:
You need to flash via USB, but you can use ethernet after that.
Multiprotocol addon has a hidden setting for network address, click the toggle at bottom of page "show unused settings"

you can then enter the address something like below (you must also select a USB port, but that can be any available option as this setting gets overridden by network address):
`tcp://slzb-06M.local:6638`

In the Web UI for SLZB-06M, you also need to go to the "Z2M and ZHA" page and set the baudrate to 460800

Tobi9111, 2024-04-08:
Hi,
thanks a lot. i've found this hide option now, but i get always thousendtimes error massages in the Log after starting the addon:
`[20:43:52] INFO: Starting socat TCP client for CPC daemon...
2024/04/08 20:43:52 socat[641] E tcp: wrong number of parameters (3 instead of 2)`

I try this by use:
tcp://slzb-06M.local:6638
tcp://192.168.xxx.xxx:6638
socket://192.168.xxx.xxx:6638
192.168.xxx.xxx:6638
After every change i've rebooted the addon.
The error Massage ist always the same. What do i wrong?
Best regards 
Tobias

darkxst, 2024-04-08:
it should be `hostname:port` format, so the last one you provided should work.

Tobi9111, 2024-04-09:
Thank you very much, i,'ve try it again with this format after deinstall the addon and rebooting ha. It seems that this work. No Error Massage anymore. 
But now i can't setting up ZHA by using "socket://core-silabs-multiprotocol:9999". I double checked all settings (incl. portspeed is set to 460800 everywere), but i get an unknown error).
[image]

`[19:25:11:316209] Info : Endpoint socket #12: Client disconnected. 1 connections
[19:25:11:316242] Info : Client disconnected
[19:25:12:317550] Info : New client connection using library v4.3.1.0
[19:25:12:322239] Info : Endpoint socket #12: Client connected. 2 connections
Reusing socket from previous instance.
[19:25:18] INFO: zigbeed ended with exit code 256 (signal 13)...
[19:25:18:463529] Info : Endpoint socket #12: Client disconnected. 1 connections
[19:25:18:463555] Info : Client disconnected
[19:25:19] INFO: Starting zigbeed...
[19:25:20:529089] Info : New client connection using library v4.3.1.0
[19:25:20:534441] Info : Endpoint socket #12: Client connected. 2 connections
Listening on port 9999 for connection...
Accepting connection.`
Furthermore i rebooted my ha Host and the slzb-06m, but it doesn't help.
Best regards 
Tobias

darkxst, 2024-04-09:
Logs from the multiprotocol addon look good, not sure why ZHA is not connecting, perhaps you could try with Zigbee2MQTT and see if that works?

Hedda, 2024-04-26:
> Also I will mention the focus at NabuCasa (Home ASsistant) is moving away from Multiprotocol and towards having a dedicated device for openThread. This is largely because of the many issues multipan can have. 

FYI, more information about that here -> https://github.com/darkxst/silabs-firmware-builder/discussions/41

darkxst, 2024-04-26:
OTBR Addon now supports network socket, I suggest you try that with your slzb-06m and the Openthread firmware.