darkxst/silabs-firmware-builder #75: Gecko 4.4.2

darkxst, 2024-04-11:
Initial builds for Gecko 4.4.2 are on these branches.

https://github.com/darkxst/silabs-firmware-builder/tree/4.4.2/firmware_builds

without flow control for Z2M Ember backend:
https://github.com/darkxst/silabs-firmware-builder/tree/ember-nohw/firmware_builds

If you test these builds, please report back results ;)

dd310, 2024-04-12:
Updated yesterday, no issues so far with zigbee and openthread on ZBDongle-E 👍

avbdr, 2024-04-25:
@darkxst is there a change log something? :) sorry for the noise

Hedda, 2024-04-26:
Changelog is available via links upstream release notes in the official Silicon Labs Gecko SDK (GSDK):

https://github.com/SiliconLabs/gecko_sdk

https://github.com/SiliconLabs/gecko_sdk/releases

closed 2026-01-20 by github-actions[bot]