darkxst/silabs-firmware-builder #76: Received ERROR from NCP, with code=RESET_ASSERT.

Outsidewall, 2024-04-14:
Zigbee2MQTT version  [1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1]
Coordinator type: EmberZNet
Coordinator revision: 7.4.1 [GA]

Here is the output from two instances of the error. It takes around 6/10 hours to happen. Wonder if it might be a memory leak in the adapter firmware code.
I am automatically now restarting zigbee2mqtt every couple of hours.

[image]

darkxst, 2024-04-15:
Can you try the latest build 7.4.2 from this branch
https://github.com/darkxst/silabs-firmware-builder/tree/4.4.2/firmware_builds

Outsidewall, 2024-04-16:
Hello,

Have updated to 7.4.2, looking good, no dongle crashes, been running for just over 24 Hrs. Also no memory leaks from zigbee2mqtt, so result.

[image]

I have got the following errors.. ROUTE_ERRORS, which I believe are normal for zigbee networks

Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31486".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "24606".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31486".
Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "3533".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "3533".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "59255".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "11181".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8632".
[NCP COUNTERS] MAC_RX_BROADCAST: 32462 | MAC_TX_BROADCAST: 807 | MAC_RX_UNICAST: 37100 | MAC_TX_UNICAST_SUCCESS: 30770 | MAC_TX_UNICAST_RETRY: 1844 | MAC_TX_UNICAST_FAILED: 244 | APS_DATA_RX_BROADCAST: 27 | APS_DATA_TX_BROADCAST: 26 | APS_DATA_RX_UNICAST: 26217 | APS_DATA_TX_UNICAST_SUCCESS: 23288 | APS_DATA_TX_UNICAST_RETRY: 0 | APS_DATA_TX_UNICAST_FAILED: 2 | ROUTE_DISCOVERY_INITIATED: 73 | NEIGHBOR_ADDED: 880 | NEIGHBOR_REMOVED: 880 | NEIGHBOR_STALE: 124 | JOIN_INDICATION: 7 | CHILD_REMOVED: 4 | ASH_OVERFLOW_ERROR: 0 | ASH_FRAMING_ERROR: 0 | ASH_OVERRUN_ERROR: 0 | NWK_FRAME_COUNTER_FAILURE: 991 | APS_FRAME_COUNTER_FAILURE: 0 | ASH_XOFF: 0 | APS_LINK_KEY_NOT_AUTHORIZED: 0 | NWK_DECRYPTION_FAILURE: 7251 | APS_DECRYPTION_FAILURE: 22 | ALLOCATE_PACKET_BUFFER_FAILURE: 0 | RELAYED_UNICAST: 0 | PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0 | PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0 | TYPE_NWK_RETRY_OVERFLOW: 0 | PHY_CCA_FAIL_COUNT: 177 | BROADCAST_TABLE_FULL: 0 | PTA_LO_PRI_REQUESTED: 0 | PTA_HI_PRI_REQUESTED: 0 | PTA_LO_PRI_DENIED: 0 | PTA_HI_PRI_DENIED: 0 | PTA_LO_PRI_TX_ABORTED: 0 | PTA_HI_PRI_TX_ABORTED: 0 | ADDRESS_CONFLICT_SENT: 0 | 
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8632".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8632".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8632".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "5731".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31486".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "33842".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8632".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31486".
Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "64009".

Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8632".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "5731".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "5731".

closed 2026-01-20 by github-actions[bot]