darkxst/silabs-firmware-builder #79: zb-gw04-1v2 open thread router crashes

francisp2, 2024-04-25:
Tried both the one offered on 
https://darkxst.github.io/silabs-firmware-builder/
and a manual download of ot-rcp-v2.4.2.0-zb-gw04-1v2-230400.gbl

After a while the OTBR add-on stops itself

[log.txt](https://github.com/darkxst/silabs-firmware-builder/files/15104638/log.txt)

darkxst, 2024-04-25:
I wouldnt really recommend the ZB-GW04 for thread. OTBR is not very resilient towards dropped frames and or frame errors, and will stop itself at the first sign of issues.

You would be better off with ZBDongle-E or SLZB-07 @460800 baud for Thread.

francisp2, 2024-04-25:
> You would be better off with ZBDongle-E or SLZB-07 @460800 baud for Thread.

I do have 2 nRF52840 dongles that work perfect, so I guess I have to replace the zb-gw04-1v2 with another one.

darkxst, 2024-04-25:
Both ZBDongle-E or SLZB-07 work perfect as well. zb-gw04 is just generally problematic across the board.

closed 2026-01-12 by francisp2