darkxst/silabs-firmware-builder #82: Flashing slzb-06m fails

fir3drag0n, 2024-04-28:
When I try to flash , I get the following error 

`00ncp-uart-hw-v7.4.2.0-slzb-06m-115200.gbl  [##----------------------------------]    7%  00:08:22
Error: Firmware image was rejected by the device. Ensure this is the correct image for this device.`

It is the gbl file for slzb-06m. What I need to change in order to flash it successfully?

darkxst, 2024-04-28:
Are you flashing over USB? it should just work.

Can you run universal-silabs-flasher with `-v` option and post logs.

fir3drag0n, 2024-04-28:
I am flashing over Ethernet

darkxst, 2024-04-28:
try USB (after selecting USB mode) there is less things to go wrong there.

fir3drag0n, 2024-04-28:
And how is the smlight zigbee radio firmware different from yours?

fir3drag0n, 2024-04-28:
I guess I have to set 

`--allow-cross-flashing`

in order to bypass the check?

darkxst, 2024-04-28:
smlight firmware is built against an older release of EmberNet(EZSP) than my builds. 7.4.2 is the very latest version available.

> --allow-cross-flashing

that is only if you are changing firmware type, ie zigbee to multiprotocol or thread.

I dont know why you get that error after it started though. Would need to see logs...

kalpik, 2024-05-15:
For me, this error was because the z2m container had restarted and connected back to the stick. After shutting down z2m, and then updating, it went fine.

closed 2026-01-19 by github-actions[bot]