darkxst/silabs-firmware-builder #83: Error after completing update

Jumperm, 2024-04-30:
I try to install 7.4.2.0 on the Sonoff dongle E

`2024-04-30 14:13:15 emscripten universal_silabs_flasher.xmodemcrc[42] DEBUG Firmware upload progress: 99.95%
2024-04-30 14:13:15 emscripten universal_silabs_flasher.xmodemcrc[42] DEBUG Sending data b'\x01\x1f\xe0sp_version": "7.3.1.0"\n}\n\xfc\x04\x04\xfc\x04\x00\x00\x00\x17 \x95\x9a\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xe8\xac' (attempt 0)
2024-04-30 14:13:15 emscripten universal_silabs_flasher.xmodemcrc[42] DEBUG Got response: b'\x06'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.xmodemcrc[42] DEBUG Firmware upload progress: 100.00%
2024-04-30 14:13:15 emscripten universal_silabs_flasher.xmodemcrc[42] DEBUG Sending data b'\x04' (attempt 0)
2024-04-30 14:13:15 emscripten universal_silabs_flasher.xmodemcrc[42] DEBUG Got response: b'\x06'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\r\nSerial upload complete\r\n\x00\r\nGecko Boo'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_UPLOAD_DONE: bytearray(b'\r\nSerial upload complete\r\n\x00\r\nGecko Boo')
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Boo')
2024-04-30 14:13:15 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'tloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Detected version string '1.12.00'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'2'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\r'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r')
2024-04-30 14:13:15 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\nSonoff v1.0.1\r\n\x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nSonoff v1.0.1\r\n\x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2024-04-30 14:13:15 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Detected version string '1.12.00'
2024-04-30 14:13:15 emscripten webserial_transport[42] DEBUG Closing serial port
2024-04-30 14:13:15 emscripten webserial_transport[42] DEBUG Closed serial port`

darkxst, 2024-04-30:
It looks to have completed, but then seems to be stuck in the bootloader, does it come good if you power cycle the dongle?

You can probe it after:
```
universal-silabs-flasher --device /dev/ttyACM0 probe
```

Jumperm, 2024-04-30:
`universal-silabs-flasher --device com7 probe
2024-04-30 14:32:07 Tobi-PC universal_silabs_flasher.flasher[9152] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-04-30 14:32:07 Tobi-PC universal_silabs_flasher.flasher[9152] WARNING No application can be launched
2024-04-30 14:32:07 Tobi-PC universal_silabs_flasher.flasher[9152] INFO Detected bootloader version '1.12.0'
2024-04-30 14:32:07 Tobi-PC universal_silabs_flasher.flasher[9152] INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)`

darkxst, 2024-04-30:
So it hasnt flashed, but there was no error flashing....

Just to check you downloaded the raw firmware file from GH?
https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl

Jumperm, 2024-04-30:
What does this mean?

"WARNING No application can be launched"

Jumperm, 2024-04-30:
`PS C:\Users\Tobi\Downloads\universal-silabs-flasher-0.0.15> universal-silabs-flasher --device COM7 flash --firmware ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl --sonoff-reset
2024-04-30 14:52:20 Tobi-PC universal_silabs_flasher.flash[7644] INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.2', ezsp_version='7.4.2.0', ot_rcp_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-04-30 14:52:20 Tobi-PC universal_silabs_flasher.flasher[7644] INFO Triggering Sonoff bootloader
2024-04-30 14:52:20 Tobi-PC universal_silabs_flasher.flasher[7644] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-04-30 14:52:21 Tobi-PC universal_silabs_flasher.flasher[7644] WARNING No application can be launched
2024-04-30 14:52:21 Tobi-PC universal_silabs_flasher.flasher[7644] INFO Detected bootloader version '1.12.0'
2024-04-30 14:52:21 Tobi-PC universal_silabs_flasher.flasher[7644] INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)
ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl  [####################################]  100%
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Python311\Scripts\universal-silabs-flasher.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python311\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\universal_silabs_flasher\flash.py", line 36, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\base_events.py", line 650, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\universal_silabs_flasher\flash.py", line 407, in flash
    await flasher.flash_firmware(
  File "C:\Program Files\Python311\Lib\contextlib.py", line 222, in __aexit__
    await self.gen.athrow(typ, value, traceback)
  File "C:\Program Files\Python311\Lib\site-packages\universal_silabs_flasher\common.py", line 181, in connect_protocol
    yield protocol
  File "C:\Program Files\Python311\Lib\site-packages\universal_silabs_flasher\flasher.py", line 326, in flash_firmware
    await gecko.run_firmware()
  File "C:\Program Files\Python311\Lib\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 97, in run_firmware
    raise NoFirmwareError("No firmware exists on the device")
universal_silabs_flasher.gecko_bootloader.NoFirmwareError: No firmware exists on the device
PS C:\Users\Tobi\Downloads\universal-silabs-flasher-0.0.15> universal-silabs-flasher --device com7 probe
2024-04-30 14:54:35 Tobi-PC universal_silabs_flasher.flasher[11928] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-04-30 14:54:35 Tobi-PC universal_silabs_flasher.flasher[11928] WARNING No application can be launched
2024-04-30 14:54:35 Tobi-PC universal_silabs_flasher.flasher[11928] INFO Detected bootloader version '1.12.0'
2024-04-30 14:54:35 Tobi-PC universal_silabs_flasher.flasher[11928] INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)`

darkxst, 2024-04-30:
> universal-silabs-flasher-0.0.15

This is an old version, try upgrade to 0.0.19, there was a bug fix around 0.0.16 that might fix this.

Jumperm, 2024-04-30:
With the old and the new version it don't know the command `--bootloader-reset`

darkxst, 2024-04-30:
then your still running the old version. `--bootloader-reset` was added in 0.0.17 or 18.

Jumperm, 2024-04-30:
Didn' work either...

darkxst, 2024-04-30:
I not sure what is going on then, 7.4.2 firmware has been validated by multiple users at this point. So something seems up with your universal-silabs flasher.

Jumperm, 2024-04-30:
You are right! I downloaded the 0.0.19 zip but i didn't generate a new *exe from Python. I didn't manage it right now. Do you know how i rebuild the new version under windows?

darkxst, 2024-05-01:
I would think you can use `pip install universal-silabs-flasher` on windows also? but I really not familiar with windows usage, I dont use it myself.

Jumperm, 2024-05-01:
Seems like I damage the Bootloader or something...
`universal-silabs-flasher --device /dev/cu.wchusbserial51850245491 --bootloader-reset sonoff flash --firmware ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl 
2024-05-01 21:08:36.546 Tobias-MBAir.local universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.2', ezsp_version='7.4.2.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-05-01 21:08:36.546 Tobias-MBAir.local universal_silabs_flasher.flasher INFO Triggering sonoff bootloader
2024-05-01 21:08:37.162 Tobias-MBAir.local universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-05-01 21:08:37.208 Tobias-MBAir.local universal_silabs_flasher.flasher WARNING No application can be launched
2024-05-01 21:08:37.209 Tobias-MBAir.local universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2024-05-01 21:08:37.209 Tobias-MBAir.local universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)
ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl  [####################################]  100%          
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/universal_silabs_flasher/flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/universal_silabs_flasher/flash.py", line 423, in flash
    await flasher.flash_firmware(
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/universal_silabs_flasher/flasher.py", line 299, in flash_firmware
    await gecko.run_firmware()
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/universal_silabs_flasher/gecko_bootloader.py", line 101, in run_firmware
    raise NoFirmwareError("No firmware exists on the device")
universal_silabs_flasher.gecko_bootloader.NoFirmwareError: No firmware exists on the device
`

darkxst, 2024-05-01:
I dont think the bootloader is damaged, but not sure why its not working successfully. Does the same thing happen with 7.3.x firmware?

Otherwise, You could try manual method as documented by sonoff to flash manually using xmodem.

Jumperm, 2024-05-02:
Yes, I tried the 7.3.1.0 and Thread and Multi FW via web. Always the same result.
 
Here is the output:
`CCCCCCCCCCCCCCCCCCCC
Sonoff v1.0.1

Gecko Bootloader v1.12.00
1. upload gbl
2. run
3. ebl info
BL > 
begin upload
CCCCCCCCCCCCC
Starting xmodem transfer.  Press Ctrl+C to cancel.
Transferring /Users/tobiasmaroschek/Downloads/ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl...
  100%     233 KB       8 KB/sec    00:00:27       0 Errors  

Serial upload complete

Gecko Bootloader v1.12.00
1. upload gbl
2. run
3. ebl info
BL > `

Seems like it always fall back into bootloader mode.

darkxst, 2024-05-03:
One more thing you could try is to wipe NVME by flashing:
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl

The after that try flash again my firmware.

Jumperm, 2024-05-03:
> If you previously had a 230400 firmware installed on dongle you need additional flags to univeral-silabs-flasher for the flash to work.

What are these flags. Maybe they help me!

Jumperm, 2024-05-03:
```
Sonoff v1.0.1

Gecko Bootloader v1.12.00
1. upload gbl
2. run
3. ebl info
BL > 
begin upload
CCCCCCCCCCCCCCCCCCC
Starting xmodem transfer.  Press Ctrl+C to cancel.
Transferring /Users/tobiasmaroschek/Downloads/ncp-uart-sw_7.3.1.0_115200.gbl...
  100%     227 KB       7 KB/sec    00:00:29       0 Errors  

Serial upload complete

Gecko Bootloader v1.12.00
1. upload gbl
2. run
3. ebl info
BL > 
begin upload
CCCCCCCCCCC
Starting xmodem transfer.  Press Ctrl+C to cancel.
Transferring /Users/tobiasmaroschek/Downloads/nvm3_initfile (1).gbl...
  100%      32 KB       8 KB/sec    00:00:04       0 Errors  

Serial upload complete

Gecko Bootloader v1.12.00
1. upload gbl
2. run
3. ebl info
BL > ```

Jumperm, 2024-05-03:
[image]

sstidl, 2024-05-03:
> <img alt="Bildschirmfoto 2024-05-03 um 18 04 25" width="286" src="https://private-user-images.githubusercontent.com/6853816/327815030-93f264e7-5be4-4110-8217-e070e37da70d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ3NTg3NDAsIm5iZiI6MTcxNDc1ODQ0MCwicGF0aCI6Ii82ODUzODE2LzMyNzgxNTAzMC05M2YyNjRlNy01YmU0LTQxMTAtODIxNy1lMDcwZTM3ZGE3MGQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDNUMTc0NzIwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjZhY2U0NGNhNDRmZWRkZjg5NTFjOTRiZWE0ZmViNjI4ZTZmN2QxOWJlNjk2YmVlNWRiZjZlODNlM2M2YWY5NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.hiXuhMnkcJ2PQIaB_7DFcuGj_W7tdKcw7llAHC6OOTM">

Why are you posting this?

Jumperm, 2024-05-03:
Sorry for irritating you.

This is the Webview after just plug the Dongle in and Connect it in chrome.
I want to show, that my dongle directly go into bootloader mode without trigger it via hardwarebutton or serial trigger (1).

darkxst, 2024-05-04:
Its triggering bootloader due to issue with NCP app.

If nvm3_init didnt help, not sure what else can try. I can't reproduce this issue on my ZBDongle-E's.

Jumperm, 2024-05-04:
> Its triggering bootloader due to issue with NCP app.
> 
> If nvm3_init didnt help, not sure what else can try. I can't reproduce this issue on my ZBDongle-E's.

Big thanks for your intensive help. I also think that's a hardware problem and not your gbl.

sstidl, 2024-05-04:
@Jumperm try to follow the steps I wrote here: #84

Jumperm, 2024-05-05:
> @Jumperm try to follow the steps I wrote here: #84

bellows also shows errors.

Jumperm, 2024-05-06:
With: `universal-silabs-flasher --device com7 --ezsp-baudrate 230400 --probe-method ezsp probe`

i got: `2024-05-06 13:08:37.222 Tobias-PC bellows.uart ERROR CRC error in frame b'061886061878f8860018981e98800618860098e606069efe9e06067800987818069ee68098801818860018f8667e' (b'f866' != b'c689')`

darkxst, 2024-05-06:
this would be normal if a 115200 firmware is installed and running (however if the installed firmware is actually 230400 then may be hardware issue or bad usb etc)

closed 2026-01-15 by github-actions[bot]