darkxst/silabs-firmware-builder #84: ZBDongle-E SONOFF Zigbee 3.0 USB Dongle Plus V2 not booting NCP-App-Firmware

sstidl, 2024-05-02:
Hi!

I have a 
`Bus 001 Device 008: ID 1a86:55d4 QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2`
aka zbdonglee

I tried to flash any of the following firmwares:
| md5|filename|
|--------|----|
|8446db9f6c0e5755b9eb0f83b0fdbdc5  |ncp-uart-hw-v7.3.1.0-zbdonglee-115200.gbl|
|4a6832e40b861cf7fb877ccb50b7f588  |ncp-uart-hw-v7.4.0.0-zbdonglee-115200.gbl|
|ba133078833891e1da160f63c32ff081 | ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl|
|49cfc83a2e70f99614de8ec55fde3351  | ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl|

I used the command:
`universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware ./ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl`

It always reported success, but the stick got stuck in bootloader.

I also tried the webflasher which reported success, but same effect. Not booting the EZSP App.

when I flash the firmware from 
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.3.1.0_115200.gbl
it works.

I dont know what's the difference. xsp1989 has no firmware for 7.4.x so I wanted to flash the one over here...

sstidl, 2024-05-02:
another reason to reflash 7.4 is https://github.com/Koenkk/zigbee2mqtt/discussions/21462 
they deprecated `ezsp` driver for the new `ember` one which needs EZSP13:
> To keep things simple, this driver currently only supports EmberZNet firmware 7.4.x (EZSP 13), older versions will not be supported. Initially, multi-protocol will also not be supported as this is not considered stable yet.

darkxst, 2024-05-03:
same issue in #83 but I really not sure what is causing this, I have not been able to reproduce myself.

One thing you could try is to wipe NVME by flashing:
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl

The after that try flash again my firmware.

EnzovdWetering, 2024-05-03:
I also had issues flashing 7.4.x. I did the following (in this order) and was successful for some reason...

1. ncp-uart-hw-v7.4.2.0-zbdonglee-230400.gbl via web tool (failed)
2.ncp-uart-hw-v7.3.1.0-zbdonglee-none-115200.gbl via universal-silabs-flasher (successful, but I really need 7.4 so moving on..)
3. ncp-uart-hw-v7.4.1.0-zbdonglee-230400.gbl via universal-silabs-flasher (failed)
4. ncp-uart-hw-v7.3.1.0-zbdonglee-none-115200.gbl (success)
5. ncp-uart-hw-v7.4.0.0-zbdonglee-115200.gbl (success)

I guess I was lucky.. Not sure where things go wrong.

darkxst, 2024-05-03:
If you previously had a 230400 firmware installed on dongle you need additional flags to univeral-silabs-flasher for the flash to work. However apart from that, there is no difference between any images you listed in terms of flashing that would cause some to work and others to fail.

On a side note, generally recommend NCP 115200 or Thread 460800 buadrates. There is no evidence at this point that the 230400 baud really improves anything.

Jumperm, 2024-05-03:
> If you previously had a 230400 firmware installed on dongle you need additional flags to univeral-silabs-flasher for the flash to work. However apart from that, there is no difference between any [...]

What are these flags. Maybe they help me!

Jumperm, 2024-05-03:
> > What are these flags. Maybe they help me!
> 
> ```
> --bootloader-reset sonoff
> ```
> 
> Is should be enough, but you already tried that?
Yes, its just trigger the Bootmode. And that's my problem. the device ist immediately in Bootmode. It never start the NCP App.

> The following is required to probe a running ncp 230400 firmware:
> 
> ```
> --ezsp-baudrate 230400
> ```
As mentions. No NCP is running.
> 
> Did you try flash `nvm3_init.gbl`?
Yes, via web, Terminal, shell and SecureCRT.

I'm now ordering a new Sonoff or  SMLIGHT SLZB-06M...

darkxst, 2024-05-04:
> Yes, via web, Terminal, shell and SecureCRT.

Once would have been sufficient, then flash NCP again?

sstidl, 2024-05-04:
Success!!!

Thank you so much for the hint of erasing the nvm!

I did the following:
Started with a running firmware from https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.3.1.0_115200.gbl

stopped all services using the stick (zigbee2mqtt in my case)

did a backup with bellows (pip install bellows)
`bellows -d /dev/ttyACM0 -b 115200 backup | tee bellow.backup.json`

flashed the nvm3 cleaner
`universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware nvm3_initfile.gbl`
checked if running with bellows
`bellows -d /dev/ttyACM0 -b 115200 info`
now you should see there is no joined network (because we erased it)

flashed the 7.4x version from this repo
`universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl`

again check if stick is booting correctly with
`bellows -d /dev/ttyACM0 -b 115200 info`
now it should look the same as above but with new version, but still no joined network

now restore the config:
`bellows -d /dev/ttyACM0 -b 115200 restore -B bellow.backup.json`

check again
`bellows -d /dev/ttyACM0 -b 115200 info`

now i am back in my network (joined)

start zigbee2mqtt and cross fingers

done :rocket:

closed 2024-05-04 by sstidl

darkxst, 2024-05-05:
> Success!!!
> 
> Thank you so much for the hint of erasing the nvm!

Good to hear its now working, I think this has happened before for users migrating from xsp1989 firmwares.