darkxst/silabs-firmware-builder #86: Gecko 4.4.3

darkxst, 2024-05-04:
Initial builds for Gecko 4.4.3 are here, release notes are broken currently, but this appears to be only a very minor bug fix release.

https://github.com/darkxst/silabs-firmware-builder/tree/4.4.3

mamrai1, 2024-05-05:
Flashed! All good..

derwolff, 2024-05-07:
Is this a Ember backend build? I noticed in version 4.4.2 different folders.

darkxst, 2024-05-07:
Yes these will work for ember.

There is also a separate branch that disables hardware flow control on some dongles (slzb-07 etc), not sure what the current status is on these in ember. But only use these if you are getting crashes on startup.
https://github.com/darkxst/silabs-firmware-builder/tree/ember-nohw

mamrai1, 2024-06-19:
@darkxst are the firmwares built for ember config updates like 7.4.1?
Why builds are not in main branch?