darkxst/silabs-firmware-builder #87: Aqara battery always 100%

Gobolin80, 2024-05-04:
HI,
I have been using the Sonoff ZBDongle-E by ITead (now with this firmware https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl) on Home Assistant via the Zigbee2Mqtt addon, for a year.
Everything works correctly except for the battery percentage, which for all 10 Aqara devices is fixed at 100%, even when the battery is completely empty.
I don't know if it's a firmware bug, but I'll point it out.

This is what I see for a **fully discharged** device:

{
    "battery": 100,
    "device_temperature": 25,
    "illuminance": 0,
    "illuminance_lux": 0,
    "linkquality": 216,
    "occupancy": false,
    "power_outage_count": 38,
    "voltage": 3115
}

Thanks for your work

Gobolin80, 2024-05-04:
I was afraid the problem might be something like this. 
Thanks for the feedback

darkxst, 2024-05-05:
ZBDongle-P does fair a little better with Aqara sensors compared to ZBDongle-E