darkxst/silabs-firmware-builder #88: Error after Update to ncp-uart-hw-v7.4.2.0-zbdonglee-230400.gbl

andre1808, 2024-05-04:
Hi there.

I'm not really an expert in this and just tried updating my ZBDongle-E to a 7.4+ version to be able to use it with HomeAssistant and Zigbee2MQTT.
I was previously on MultiPAN (which I didn't really need). I've downloaded the [ncp-uart-hw-v7.4.2.0-zbdonglee-230400.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.2.0-zbdonglee-230400.gbl) which after reading [this thread ](https://github.com/darkxst/silabs-firmware-builder/issues/83)may have been a fault and I should probably have picked the 115200.

Anyway, the flashing was working fine according to the webflasher. However, the device never got back.
So I've installed python, pip and finally the universal-silabs-flasher (under windows) and seem to have the flasher working.
I've tried now to "probe" as well as flash a new firmware. Neither work and throw the same error:
`PS C:\users\Andre\Downloads> universal-silabs-flasher --device COM4 flash --firmware ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl --sonoff-reset
2024-05-04 18:10:41.752 Andre-PC universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.2', ezsp_version='7.4.2.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-05-04 18:10:41.752 Andre-PC universal_silabs_flasher.flash INFO The '--sonoff-reset' flag is deprecated. Use '--bootloader-reset' instead, see --help for details.
2024-05-04 18:10:41.752 Andre-PC universal_silabs_flasher.flasher INFO Triggering sonoff bootloader
2024-05-04 18:10:42.356 Andre-PC universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-05-04 18:10:42.465 Andre-PC universal_silabs_flasher.flasher INFO Launched application from bootloader
2024-05-04 18:10:43.509 Andre-PC universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2024-05-04 18:10:43.510 Andre-PC universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-05-04 18:10:48.521 Andre-PC universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\Scripts\universal-silabs-flasher.exe\__main__.py", line 7, in <module>
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\universal_silabs_flasher\flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\asyncio\base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\universal_silabs_flasher\flash.py", line 341, in flash
    await flasher.probe_app_type()
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\universal_silabs_flasher\flasher.py", line 206, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\universal_silabs_flasher\flasher.py", line 135, in probe_cpc
    async with self._connect_cpc(baudrate) as cpc:
  File "C:\Program Files\Python312\Lib\contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\universal_silabs_flasher\common.py", line 181, in connect_protocol
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\zigpy\serial.py", line 48, in create_serial_connection
    transport, protocol = await pyserial_asyncio.create_serial_connection(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\serial_asyncio\__init__.py", line 448, in create_serial_connection
    serial_instance = serial.serial_for_url(*args, **kwargs)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Users\Andre\AppData\Roaming\Python\Python312\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)`

I'm running Powershell as Admin, so the Access issue shouldn't come from Windows (I suppose?).
Does anyone have an idea how to resolve this issue? I'd really appreciate any help!

Thank you and kind regards
André

mamrai1, 2024-05-04:
You need to specify speed 230400 when flashing

andre1808, 2024-05-04:
> You need to specify speed 230400 when flashing

Sorry can you explain a little bit further? I’ve tried flashing with the parameter --bootloader-baudrate 115200. so I guess I’m using 230400 instead?

Nerivec, 2024-05-04:
You might want to stay with 115200 firmware for Dongle-E. It's the most tested one.

andre1808, 2024-05-04:
Thank you. I will try tomorrow and let you know the result

darkxst, 2024-05-05:
> Yes, use parameter --bootloader-baudrate 230400 and flash the 115200 firmware

This should be
```
--ezsp-baudrate 230400
```

darkxst, 2024-05-05:
web flasher will now probe 230400.

andre1808, 2024-05-05:
> web flasher will now probe 230400.

Thank you so much. I’ve used the webflasher and it worked immediately!
Now I just need to make it work with HA again - but that’s not your business I guess :)

thanks again for your great support (to all :))

darkxst, 2024-05-19:
Good to hear its sorted now, will close this issue.

closed 2024-05-19 by darkxst