darkxst/silabs-firmware-builder #89: Firmware ncp-uart-hw-v7.4.2.0-zb-gw04-1v2-230400.gbl bricked ZB-GW04 v1.2

uwen70, 2024-05-04:
Today I bricked my ZB-GW04 v1.2 with firmware **ncp-uart-hw-v7.4.2.0-zb-gw04-1v2-230400.gbl**. The update ran successfully via the Web Flasher (https://darkxst.github.io/silabs-firmware-builder/). Afterwards the stick is not recognized by the Web Flasher, universal-silabs-flasher or z2m.

Flash Logfile: 
[silabs_flasher.log](https://github.com/darkxst/silabs-firmware-builder/files/15211130/silabs_flasher.1.log)

Bricked Log: 
[silabs_flasher.log](https://github.com/darkxst/silabs-firmware-builder/files/15211135/silabs_flasher.3.log)

darkxst, 2024-05-05:
Its not bricked, but the web flasher doesnt recognise 230400 baud. If you configure it as 230400 in Z2m it will work

If you want to go back to 115200 then:

```
universal-silabs-flasher --device </dev/ttyACM0 or COM#> --ezsp-baudrate 230400 flash --firmware <file.gbl>
```

darkxst, 2024-05-05:
Webflasher is now configured to also probe ezsp 230400, given the number of issues this causes. `universal-silabs-flasher` will still require specifying baudrate for 230400.

closed 2024-05-05 by darkxst