darkxst/silabs-firmware-builder #90: [7.4.x] SLZB-06M inconsistencies when setting config IDs on startup

Nerivec, 2024-05-05:
Reporting this since I've seen a few logs now, from different users, that show this. (I don't have the device myself, so I can't run any test.)
Seems something is going wrong, that first one is the first being set during startup sequence, and is set to the same as the [firmware default](https://github.com/darkxst/silabs-firmware-builder/blob/09f81c6252758f4346f03d36d3f565ecf8b82cd8/manifests/defaults.json#L12). Not sure why it would return as out of memory at that point?

```log
[2024-05-05 12:07:00] warning: zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-05-05 12:07:00] warning: zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-05-05 12:07:00] warning: zh:ember: [EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-05-05 12:07:00] warning: zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
[2024-05-05 12:07:00] warning: zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```

_I'm guessing the `ERROR_INVALID_VALUE` for `SOURCE_ROUTE_TABLE_SIZE` is a bug in silabs firmware, reporting the wrong status, and is likely really `ERROR_OUT_OF_MEMORY` too._

When the device is a state where it does this during startup (it doesn't always do it seems like), it then misbehaves in various ways.
Any ideas?

darkxst, 2024-05-05:
Its strange that is only happening with SLZB-06M, and for settings that are already set! however I can easily reproduce here on 7.4.1.

Also tried with 7.4.2 I only get a single error (which is actually the only value that is changed compared to firmware):
```
[2024-05-06 00:18:34] info:     zh:ember:ezsp: ======== EZSP started ========
[2024-05-06 00:18:34] warning:     zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```

Now I have [updated settings](https://github.com/darkxst/silabs-firmware-builder/commit/5608ea321593ace2d3667b53c652a5ba5cc8f418) to match ZBDongle-E

https://github.com/darkxst/silabs-firmware-builder/raw/4.4.3/firmware_builds/slzb-06m/ncp-uart-hw-v7.4.3.0-slzb-06m-115200.gbl

(7.4.3 doesnt appear to have any changes in Zigbee firmware compared to 7.4.2)

Let me know if that resolves the issues for people.

itwtds, 2024-05-10:
> Its strange that is only happening with SLZB-06M, and for settings that are already set! however I can easily reproduce here on 7.4.1.
>
> Also tried with 7.4.2 I only get a single error (which is [...]

> ZBDongle-E

> 
> https://github.com/darkxst/silabs-firmware-builder/raw/4.4.3/firmware_builds/slzb-06m/ncp-uart-hw-v7.4.3.0-slzb-06m-115200.gbl
> 
> (7.4.3 doesnt appear to have any changes in Zigbee firmware compared to 7.4.2)
> 
> Let me know if that resolves the issues for people.

This is happening on the Skyconnect as well on 7.4.x (all tested).

`[2024-05-10 21:35:31] info: 	zh:ember:ezsp: ======== EZSP started ========

[2024-05-10 21:35:32] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-05-10 21:35:32] info: 	zh:ember: [STACK STATUS] Network up.
[2024-05-10 21:35:32] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-05-10 21:35:32] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-05-10 21:35:32] info: 	z2m: zigbee-herdsman started (resumed)
[2024-05-10 21:35:32] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":3,"revision":"7.4.3 [GA]","special":0,"type":170},"type":"EmberZNet"}'`

Thanks

GigiPompieru, 2024-06-20:
Hi, just updated today and saw this in the logs. any update on fixing it?

Nerivec, 2024-07-07:
Seems Silabs made some changes to their firmware which made a lot of values impossible to increase, only decrease. In 8.0.x, it appears, most can also no longer be set to the same value as the firmware's built-in.
Z2M `ember` will no longer be setting these values in future releases to avoid issues and will instead rely on values built into the firmware (easy enough to customize with the new build system).

Remains the issues of the SLZB-06M that sets these values just fine in 7.4.x except sometimes after an NCP reset, when it fails on multiple ones (the ones in first post here). Seems there is an actual issue in this specific scenario (some improper state after reset? might be linked to stress prior to reset).

darkxst, 2024-07-08:
> Seems Silabs made some changes to their firmware which made a lot of values impossible to increase
Yes they have incrementally been doing this to variables through out the 7.x releases.

> (easy enough to customize with the new build system).
Was just as easy with my old builder :)

>  SLZB-06M that sets these values just fine in 7.4.x except sometimes after an NCP reset

Is this still happening with updated config, that sets all values in OP as default? Which incidentally was the default config on my ZBDongle-E builds for a long time.

Nerivec, 2024-07-08:
It is, but on restart, after an error triggered said restart (seems to be the pattern anyway); as described in first post. Might be more about memory, than the configs themselves.
Though now we won't see it anymore, since these will no longer be set by the app (except `APS_UNICAST_MESSAGE_COUNT`). That's not to say it might not still be causing troubles because of the underlaying issue. As mentioned in first post, it's not just the configs failing when it gets in that state, it's general instability after start.

closed 2024-11-21 by Nerivec