darkxst/silabs-firmware-builder #94: unable to update a ZB-GW04 1.2 fromt 7.3 to 7.4.1

Claudio1L, 2024-05-19:
HI

after updating to 7.3 some days ago, I'm trying to update it to 7.4.2 to enable ember driver support.
I'm not able to update it. I tried with chrome on Macos, and chrome and edge on windows.

it sees the device:
[image]

but after some time
[image]

it fails:

[image]

the debug log is the following:
[silabs_flasher (4).log](https://github.com/darkxst/silabs-firmware-builder/files/15370426/silabs_flasher.4.log)

The stick is working with Z2M, but I'm not able to upgrade anymore

darkxst, 2024-05-20:
I dont know much about Mac, but on windows you need to install drivers for CH340.

However your probably better just flashing using your linux VM and universal-silabs-flasher

```
universal-silabs-flasher --device /dev/ttyUSB0 flash --firmware <file.gbl>
```
make sure you user is added to `dialout` group and you dont even need to use `root` ;)

Claudio1L, 2024-05-20:
> Thanks TIM
>
> I tried, but still no way
>
> 2024-05-20 17:59:58.843 pve asyncio DEBUG Using selector: EpollSelector
> Error: 'ncp-uart-hw-v7.4.2.0-zb-gw04-1v2-115200.gbl' does not appear to be a valid firmware image: ValueError('Unknown firmware image type')
>
> I see could be used an option --allow-cross-flashing, but not clear where to position it. Tried any position in the flash command but seems not working

Ok, solved.

Downloading the firmware as raw did the job

Thanks

Claudio