darkxst/silabs-firmware-builder #95: Webflasher connection failed with SLZB-06M

liouma, 2024-05-27:
I use a SLZB-06M with latest original coordinator firmware with Zigbee only.
As I'm trying to run at the same time ZHA and Z2M I'm trying to flash ncp-uart-hw-v7.4.1.0-slzb-06m-115200.gbl firmware that is recommended on this page: https://www.zigbee2mqtt.io/guide/adapters/emberznet.html

I used this web flasher: https://darkxst.github.io/silabs-firmware-builder/

After "trying to connect" for a while I get the following error:
_Connection failed
The running firmware could not be detected.

Make sure the USB port works and if you are using a USB extension cable, make sure the cable can transfer data. Unplug the SMLIGHT SLZB-06M and plug it back in to reset and try again._

Debug log is attached.

Thanks in advance for your help :)

[
[silabs_flasher (1).log](https://github.com/darkxst/silabs-firmware-builder/files/15450946/silabs_flasher.1.log)
](url)

darkxst, 2024-05-27:
> I'm trying to flash ncp-uart-hw-v7.4.1.0-slzb-06m-115200.gbl firmware

This is the exact same firmware as latest SMLIGHT ota firmware via slzb web ui (that is marked as for ember testing).

I dont know why the webflasher is failing but you can try pressing the "Zigbee flash mode" button in the slzb web ui first before running web flasher.

liouma, 2024-05-27:
Thanks a lot for your very fast answer :)
I already flashed the following version in OTA:
- zigbee : 20240510
- ESP: v2.2.2.dev

So this is the same as ncp-uart-hw-v7.4.1.0-slzb-06m-115200.gbl ?

I hoped this firmware would solve my problem but I'll have to find the solution somewhere else...
Thank you

darkxst, 2024-05-27:
Yes it is the exact same build.

What is the problem you are having?

There is a newer firmware available for testing that includes some bug fixes over 7.4.1 here:
https://github.com/darkxst/silabs-firmware-builder/raw/4.4.3/firmware_builds/slzb-06m/ncp-uart-hw-v7.4.3.0-slzb-06m-115200.gbl

liouma, 2024-05-27:
Thanks for the new firmware but as I can't flash it on OTA I would need to solve my webflasher issue...

The problem I'm having is the following:
I use ZHA with the SLZB-06M and my goal is to to what is described in this article:
https://smarthomescene.com/guides/how-to-use-zigbee2mqtt-and-zha-with-a-single-coordinator/

I configured Z2M using the same pan, pan_id, network_key, ieee and channel.

Everything work fine untill ZHA and Z2M are run at the same time.
I tried with z2m as a HA integration and on a separate VM. I tried with ezsp and ember. With EZSP, z2m crashes. When z2m uses ember, it's zha that crashes.

In case you have an idea, this is the Z2M log when ZHA is already running and I start Z2M (ezsp version):
`[2024-05-22 21:59:40] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":6} after 4000ms
[2024-05-22 21:59:40] error: 	zh:ezsp:uart: -!- break waiting (6)
[2024-05-22 21:59:40] error: 	zh:ezsp:uart: Can't send DATA frame (5,7,0): 3d000164000efdff0100
[2024-05-22 22:00:06] error: 	zh:ezsp:uart: Unrecognized frame FrameID 258
[2024-05-22 22:00:06] error: 	zh:ezsp:uart: Error while parsing to NpiFrame 'Error: Unrecognized frame FrameID 258
    at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:145:19)
    at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
    at SerialDriver.emit (node:events:517:28)
    at SerialDriver.handleDATA (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:250:14)
    at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:174:22)
    at Parser.emit (node:events:517:28)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)'`

And when ZHA is already started if I start z2m in ember mode:
`
[image]
`

On its side ZHA says:
[image]
[image]

I tried with SLZB-06M original firmware then I upgraded the firmware as adviced in z2m adapters page. 
And here I am :)

darkxst, 2024-05-27:
> Everything work fine untill ZHA and Z2M are run at the same time.

You can't do this, it won't work and even if did its a bad idea (and yes I am aware there is content on the internet suggesting it does work). 

you can only connect a single instance to each coordinator. Thus you would need a second coordinator (running a separate zigbee network) to run both ZHA and Z2M at the same time.

liouma, 2024-05-27:
Do you mean I can't do this with this chipset?

I'm in touch with persons on other forums who are running it sucessfully, and as you say there is also an article and some videos.
I will try with a SLZB-06 or a UZG-01 adapter.

I have a large house with a mesh network including routers... 
Some devices work better with Z2M others with ZHA. Using 2 coordinators would mean doubling all repeaters...

That's why using boths softwares would be a perfect solution for me. Zha would manage 90% of the devices. Z2M would manage the other 10%. 
-> None of the devices would be managed by boths software. <-
I don't want to play with it, I just want to have all my devices on the same mesh network.

-> I think you are one of the best experts on this subject, so I would be very interested to understand why you say this is a bad idea. Given your level of expertise, what you tell me is decisive!

darkxst, 2024-05-27:
It wont work at all with SLZB-06M/Silabs chipset, the protocol simply doesnt allow it. 

Apparently it works with the CC2652 in SLZB-06 or UZG-01, but it is still a bad idea to be running it like that.

> would mean doubling all repeaters...

Depending on your network that may not be all that bad, it is possible to have too many routers, especially if you have lots of lights/wall switches etc

liouma, 2024-05-27:
Thank you I'll give up the 06M, you just saved me some time.
I was planning to buy the UZG-01 but the current version is based on CC2652P7 chip (since hardware v0.2). Would you go for this one or an older CC2652 ?

Why is it a bad idea? 
is it a bad idea even if every device is managed only by one software (zha OR z2m)?

About doubling the devices: 
- there are some areas with high redondancy where I could split the routers without problems
- and other "weak" areas / (remote areas or areas with big walls and too few devices) where I had to make (CC2530 + CC2591) antenna routers to cover it. In those areas, abandoning half of the routers won't be possible. I'd need to double them on 2 different channels.
- "medium" areas that may work with half the routers but with weak connexion especialy when covers / doors are down or it's raining.

darkxst, 2024-05-27:
> I was planning to buy the UZG-01 but the current version is based on CC2652P7 chip (since hardware v0.2). Would you go for this one or an older CC2652 ?

SLZB-06p7 is also based on CC2652P7, right now there is not much difference between firmware, but there is some scope to allow for larger networks on the P7 in the future. Its less tested though so may have new issues that dont exist in the very well tested and mature CC2652P. But probably I would go with the P7 ;)

> Why is it a bad idea?
> is it a bad idea even if every device is managed only by one software (zha OR z2m)?

Every device has information stored in the NVM of the coordinator, it is possible that ZHA or Z2M might have conflicting versions of this state or other options, you could get collisions between commands sent from either. It could cause weird issues that are hard to debug, and no one is going to give you support on bugs/issues in a configuration such as this. Even if each device was only paired with one stack there could be still many possible issues.

> * where I had to make (CC2530 + CC2591) antenna routers to cover it.

You can use the SLZB-06(or M) as router although its a bit overkill, other cheaper options are
Sonoff ZBDongle-E
New Sonoff ZBMicro
or any mains power smart plug

liouma, 2024-05-27:
Thanks a lot, that much more clear for me :)
- So I will try the P7.
- I will reduce the risk of collisions enabling each device only on one software.
- I understood that there may still be some conflicts and if I can't do otherwise I'll split the network between ZHA (slzb) and z2m (P7).

Thanks, I'll check the sonoff. When there's no powered device nearby I use this as router (cheap with antenna): https://www.zigbee2mqtt.io/advanced/zigbee/05_create_a_cc2530_router.html

closed 2026-01-13 by github-actions[bot]