darkxst/silabs-firmware-builder #96: Don't find SONOFF dongle E

Amonix87, 2024-05-30:
Web flasher can't find usb sonoff dongle E device.

Souks, 2024-06-04:
same for me and with the python package i got an error like "port com5 busy or already in use" 
i can't figure out why

GauthierCrp, 2024-06-04:
Hello,

Using Google Chrome on Fedora 40, I'm facing the error following : 
The dongle seems to be detected, but impossible to check the firmware installed on it. The firmware is the factory one, installed by Sonoff.

[image]

But : 

[image]

[image]

[image]

Someone can help me ? I try to upload the firmware because with the new version of Zigbee2Mqtt (Current version: 1.38.0-1) the Dongle is not working, detected but not functional. But, it was working just fine with the passed version.

Thank you very much for yout help.

Best Regards.
Gauthier.

darkxst, 2024-06-05:
If you can't connect to Dongle its usually one of these:
On Linux you need to add your user to the `dialout` group.
On Windows you probably need to install drivers for CH9102 chip.

darkxst, 2024-06-05:
Interesting, I have not seen a ZBdongle-E with cp2102 before, however this should have no impact on web flasher.

@andy858585 Are you on Windows or Linux? If Windows could be a driver issue?

darkxst, 2024-06-05:
Oh right, I have updated usb filters for zbdongle-e, it should work now.

Amonix87, 2024-06-05:
Thankx, Now it found (it's a CP2102N) but flash not working 
[silabs_flasher.log](https://github.com/user-attachments/files/15573440/silabs_flasher.log)

darkxst, 2024-06-05:
@Amonix87 Its failing to connect, but I not too sure what else could effect it on Windows. Can you try directly with `the python `universal-silabs-flasher -v` this might give a bit more useful logs.

Souks, 2024-06-05:
i managed to get a new firmware with the following method seen [here ](https://github.com/darkxst/silabs-firmware-builder/issues/56#issuecomment-1963133886)

```
Install SecureCRT (trial version is fine), or another similar software
Use it to flash Sonoff [factory NCP firmware](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP) (procedure described in [Sonoff PDF](https://sonoff.tech/wp-content/uploads/2022/07/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf))
Come back to darkxst flasher and flash the new firmware you want
```

Just the last step "Come back to darkxst flasher and flash the new firmware you want" didn't worked for me 
And flashing a rcp firmware didn't worked too, i end up flashing the stock firmware again to use it in home assistant (the key is working with stock firmware) 

I will try with the last update when i came home

Souks, 2024-06-05:
For me its working now !

Amonix87, 2024-06-06:
> @Amonix87 Its failing to connect, but I not too sure what else could effect it on Windows. Can you try directly with `the python `universal-silabs-flasher -v` this might give a bit more useful logs.

I don't know how make this. never used Python....

Amonix87, 2024-06-06:
`PS C:\Users\j_bp_> universal-silabs-flasher --device COM7 --bootloader-reset sonoff flash --firmware "D:\Téléchargement\ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl"
2024-06-06 18:07:53.759 DESKTOP-I7B50NR universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.3', ezsp_version='7.4.3.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-06-06 18:07:53.760 DESKTOP-I7B50NR universal_silabs_flasher.flasher INFO Triggering sonoff bootloader
2024-06-06 18:07:54.377 DESKTOP-I7B50NR universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-06-06 18:07:54.487 DESKTOP-I7B50NR universal_silabs_flasher.flasher INFO Launched application from bootloader
2024-06-06 18:07:55.528 DESKTOP-I7B50NR universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2024-06-06 18:07:55.528 DESKTOP-I7B50NR universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-06-06 18:07:56.836 DESKTOP-I7B50NR universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '6.10.3.0 build 297' (6.10.3.0.297) at 115200 baudrate (bootloader baudrate 115200)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Scripts\universal-silabs-flasher.exe\__main__.py", line 7, in <module>
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flash.py", line 408, in flash
    await flasher.enter_bootloader()
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flasher.py", line 261, in enter_bootloader
    async with self._connect_ezsp(self.app_baudrate) as ezsp:
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\emberznet.py", line 41, in connect_ezsp
    ezsp = await bellows.ezsp.EZSP.initialize(app_config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\bellows\ezsp\__init__.py", line 133, in initialize
    await ezsp.connect(use_thread=zigpy_config[conf.CONF_USE_THREAD])
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\bellows\ezsp\__init__.py", line 145, in connect
    self._gw = await bellows.uart.connect(self._config, self, use_thread=use_thread)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\bellows\uart.py", line 187, in connect
    protocol, _ = await _connect(config, application)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\bellows\uart.py", line 158, in _connect
    transport, _ = await zigpy.serial.create_serial_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\zigpy\serial.py", line 48, in create_serial_connection
[... cut ...]
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Users\j_bp_\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM7': PermissionError(13, 'Accès refusé.', None, 5)
PS C:\Users\j_bp_>`

Amonix87, 2024-06-07:
I was already in admin and I reboot pc. I don’t know how to find if There is a program who use the port…

Hedda, 2024-06-10:
FYI, while probably not your issue from the sounds of it, be aware that the USB-plug on these Sonoff dongles dongles and well as some others (as well as aother USB-devices/peripherals) are known to be too short so do not make proper physical contact when plugged into a computer with sunken USB-ports.

For example, the The chassi/enclosure walls of the Home Assistant Blue are very thick due to poor design that makes the USB-ports sunken, therefore these USB-dongles will not make proper contact when plugged into the Home Assistant Blue.

The workaround is to use a USB extension cable (which you should use regardless) that has a longer USB-plug to make sure it makes proper physical contact (i.e. that it can be plugged in all the way despite the thick chassi/enclosure walls of the Home Assistant Blue ). I mention this under tips here -> https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-v2-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-soc-mcu/442695

 *To avoid EMF interference strongly recommended buy and use → [Long USB extension cable ](https://itead.cc/product/1-5m-usb-male-to-female-extension-cable/)*
 * ***Another reason for using a USB extension cable is that the USB-plug design of ITead’s Sonoff Zigbee 3.0 USB dongle hardware adapters is a little short which makes it harder and sometimes impossible to plug it into some USB ports if the computer enclosure/chassis/casing is to thick around the USB-port as that will physically prevent the USB dongle from actually making a proper connection inside the USB port even if it looks like it is plugged in all the way that is possible.***
 * *Also follow all the general tips in [https://github.com/zigpy/zigpy/wiki/Generic-best-practice-tips-on-improving-Zigbee-network-range-and-general-stability ](https://github.com/zigpy/zigpy/wiki/Generic-best-practice-tips-on-improving-Zigbee-network-range-and-general-stability) as all Zigbee Coordinator adapters is very sensitive to EMI/RFI interference (e.g. a noisy radio frequency environment will jam the signal and prevent it from receiving all Zigbee messages to it without errors). Connecting the dongle via a long “shielded” USB extension cable in a USB 2.0 port or USB 2.0 hub (and not a USB 3.0 port) to get it away from EMF sources will usually help a lot if experiencing connection or pairing symptoms/issues.*

PS: I seen posts from some people than have taken a Dremel, power-drill or other power-tool to physically modify the Home Assistant Blue chassi/enclosure to fix the issue of sunken USB-ports. See picture from issue with ConBee and Aeotec dongles -> https://community.home-assistant.io/t/solved-conbee-ii-not-recognized-on-ha-blue-odroid-n2/265883

[image]

Hedda, 2024-06-10:
> _First dongle uses CH9102. Web flasher can detect it and flash firmware._ 
>
> ...
>
> _**Second dongle uses CP2102N. Web flasher cannot detect it.**_

hmm, never seen that before but I see now that they updated the specification list on their website to mention both those chips:

https://itead.cc/product/zigbee-3-0-usb-dongle/

[image]