darkxst/silabs-firmware-builder #99: Help with flashing the

FrozenGalaxy, 2024-06-11:
When I try to flash any firmware I run into this issue (see bottom), I got a ZB-GW04 (v1.2)

When I serial into it, I get the router firmware and can run commands
```
Z3RouterUSBDonlge>version
stack ver. [6.10.3 GA build 297]
```
But i want to flash the stock version again, My end goal is to use Zigbee on HA.

flashing error:

```
> python Elelabs_EzspFwUtility.py flash -f .\firmware.gbl -p COM7
2024/06/11 21:22:27 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2024/06/11 21:22:27 Elelabs_EzspFwUtility:   Usage:
2024/06/11 21:22:27 Elelabs_EzspFwUtility:   Allready in bootloader mode. No need to restart
2024/06/11 21:22:28 Elelabs_EzspFwUtility:   Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)
send error: expected NAK, CRC, EOT or CAN; got b' '
send error: expected NAK, CRC, EOT or CAN; got b'1'
send error: expected NAK, CRC, EOT or CAN; got b'0'
send error: expected NAK, CRC, EOT or CAN; got b' '
send error: expected NAK, CRC, EOT or CAN; got b's'
.. etc
2024/06/11 21:22:37 Elelabs_EzspFwUtility:   Firmware upload failed. Please try a correct firmware image or restart in normal mode.
```

darkxst, 2024-06-12:
I am not familiar with the Elelabs script, please try with `universal-silabs-flasher`

```
universal-silabs-flasher --device COM7 flash --firmware ./firmware.gbl
```
Also make sure you have downloaded the firmware via raw link on github (and not the html page!)

darkxst, 2024-06-13:
@FrozenGalaxy Your running Router firmware? you need to activate bootloader (boot button) before probe or flash

FrozenGalaxy, 2024-06-14:
> @FrozenGalaxy Your running Router firmware? you need to activate bootloader (boot button) before probe or flash

I did press that and then ran flashing but that didn't work, is there a button combination or so? And yes Router firmware

darkxst, 2024-06-14:
Hold Boot button, then tap reset button and release boot a few seconds later. Or you can just hold boot button while plugging into USB.

FrozenGalaxy, 2024-06-28:
Sorry for the delay, flashing works now, but .. while offtopic, which firmware is needed to have it working in Home Assistant either zigbee2mqtt or ZHA

closed 2024-06-28 by FrozenGalaxy