darkxst/silabs-firmware-builder #107: Impossibile to flash on SonoZigbee 3.0 USB

PetFra, 2024-07-15:
Hello there, I'm trying to flash a classic Sonoff to have Thread on it.
I tried on windows and Mac, but doesn't work.
I attach the log, I also tried with the python script but it didn't work for sure because of my poor Python Skill.

[silabs_flasher.log](https://github.com/user-attachments/files/16236188/silabs_flasher.log)

Any help would be highly apprecciated. Thanks

darkxst, 2024-07-16:
The flasher is not getting any response from the dongle. Is this ZBdongle-E? what firmware is on it currently?

PetFra, 2024-07-16:
Ho thanks, is the classic Sonoff ZigBee 3.0 dongle

as far as I remember is the original firmware on it. 

Would using the python script bypass the issue?

darkxst, 2024-07-16:
Sonoff have two different dongles, ZBDongle-E is compatible with my web flsher and these firmwares. The other one ZBDongle-P is a different chipset and requires different flasher and firmware from:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Make sure you have ZBDongle-E first!

PetFra, 2024-07-16:
I think problem is I have Dongle P
This is mine: https://amzn.eu/d/0d47x3yq

and it says dongle P
So I need to understand if I can make thread over it

thanks a lot for your input

darkxst, 2024-07-17:
Currently there are no Thread firmwares for the P dongles (that I am aware of anyway). I might release Thread firmware for these sometime in the future...

For now, you can however build your own firmware from this repo:
https://github.com/TexasInstruments/ot-ti/commits/main

PetFra, 2024-07-20:
Hi thanks! 
I’ll look into it!

thank very much

TarasRybak, 2024-08-08:
Hello.
look at my logs too, please. for some reason it does not want to connect to Windows 11, Chrome or Edge. I installed universal drivers on 07/19/2024. CP210x Universal Windows Driver v11.4.0.
[silabs_flasher (1).log](https://github.com/user-attachments/files/16553882/silabs_flasher.1.log)
[image]

TarasRybak, 2024-08-09:
i tried to update locally. on a Proxmox machine. on a home assistant. installed venv, installed the flasher. tried two versions 7.4.2 and 7.4.3. but i have another problem.
[image]

darkxst, 2024-08-09:
@TarasRybak You need to use the raw link from github, otherwise its just downloading the html page.

https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

Dizzy49, 2024-08-10:
[silabs_flasher.log](https://github.com/user-attachments/files/16568718/silabs_flasher.log)

I am having the exact same issue as [Cannot flash my Sonoff ZBDongle-E with web flasher](https://github.com/darkxst/silabs-firmware-builder/issues/27) except mine is not V2
[image]

I have tried every USB port on my desktop and get the same error.  I just bought this from itead less than two weeks ago.  Is it possible it is already updated?  How can I find out?

darkxst, 2024-08-11:
> except mine is not V2

@Dizzy49 Are you sure this is ZBDongle-E and not ZBDongle-P (which requires different firmware and flashing tools)?

Dizzy49, 2024-08-12:
Yes, I ordered it from ITEAD and the item is "Zigbee 3.0 USB Dongle Plus--
ZBDongle-E"
I also verified on the back of the device.

Mdumas81, 2024-08-27:
Having the same issue. Been trying to troubleshoot all night.

I have the dongle-E and out of the box it was working fine. I started reading up on things and was suggested to flash with the newer firmware to use ember or whatever. Tool recognizes the dongle it looks like it connects for a second, then spins out and says cannot read firmware.

Tried 3 cables, ports on both front of my PC, back of PC, some USB 2.0 and some 3.1 and 3.2.

darkxst, 2024-08-27:
I suppose this is causing it to fail, but not sure why...
```
2024-08-27 01:18:49 emscripten bellows.ezsp[42] DEBUG Resetting EZSP
2024-08-27 01:18:49 emscripten bellows.uart[42] DEBUG Resetting ASH
2024-08-27 01:18:49 emscripten bellows.uart[42] DEBUG Sending: b'1ac038bc7e'
2024-08-27 01:18:50 emscripten bellows.uart[42] DEBUG RSTACK Version: 2 Reason: RESET_POWER_ON frame: b'c102029b7b7e'
2024-08-27 01:18:50 emscripten bellows.ezsp[42] INFO NCP entered failed state. No application handler registered, ignoring...
2024-08-27 01:18:54 emscripten bellows.uart[42] DEBUG Connection lost: None
2024-08-27 01:18:54 emscripten bellows.uart[42] DEBUG Closed serial connection
```

> I have a HA green and I am brand new to all this, so I will totally give that a try.

There is a hidden setting in the silabs flasher addon (Click the button to show unused settings) that you can use to put a custom URL into. Just make sure you use the RAW url from github page.

Mdumas81, 2024-08-27:
thank you, i think the firmware updated but i cant tell as the only way i know is to look from within z2m but z2m is now throwing errors

darkxst, 2024-08-27:
> thank you, i think the firmware updated but i cant tell as the only way i know is to look from within z2m but z2m is now throwing errors

Are you using `ember` backend in Z2M? it should work fine with my 7.4.x firmwares

Mdumas81, 2024-08-27:
[image]

darkxst, 2024-08-27:
which firmware did you flash?

Mdumas81, 2024-08-27:
ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

Mdumas81, 2024-08-27:
ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl seems to be flashing properly, Going to try that

darkxst, 2024-08-27:
both those should work, there werent really any changes between 7.4.2 and 7.4.3.

Mdumas81, 2024-08-27:
ok so 7.4.2 worked it looks like. Im getting a ton more info

```
Starting Zigbee2MQTT without watchdog.
[2024-08-27 02:11:16] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-27 02:11:16] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-27 02:11:16] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-27 02:11:17] info: 	zh:ember: Using default stack config.
[2024-08-27 02:11:17] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-08-27 02:11:17] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-08-27 02:11:17] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-08-27 02:11:17] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-08-27 02:11:17] info: 	zh:ember:uart:ash: Serial port opened
[2024-08-27 02:11:17] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-08-27 02:11:18] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-08-27 02:11:18] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-08-27 02:11:18] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-08-27 02:11:19] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-08-27 02:11:19] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.2 [GA]","build":0,"major":7,"minor":4,"patch":2,"special":0,"type":170}
[2024-08-27 02:11:19] info: 	zh:ember: [STACK STATUS] Network up.
[2024-08-27 02:11:19] info: 	zh:ember: [INIT TC] Adapter network does not match config. Leaving network...
[2024-08-27 02:11:20] info: 	zh:ember: [STACK STATUS] Network down.
[2024-08-27 02:11:20] info: 	zh:ember: [INIT TC] No valid backup found.
[2024-08-27 02:11:20] info: 	zh:ember: [INIT TC] Forming from config.
[2024-08-27 02:11:20] info: 	zh:ember: [INIT FORM] Forming new network with: {"panId":28037,"extendedPanId":[230,91,206,189,106,92,214,131],"radioTxPower":5,"radioChannel":11,"joinMethod":0,"nwkManagerId":0,"nwkUpdateId":0,"channels":134215680}
[2024-08-27 02:11:20] info: 	zh:ember: [STACK STATUS] Network up.
[2024-08-27 02:11:20] info: 	zh:ember: [INIT FORM] New network formed!
[2024-08-27 02:11:20] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1247ms until next broadcast.
[2024-08-27 02:11:20] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-08-27 02:11:20] info: 	z2m: zigbee-herdsman started (reset)
[2024-08-27 02:11:20] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":2,"revision":"7.4.2 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-08-27 02:11:20] info: 	z2m: Currently 0 devices are joined:
[2024-08-27 02:11:20] info: 	z2m: Zigbee: disabling joining new devices.
[2024-08-27 02:11:21] info: 	z2m: Connecting to MQTT server at mqtt://192.168.1.4:1883
[2024-08-27 02:11:21] info: 	z2m: Connected to MQTT server
[2024-08-27 02:11:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-08-27 02:11:21] info: 	z2m: Started frontend on port 8099
[2024-08-27 02:11:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-08-27 02:11:22] info: 	z2m: Zigbee2MQTT started!
[2024-08-27 02:11:27] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x6c5cb1fffe992e53/connection_state/config', payload '{"device":{"hw_version":"EmberZNet 7.4.2 [GA]","identifiers":["zigbee2mqtt_bridge_0x6c5cb1fffe992e53"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.39.1"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.39.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x6c5cb1fffe992e53_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2024-08-27 02:11:27] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x6c5cb1fffe992e53/restart_required/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"EmberZNet 7.4.2 [GA]","identifiers":["zigbee2mqtt_bridge_0x6c5cb1fffe992e53"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.39.1"},"device_class":"problem","enabled_by_default":false,"entity_category":"diagnostic","name":"Restart required","object_id":"zigbee2mqtt_bridge_restart_required","origin":{"name":"Zigbee2MQTT","sw":"1.39.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x6c5cb1fffe992e53_restart_required_zigbee2mqtt","value_template":"{{ value_json.restart_required }}"}'
[2024-08-27 02:11:27] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/button/1221051039810110150109113116116_0x6c5cb1fffe992e53/restart/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/restart","device":{"hw_version":"EmberZNet 7.4.2 [GA]","identifiers":["zigbee2mqtt_bridge_0x6c5cb1fffe992e53"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name [... cut ...] fe992e53"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.39.1"},"icon":"mdi:human-greeting-proximity","name":"Permit join","object_id":"zigbee2mqtt_bridge_permit_join","origin":{"name":"Zigbee2MQTT","sw":"1.39.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x6c5cb1fffe992e53_permit_join_zigbee2mqtt","value_template":"{{ value_json.permit_join | lower }}"}'
```

darkxst, 2024-08-28:
> I suppose this is causing it to fail, but not sure why...
>
> ```
> 2024-08-27 01:18:49 emscripten bellows.ezsp[42] DEBUG Resetting EZSP
> 2024-08-27 01:18:49 emscripten bellows.uart[42] DEBUG Resetting [...]

can someone with the above errors, try test with this patched flasher
https://darkxst.github.io/cp-test/

NickFurlo, 2024-08-31:
Came across this issue while trying to flash my new ZBDongle-E. cp-test appears to work in that it recognizes the firmware version and provides an options to upgrade, but the install immediately fails for me. 

[silabs_flasher (2).log](https://github.com/user-attachments/files/16826093/silabs_flasher.2.log)

yvess16, 2024-09-08:
> can someone with the above errors, try test with this patched flasher https://darkxst.github.io/cp-test/

just came across this, as i had the same issues. Exact same logs as you mentioned. Tried to flash on https://darkxst.github.io/cp-test/ and it worked fine! Thanks so much!
Out of curiosity, what changed?

giandena, 2024-10-02:
> just came across this, as i had the same issues. Exact same logs as you mentioned. Tried to flash on https://darkxst.github.io/cp-test/ and it worked fine! Thanks so much! Out of curiosity, what [...]

Using this tool I finally saw the update page, unfortunately, the update is not completed anyway. I've been waiting for 30 minutes but nothing happens. 
[image]
[image]

darkxst, 2024-10-02:
> Out of curiosity, what changed?

Sonoff changed the usb serial chip, the windows driver for this new chip (cp2102) has some quirks that reset device at an inconvenient time.

darkxst, 2024-10-02:
> Using this tool I finally saw the update page, unfortunately, the update is not completed anyway. I've been waiting for 30 minutes but nothing happens.

Try get logs from developer tools (F12)

giandena, 2024-10-02:
[darkxst.github.io-1727861819807.log](https://github.com/user-attachments/files/17227462/darkxst.github.io-1727861819807.log)

pashdown, 2024-10-06:
- MacOS Sequoia 15.0.1
- SiLabs VCP MacOSX Driver
- cu.SLAB_USBtoUART
- Sonoff ZBDongle-E

Unable to identify existing default firmware with normal and cp-test flasher.  Complains about the buggy macOS driver even though I'm using SiLabs.  LED changes from red to green in the process.  Here are the cp-test logs:
[silabs_flasher (1).log](https://github.com/user-attachments/files/17270554/silabs_flasher.1.log)

giandena, 2024-10-09:
Solved ! I was able to update the firmware using Secure CRT on windows10.
The official instructions do not explain exactly how to start the boot phase and then press the key 1.
The most important step! 
Once the device is connected you need to remove it and reinsert it into the USB port again while holding down the boot key on the device. 
At this point in SecureCRT you need to press the reconnect button. Only in this way the menu appear that allows you to update the firmware.
[image]

quickquest88, 2024-10-15:
I'm having issues with this too.  Def have the zbdongle-e, when I open the web tool, it will recognize the dongle, then throw an error about not connecting.

[silabs_flasher (1).log](https://github.com/user-attachments/files/17370450/silabs_flasher.1.log)
[silabs_flasher.log](https://github.com/user-attachments/files/17370451/silabs_flasher.log)

can some one help, please?

darkxst, 2024-10-15:
There are known issues with the windows driver resetting device when we open serial port. Not managed to work around this yet. 

If you are using Home Assistant, easiest option is to use the "Silabs Flasher" addon to flash dongle. This has a hidden option to specify custom URL to flash.

quickquest88, 2024-10-15:
That didn't work either. "Universal-silabs-flasher-up exited code 6."

darkxst, 2024-10-15:
can you paste full logs

quickquest88, 2024-10-15:
Silicon Labs Flasher
[Info](http://192.168.50.88:8123/hassio/addon/core_silabs_flasher/info)
[Documentation](http://192.168.50.88:8123/hassio/addon/core_silabs_flasher/documentation)
[Configuration](http://192.168.50.88:8123/hassio/addon/core_silabs_flasher/config)
[Log](http://192.168.50.88:8123/hassio/addon/core_silabs_flasher/logs)
Silicon Labs Flasher
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: Silicon Labs Flasher
 Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.10.2
 Home Assistant Supervisor: 2024.10.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[19:51:38] INFO: universal-silabs-flasher-up script exited with code 6
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 6
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: Silicon Labs Flasher
 Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.10.2
 Home Assistant Supervisor: 2024.10.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[19:51:42] INFO: universal-silabs-flasher-up script exited with code 6
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 6
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: Silicon Labs Flasher
 Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.10.2
 Home Assistant Supervisor: 2024.10.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[19:53:22] INFO: universal-silabs-flasher-up script exited with code 6
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 6
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

darkxst, 2024-10-15:
Is something else connected to dongle? make sure ZHA or Z2M dont have active connection to dongle while flashing

quickquest88, 2024-10-15:
I do not think I have any zigbwe integrations or add ones.  I did uninstall silicon labs multiprotocol.

Do I need to configure the dongle on the devices page?

darkxst, 2024-10-15:
> Do I need to configure the dongle on the devices page?

yes you need to configure the dongle as device in silabs flasher addon

quickquest88, 2024-10-15:
Sorry for the delay, still no go. 

 The flasher just sits there and doesn't do anything.  I set the device up on the devices page and started a new network.  Still nothing.

Vaiperu, 2024-10-18:
I have the same issue. Ordered the ZBDongle-E 3 with the SL chip EFR32MG21

Device Manager shows it as "Silicon Labs CP210x USB to UART Bridge (COM5)" (I installed the driver manually from slabs.com

Logs:
`Connection failed
The running firmware could not be detected.

Make sure the USB port works and if you are using a USB extension cable, make sure the cable can transfer data. Unplug the Sonoff ZBDongle-E and plug it back in to reset and try again.`

`2024-10-18 21:11:58 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-10-18 21:11:58 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2024-10-18 21:11:58 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x15be540>
2024-10-18 21:11:58 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2024-10-18 21:12:00 emscripten webserial_transport[42] DEBUG Closing serial port
2024-10-18 21:12:00 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2024-10-18 21:12:00 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
2024-10-18 21:12:00 emscripten webserial_transport[42] DEBUG Closed serial port
2024-10-18 21:12:00 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1aab508>
2024-10-18 21:12:00 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-10-18 21:12:00 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-10-18 21:12:00 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'x'
2024-10-18 21:12:00 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x00'
2024-10-18 21:12:01 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2024-10-18 21:12:01 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\x80'
2024-10-18 21:12:01 emscripten universal_silabs_flasher.common[42] DEBUG Received data b'\xf8\x00x\x80\x00\x80\x00\xf8\xf8\x80\xf8\xf8x\xf8'
2024-10-18 21:12:01 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'\xf8'): ValueError('Invalid flag')
2024-10-18 21:12:01 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-10-18 21:12:01 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-10-18 21:12:02 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2024-10-18 21:12:02 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-10-18 21:12:02 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-10-18 21:12:03 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2024-10-18 21:12:03 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-10-18 21:12:03 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-10-18 21:12:04 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
[... cut ...]
2024-10-18 21:12:27 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'~\x83\x02\x02\xe65~'
2024-10-18 21:12:28 emscripten universal_silabs_flasher.spinel[42] DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
2024-10-18 21:12:28 emscripten webserial_transport[42] DEBUG Closing serial port
2024-10-18 21:12:28 emscripten webserial_transport[42] DEBUG Closed serial port
`

quickquest88, 2024-10-18:
When I try to flash using the web tool, i get this log

[silabs_flasher (3).log](https://github.com/user-attachments/files/17441464/silabs_flasher.3.log)

quickquest88, 2024-10-19:
Now I have found this 
"For ubuntu, you have to allow chromium to acces your usb devices: Settings > Apps > Chromium Web Browser > Permissions > Access USB Hardware directly"

WTF does that mean? I have installed Ubuntu.  There is no "settings". There is "system settings", but there is no "applications" there, only "applets", and chromium doesn't appear on that list.  How do I find this setting?

tootla, 2024-10-25:
I've been trying to troubleshoot the issue for the past week, and up until yesterday, I was still unable to flash the firmware. But this morning, when I tried again on the website([https://darkxst.github.io/silabs-firmware-builder/](https://darkxst.github.io/silabs-firmware-builder/)), it just worked! Then I plugged in my other ZBDongle-E to test, and it also succeeded! Did you fix it? @darkxst 
[image]

This is the information of my device:
- ZBDongle-E: EFR32MG21, CP2102N, Windows 11 , Chrome browser . And the driver version of CP2102N is 6.7.0.0.

Here is the success logs:
[ZBDonglee_success_silabs_flasher.log](https://github.com/user-attachments/files/17514724/ZBDonglee_success_silabs_flasher.log)

darkxst, 2024-10-25:
> Did you fix it? @darkxst

No, I havent changed anything in main site.

> * And the driver version of CP2102N is 6.7.0.0.

Was there a driver update recently perhaps? I had reported these issues to Sonoff, dont know if they managed to perhaps get  Silabs to fix driver issues.

tootla, 2024-10-25:
> > Did you fix it? @darkxst
> 
> No, I havent changed anything in main site.
> 
> > * And the driver version of CP2102N is 6.7.0.0.
> 
> Was there a driver update recently perhaps?
> 
> EDIT: That seems to be an older driver version, perhaps from before the reset bug was introduced in the driver.

Yes, I uninstalled the original 11.x version of the Windows driver and then installed the driver named CP210x VCP Windows from the [Silabs website](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads). 
After the installation was complete, I checked and found that the driver version had changed to 6.7.0.0, and then I was able to flash the firmware through the web tool.
[image]

To confirm that it was a driver issue, I reinstalled the previous 11.x version driver, and sure enough, I encountered an error when flashing the firmware.

If someone encounter the same issue, check your driver version, and try this:

> Go to the Windows Device Manager and uninstall the existing driver for ZBDongle-E (Silicon Labs CP210x USB to UART Bridge), remember to select the option to try to remove the driver software.
> Download and install the 6.7.0.0 driver ([CP210x VCP Windows](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads)).

miquelromero, 2024-10-25:
@tootla Same solution:
- Uninstall universal drivers (uninstall the device from windowsdevice manager)
- Install the CP210X VCP Windows v6.7 (.exe)
- Upgrade firmware from https://darkxst.github.io/cp-test/ (the [original one](https://darkxst.github.io/silabs-firmware-builder/) is still not working)

giandena, 2024-10-28:
> Yes, I uninstalled the original 11.x version of the Windows driver and then installed the driver named CP210x VCP Windows from the [Silabs [...]

This solution is working !
Thanks

erkr, 2024-10-30:
Thanks, reverting the device driver to "CP210x VCP Windows" fixed it for me as well

erkr, 2024-11-04:
Good summary of the options 👌

tootla, 2024-11-04:
> I installed windows on an old laptop to look into this further. Unfortunately I can't work around the Silabs driver issues, at least not without a major rewrite of the underlying flashing tools, which I wont be doing.我在一台旧笔记本电脑上安装了 Windows 以进一步研究这个问题。不幸的是，我无法解决 Silabs 驱动程序问题，至少如果不对底层刷写工具进行重大重写，我不会这样做。
> 
> So if your having trouble flashing on windows you have the following options:因此，如果您在 Windows 上刷写时遇到问题，您有以下选择：
> 
> * If you use Home Assistant, install the `Silabs flasher addon` and use that.如果您使用 Home Assistant，请安装 `Silabs flasher 插件`并使用它。
> * or Install the `cp210x VCP drivers` as detailed above.或如上所述安装 `cp210x VCP 驱动程序`。
> * or Remove the case from your dongle and just use the boot button或从加密狗中取出外壳，然后使用启动按钮

What puzzles me is why SMLIGHT SLZB-07 wouldn't have this problem.In my view, their hardware is identical.

darkxst, 2024-11-04:
> What puzzles me is why SMLIGHT SLZB-07 wouldn't have this problem.

Sonoff dongles have a bootloader reset functionality hooked up the flow control lines, the driver is triggering a reset via this on opening the serial port.  This wasnt a problem with the previous usb chip, but now they changed to cp2102 and it is affected by the windows driver bug.

SLZB-07 has actual hard flow control, so can't be accidentally reset in the same way.

R4fte, 2024-11-06:
Hi back I add a comment here @darkxst https://github.com/darkxst/silabs-firmware-builder/issues/142, I'm not big user of Git so idk if its better to speak here and not on my comment, so I repost here. 
Thx for answere !
(Sry for my English)

tootla, 2024-11-07:
> Hi back I add a comment here @darkxst #142, I'm not big user of Git so idk if its better to speak here and not on my comment, so I repost here. Thx for answere ! (Sry for my English)

Could you reinsert the dongle and then go to the web tool to see if it can be recognized?

R4fte, 2024-11-10:
Sorry I don't see ur message, I try it and my web tool recignized it 
[image]
@tootla

R4fte, 2024-11-10:
Okay I try it with that URL and that work https://darkxst.github.io/cp-test/
I dont know why x) but thx

R4fte, 2024-11-10:
Hi @tootla and @darkxst now the problem is different, I use the new link but its like infinit loading, I try in first to update the firmware like how comment say but nothing happend too. I dont know how I can  fix ?
[image]

darkxst, 2024-11-10:
@R4fte Its the same issue, just the test site gets a little further before failing. Use on of the workarounds https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2454076415

JohnRo79, 2024-11-12:
i've used the site quite a few times, but today, it just won't work
every time i've tried the freshly downloaded gbl, it just giving that error
(fresh windows 11 24h2 )

[image]